FormerLurker/Octolapse #225: [Request] Setting a GPIO port instead of grabbing an image

EddyMI3d, 2018-09-03:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

Is it possible to set a GPIO port of choice instead of grabbing an image from the web cam stream?
It would be possible to trigger an external camera.

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE
V 0.3.1

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE
V 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

OutlawAndy, 2018-09-03:
see issue #187

FormerLurker, 2018-09-03:
@EddyMI3d,

@OutlawAndy is correct.  The development code already supports this feature.  You can try it out from the newest rc/devel commit by installing through the plugin manager with [this url](https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev.zip).

[This guide should help you to get a DSLR working using gphoto2 and octopi.](https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera)  Even if you are using a different capture method, it should give you insight as to how it works.  There are also some additional scripts that are now supported:  Camera Initialize, Before Snapshot, Capture Snapshot, After Snapshot, Before Render, After Render.

FYI, there is something wrong with the custom script timeouts in Windows.  If your script gets stuck it will NEVER timeout on windows (at least until I'm able to fix it).

Good luck, and let me know if it works!

FormerLurker, 2019-10-12:
OK, so I've seen some people who were able to trigger via GPIO using custom scripting that has been supported since Octolapse V0.3.4.  This has been further enhanced in V0.4 (unreleased), so I'm considering this closed.  I am still planning to add more direct support for GPIO pins, and an issue exists for that enhancement, so I'm going to close this issue.  Let me know if you'd like this reopened, or simply create a new issue.  Thank you very much for your post!

closed 2019-10-12 by FormerLurker