FormerLurker/Octolapse #227: Add a default - No Rendering - Profile

FormerLurker, 2018-09-04:
Add a new rendering profile that will render nothing.  Basically it's just a regular rendering profile but will be disabled.

Horizonistic, 2019-03-17:
I would love this!  I'm transferring then deleting images as they come via scp to an unRAID server I have, but at the end of the print I want to use a Python script to start a custom render using the images I've transferred.  The problem is that the only way to have a script run at the end of a print is via the pre or post render script, but both of those only run if rendering is enabled, which I don't want!  

I suppose if I'm deleting images off the Pi as they come in anyway the render won't any anything to work with anyway, but it seems like a workaround and it also feels dirty to deny the process all the images. :(

FormerLurker, 2019-10-12:
This has been completed, and will be available in Octolapse V0.4, which is currently in pre-release!  Closed!

closed 2019-10-12 by FormerLurker