FormerLurker/Octolapse #228: Octolapse crash with custom GCODE

wjax, 2018-09-07:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. PRINT

#### What should have happened?
Octolapse to record and render a timelapse video

#### What happened instead?
Octolapse didnt
I use a custom GCODE to do some things before actually start printing. Octolapse may be crashing beause of that

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/wjax/1a31ae9c1a125ad687d84e4bcb31d044

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-09-07:
Hi, and thanks for posting your issue!

I have a few questions to make sure I'm understanding what is going on:

1.  When you say 'crashed' do you mean you server crashed, or did Octolapse just take no pictures?  Any additional details about exactly what happened (or what didn't happen) would be extremely helpful.
2.  Did you see any error messages before, during or after printing?
3.  Would you be willing to try upgrading to a development version to see if maybe your issue has been fixed since the 0.3.1 version of Octolapse (the current alpha is 131 commits ahead of the release version, and is waiting for testing feedback from users before being promoted to a beta release candidate)?  That way I could get some additional help testing the new version.  Let me know if you'd be interested in this.  If you are you can skip items 1 and 2 for now, and I'll give you some upgrade instructions.

Thanks!

AbuMaia, 2018-09-21:
I am seeing this same problem, except for me it is caused by adding a "M900 K0.2" to the beginning of the Start Gcode section in Slic3r, to activate the Linear Pressure Control feature in Marlin 1.1.9.  If I comment out the "M900 K0.2", Octolapse runs as it should.

FormerLurker, 2019-10-12:
So this issue has been dead for a while, and Octolapse has been changed a great deal since this was opened, so I'm going to go ahead and close this issue.  Please let me know if you'd like it reopened, and thanks for your submission!  If you have any other problems, feel free to create another issue.

Thanks!

closed 2019-10-12 by FormerLurker