FormerLurker/Octolapse #229: [Request] Add option for simple timelapse that won't create artifacts

JustinB4, 2018-09-08:
#### If this is a feature request describe it here

Add a feature that will do the same thing, or similar to the built-in timelapse that doesn't affect the output of the print.  I love the mp4 and fixed length capabilities but I really don't like that no matter what settings I use, even with stabilization disabled, Octolapse introduces artifacts.  I know that there are things that can be tweaked to get it to work better like, ZHop and retract only but depending on what you are printing that isn't always an option.  I'd like to be able to set it and forget it and have all of my timelapses come out without having to worry about adjusting settings every time.  I also don't want to make any of my prints take any longer.
[image]

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-09-08:
Thanks for posting @JustinB4.  I'm going to reply with a novel, so I hope you don't mind.

I would LOVE to implement your request, but unfortunately I've not been able to come up with any way to do this without some settings tweaks.  Before I get into the technical details, let's see if we can't figure out some way to reduce the artifacts you are seeing.

1.  Even if you don't want to mess with your settings for every print, you are DEFINITELY going to want to adjust these variables within your printer profile:  Retraction Length, Retraction Speed, Detraction Speed, Z-Hop Height.  It is VERY VERY important that these are set appropriately (even with stabilization disabled), else you WILL experience artifacts like the ones you are currently seeing.  Just copy the values from your slicer and make sure the units match (mm/sec vs mm/min, there is a 60X difference between the two).  If you have multiple configurations, you can create multiple printer profiles so that it's easy to switch between them (see Octoprint Settings (wrench)->Octolapse.  There you can copy or create new profiles from the defaults).
2.  The artifacts you are seeing are pretty extreme.  It suggests to me two possibilities:  Your retraction length/speeds are way off, or you have a very large snapshot delay.  Possibly both.  Can you let me know what your snapshot delay is (camera settings)?
3.  What snapshot profile are you using?  Some perform better than others when it comes to print quality.  The default profile is intended for compatibility, since new users need the simplest possible setup.  There are other profiles (zhop and retract only, gcode trigger) that deliver a higher quality.
4.  You can tweak your camera to take faster images, most likely, especially if you are using lots of automatic features (focus, white balance, exposure, etc).  These operations take time, and you are better off using 100% manual web cam settings.  Check out [this](https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-image-preferences) guide to tweaking your camera settings.  The part about changes to octopi.txt are especially important.

Take a look at [this](https://github.com/FormerLurker/Octolapse/wiki/Improving-Print-Quality) guide regarding increasing print quality.  It should explain a lot about what Octolapse can do regarding print quality, and tries to explain how it works.

Now, let's talk about some changes to a new alpha version that are being tested(which can be found [here](https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev)):
1.  Feature Detection - This allows you to restrict Octolapse to certain print features, like infill, interior perimeters, supports, etc.  I've been able to reduce artifacts pretty much to 0 using this new feature.  Unfortunately there are MORE slicer settings that have to be copied AND some may have to be changed a bit.
2.  Enter printer settings based on your slicer - I've added a drop-down box to the printer profiles page that allows you to select your current slicer, and all of the wording, units, and settings change to match your slicer, making it very easy to figure out where the settings are, and to copy them into Octolapse.
3.  Some other tweaks and bugfixes are built into the new version to deal with issues that were affecting quality.  This includes some good old-fashoned bugs as well as some improvements, like a reduction in the time it takes to process snapshot.

Now let me mention some features I hope to add soon:
1.  I want to automatically extract the slicer settings from your gocde file.  Most slicers put the settings somewhere in the file, and I could pre-process the file and extract the settings.  It's tricky business, however, and it may add some time to the beginning of the print in order to read the gcode.
2.  By pre-processing the gocde file I could also detect problems with start/end gcode, could estimate the amount of time Octolapse will add to the print, and could report any other problems.
3.  One more possibility (though much more advanced than the other two, so it will come later) is to add a new stabilization that uses gcode pre-processing.  For example, Octolapse could stabilize to the part itself, say the frontmost and leftmost corners.  That way you could get a pretty good timelapse without actually moving the X/Y axis at all!  There are lots of other possibilities here.

Finally, I'd like to discuss what Octolapse is doing and why it's difficult to remove these pesky settings.  Octolapse is provided the same gcodes in the same order as your printer.  Just as your printer would not know how to properly lift and retract without being told, Octolapse also needs to be told how much to retract/detract and at what speed.  In fact, Octolapse has to generate its own gcode, so it is more like your slicer in that case.  If you feed your slicer bad retraction/detraction settings there is a high probability that your print quality would suffer.  When developing Octolapse I'd say that roughly 50% of the development time was spent getting print quality as high as possible.  It's my number 1 priority.  If I can figure out a way to get higher quality with less work for the user, I'm 100% in.  Howerver, whenever I figure out a way to improve quality, it seems to always require more knowledge about how the file was sliced, and thus even more settings.  The pre-processing method I mentioned above could help to mitigate this, but it will take some time to develop.  

I hope that helps somewhat.  Please let me know if you have any questions, or if I can do anything more to help.

JustinB4, 2018-09-08:
Thank you very much for your prompt and thorough response.  I will definitely adjust my settings.  I had seen the guide before but couldn't find the settings in Cura to copy.  I'll have to dig deeper.  I'll reply back if I need any more help.

JustinB4, 2018-09-08:
Awesome.  I'm going to check it out now.  Thanks!

FormerLurker, 2018-09-08:
Hey, try this link to install: https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev2.zip

I just fixed that settings migration bug I mentioned in that version, as well as a windows timeout error.

JustinB4, 2018-09-09:
So, I had installed before you updated it and I made all of the settings match Cura, but after re-reading your notes and thinking about it, I think the real problem is that I had some settings disabled (on purpose) but I don't think I can disable those settings and still get a quality print.  I have a bed mounted camera and I don't want the print to pause at all for each snapshot.  I'm not sure if that is even possible with Octolapse, but that is what I was trying to do by disabling stabilization and disabling retract before move and lift before move.  Am I right that I can't change settings in Octolapse to completely disable the snapshot delay?  I'm doing a test now with those settings re-enabled and the quality appears to be much better (not perfect) but the speed is not.  Thanks again for your help.

FormerLurker, 2018-09-09:
I'd leave retract enabled (snapshot settings) since there will be a slight delay while capturing a snapshot.  It won't add hardly any time at all to your print, and if you are using high quality mode those minor defects.

You can remove the snapshot delay, but you might get some motion blur.  The delay required to get a perfect image is 1/framerate(in seconds) *1000, or 31.25 for a 30fps stream.  That means 3.125 seconds of delay for 100 snapshots.  Post processing (translation and/or overlays) adds a bit of time.  It should end up being well under 1 second per image though.  

The time it takes to lift and retract is minimal too.  You can get away with disabling zlift (snapshot settings) since you are not doing any traveling, I believe, and still get great results.

When you say the speed is not better, do you just mean the time it takes to acquire an image?

Ive been working to further reduce the snapshot acquisition time, fyi.  My goal is to have a best case of around 125MS at 1080 resolution.

JustinB4, 2018-09-09:
In this particular print, it is a spiralized outer contour so it greatly increased print time but in a normal print, I suppose it wouldn't be so bad.  Also, because it is only a spiralized outer contour, the quality is noticeably different.  With lift and retract turned on it is a lot better though.  I think in a normal print (not this specific test) with the new options to not capture on the outer walls, I think it will be fine.
[image]

JustinB4, 2018-09-09:
Also, I'm not concerned with motion in the capture because the camera is mounted to the print bed so the only thing moving would be the hot end assembly and I don't mind motion blur on that.

FormerLurker, 2018-09-09:
That makes more sense!  Have you thought about setting the layer trigger so that it triggers only after a certain height?  Look at the layer trigger under the trigger type drop down.  Set it to .2mm or so and see if that helps?

JustinB4, 2018-09-09:
Yes, I actually had it set to .25.  I may play around with some other numbers as well.  I haven't tested a normal print yet but I'm sure it will be fine.  Thanks for all of your help with this.

FormerLurker, 2018-09-09:
It's been a pleasure.  I'll leave this thread open in case you have any further issues/questions.

FormerLurker, 2018-09-16:
@JustinB4, I'm going to go ahead and close the thread.  If you have any additional problems related to this issue you can respond here and I'll re-open the issue.

Thanks!

closed 2018-09-16 by FormerLurker