FormerLurker/Octolapse #232: Feedback for release v0.3.4rc1.dev2

FormerLurker, 2018-09-08:
This issue is for any general feedback regarding the new rc/devel release of Octolapse.  Let me know if you have any bugs, issues, or if things are working just great!  [See the release notes here.](https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev2)

SystemDisc, 2018-09-11:
There should be the ability to save videos as MP4 using the H.264/AVC codec.

SystemDisc, 2018-09-11:
Also, the last print I did with the latest pre-release did not record anything.

FormerLurker, 2018-09-11:
@SystemDisc,

First, thanks a lot for trying the new build and posting feedback!  It REALLY helps.

So, you can use the h.264/AVC codec.  Check the Output Format (Codec) in the rendering profile.  You can choose your output format here.  I have to warn you that I sometimes get malloc errors from avconv (ffmpeg) using this codec if the snapshot resolution is too high.  I haven't figured out how to prevent this 100% percent of the time.  It generally doesn't affect webcams since they are usually at or under 1080P, but 4K+ snapshots seem to always fail, even on my PC with 16gb ram.

Let's get to the bottom of your issue where Octolapse isn't recording anything.  What does your position state info panel show (Octolapse tab->Expand Current Settings->Enable Position State) while printing after the first layer?  Also, can you tell me which pre-sets you are using?  Better still, post your gcode and your Octolapse settings (Octoprint Settings (wrench)->Octolapse->Download Settings) and I'll run it through the debugger and can determine what's happening :)

SystemDisc, 2018-09-11:
Yes, I know you added the H.264 codec, but I'm assuming (based on the note "When selected Octolapse will attempt to move the most recently completed timelapses into the default Timelapse plugin folder. Currently this only works with MP4 format. After synchronization is complete you can access your timelapses in the default Timelapse tab.") that you're not using the mp4 container with the H.264 codec, but I think you should.

Regarding errors with avconv, perhaps you should add a way for people to create a custom command for converting with placeholder variables - that way they can use avconv, or any other software, any way they want.

The next print I do, I'll come here first to upload the gcode and post my settings. What's the best logging option for you?

FormerLurker, 2018-09-11:
@SystemDisc, h264 DOES use the mp4 container and will be synchronized with the default timelapse plugin within OcotPrint.  Sorry if that note was confusing.

I'd like to mention the fact that Gina added a hook to OctoPrint to help me with this synchronization issue (only mp4 files are displayed in the default timelapse plugin).  I've already implemented the hook (octoprint.timelapse.extensions) and as soon as it's pushed into OctoPrint, Octolapse will start synchronizing all of the files it can create.  I will probably add some conditional code to hide that message about MP4 and Sync in the rendering profile based on the OctoPrint version as soon as I know what version will include the new hook.  Here is my code for the hook in case you are interested:

```
    def get_timelapse_extensions(*args, **kwargs):
        return ["mpg", "mpeg", "mp4", "m4v", "mkv", "gif", "avi", "flv", "vob"]
```

Regarding your next print, uploading your settings and gcode will be good enough for starters.  To save you trouble just leave the logging set to 'No Logging' for now.  If I run the gcode/settings through the debugger and can't find the issue I will probably ask you to turn on logging then.  Usually it's pretty easy to debug these 'no snapshots were taking' issues.  Hopefully I'll find a good way to prevent them entirely in the future.  

Here is some background on start gcode issues if you are interested:  Usually (but not always) the problem is a missing G90/G91 or M82/M83, which I could fix by setting the default profile XYZ and E axis modes to 'absolute', but I'm still not convinced that this is a safe thing to do.  I'm considering this, however, since this is such a common problem, and every printer I know of defaults to absolute mode when it is reset.  The problem comes when a G91 is sent for some reason, but the printer is not reset before the next print.  Obviously the gcode itself will mess up in this case, but I don't want Octolapse doing the same thing.  I've considered sending a G90/G91 or M82/M83 right before the print starts to somewhat mitigate issues, but that could mess up start gcode maybe?  Thoughts?

SystemDisc, 2018-09-12:
For reference, this is very confusing if indeed the H.264 uses the MP4 container: [image]

SystemDisc, 2018-09-12:
It's also worth noting that using mm/sec in printer settings now causes the values to change on save. The Z-Hop value is also changing (I put in 0.2). [Here's a screenshot](https://i.imgur.com/cA2VXwl.png)

SystemDisc, 2018-09-12:
Here're my settings, a gcode script where Octolapse was never triggered (keep in mind that I use a BLTouch probe and that my printer is set-up to set the X coordinate to -48 and Y coordinat to 2 after it finishes probing - because the BLTouch is offset 48mm to the right and 2mm in front of the nozzle), and a bunch of screenshots of how the settings show up for me.

[Archive.zip](https://github.com/FormerLurker/Octolapse/files/2373159/Archive.zip)

FormerLurker, 2018-09-12:
@SystemDisc, I ran your code through the debugger and found an issue where leaving some of the 'optional' speeds blank caused an error in the 'other slicer' printer profile (possibly others).  I need to do a bit more testing still, but I'll keep you updated.

FormerLurker, 2018-09-12:
OK, I've figured out what I think is the last issue that will prevent you from being able to record a snapshot.  There was a missing property in the default snapshot profiles (is_default) due to a mistake I made when updating the default settings.  It evaded me because the GUIDs on the defaults where the same as the ones that were in my personal settings.json file.  I have since removed this optional property on the default settings and use a different (sane) method to select the default which prevents me from having to add a property (is_default) manually when I alter the defaults.

Unfortunately your settings are still broken since they were migrated with the existing bug, but the fix is easy:

1.  Install this update from the plugin manager via the ... from URL box:  https://github.com/FormerLurker/Octolapse/archive/devel.zip
2.  Once Octoprint comes back up, make sure to select a different snapshot profile manually (any old profile will do), then select the one you want to use.  That will fix the broken (missing) guid issue.

Well, you found 4 bugs, congratulations!  I'll be working on Friday to fix the rounding issue you noticed in the 'Other Slicer' type within the printer profile.  I've been thinking a lot about it and I believe I have come up with a way to both fix the issue and to simplify the rounding madness in the process.  Win-Win!

Out of curiosity, what slicer are you using?  I added profiles for all of the ones I'm aware of (slic3r pe, cura, simplify 3d), but I'm sure there are others.  I'm just wondering why you used the 'Other Slicer' setting in the printer profile.  Don't get me wrong, I'm very glad you did since we fixed several issues.  However, if there's a slicer out there that I'm not supporting, I'd like to add it in the next release.

Let me know if you have any additional issues.  You know I'm here to help!

tixenre, 2018-09-12:
Can I request two things?
First, an option to save snapshots on the DSRL camera.
Second, an option to erase all snapshot from failures or not manually 

I'm asking this because I have config a DSRL camera, and I have some failures and forgot to enable "Clean Snapshots After Rendering Failure" and now I have no space left on the octoprint sd card and I don't know how to fix it.
Thanks

FormerLurker, 2018-09-12:
@tixenre, I've been planning to add browsing capability for a while, but it's gotten buried somewhat.  I think I'll enable 'clean after rendering failure' by default though, and that can be done quickly.

However, I can help you clear out the images for the time being.  Are you familiar with Linux?   If not the easiest way is via an SFTP (Filezilla anyone?).  Then you can browse to the following directory (I"m assuming you are running OctoPi):

/home/pi/.octoprint/data/octolapse/snapshots

Delete all of the sub directories in there (they will have strange names - Guids).  That should fix your problem.

If you would rather SSH you can do the following after login (again, if you are using OctoPi)
```
cd /home/pi/.octoprint/data/octolapse/snapshots
```
MAKE SURE YOU ARE IN THAT DIRECTORY by typing this command:
```
pwd
```
if pwd reports this directory: /home/pi/.octoprint/data/octolapse/snapshots
you are good.  You don't want to recursively delete the wrong sub directories!

now enter this:
```
rm -R -- */
```
That should clear out all of the subdirectories and files within them.  Feel free to run
```
ls
```
And you should only see a handfull of .jpg files.

Let me know if that helped!

tixenre, 2018-09-13:
Yes! It did work, I did the SSH way.
Thank you very much.
 There is an option to save snapshots on the DSLR camera? It would avoid all this trouble plus let me make my own timelapse and use the pictures.
Thanks, @FormerLurker  for this awesome plugin and the fast answers.

SystemDisc, 2018-09-13:
@FormerLurker 
> Out of curiosity, what slicer are you using? I added profiles for all of the ones I'm aware of (slic3r pe, cura, simplify 3d), but I'm sure there are others. I'm just wondering why you used the [...]

I'm actually using Simplify3D. Where do I pick that as my slicer? What does that change?

FormerLurker, 2018-09-13:
Look at the very top of the printer profile.  There is a drop-down box for you to choose your slicer.  Once you pick your slicer the controls are customized to reflect the actual names and values that your slicer supports.  Try it out and you'll see what I mean :)

If you want to use the 'Layer Change - High Quality' snapshot profile it's crucial that pick the correct slicer and enter all of the values requested.  That profile will restrict Octolapse so that it doesn't take snapshots over exterior perimeters, bridges, small perimeters, etc.  Octolapse will complain if your speeds are not unique, since this is how it detects what type of feature is being printed.  Simplify (and all the slicers really) does some odd rounding and has some other quirks that determine the actual speed within the gcode file, and thus it is important for Octolapse to know not only what slicer you are using, but all of the critical speed settings.

Check out [this](https://www.youtube.com/watch?v=sDyg9lMqMG8&t=1s) video for a (somewhat long) explanation of the new slicer settings and the high quality trigger.

FormerLurker, 2018-09-13:
@tixenre, 

It IS possible to leave the images on the DSLR.  I've not written up any tutorials on the subject yet (coming soon), but I have used an alternative process to reduce the substantial delays introduced by using a DSLR that is similar to what you are wanting.  Here is the process:

1.  I use a custom 'Before Print Starts' script (found in the camera profile) to set the camera's target to internal SD (the default is capture to RAM).  You could also use this script to configure other settings on your camera if you so desired.
2.  I delete all images on the camera's SD card within the same 'Before Print Starts' script.  This is optional in your case, but is useful in step 4 below if you would like Octolapse to render your timelapse.
3.  I change the 'Snapshot Acquire Script' (also found in the camera profile if you have 'External Camera - Script' selected as the camera type) so that instead of capturing and downloading, it only triggers a camera capture and returns the moment the camera finished the photo, but before it copies the image to the SD card.  This REALLY saves time, especially for ultra high res images.
4.  After the print finishes a 'Before Render Script' script is triggered that copies all of the images from the camera to the proper folder and renames them so that avconf (or ffmpeg depending on what you are using) can render a video.  The images will still be on the camera's SD at this point in case you want to render a video manually.  Alternatively you could skip this step entirely if you ONLY want to manually render a video.

There are downsides to this, though.  You'll receive no 'latest snapshot' images since they won't be downloaded from the camera until the end of the process.  Text overlays (rendering options) won't show any information since no metadata is captured for the images, though I could fix this).  Transformations (flip horizontal, flip vertical, mirror, etc) won't work either, though I could fix that too eventually.

Let me know if you would like some help getting this working since these are also new features that need to be tested.  Otherwise eventually a how-to guide will eventually appear within the [Octolapse Wiki](https://github.com/FormerLurker/Octolapse/wiki) explaining this process in detail.

FormerLurker, 2018-09-15:
@SystemDisc,

Please see Issue #240, which fixes issue you reported involving 'Other Slicer' settings changing after save.  I fixed this by applying the rounding directly to the text box as it's being entered, and by completely separating the rounding itself from the 'speed tolerance', which is how it should have worked in the first place.  There is still some odd behavior when switching back and forth from mm/min and mm/sec, but I'm not sure if I will fix that or not.

I also switched the other slicer specific settings (cura, simplify 3d, slic3r) to use the 'round in place' style extenders since I liked the way it worked so much!

FormerLurker, 2018-09-15:
I've pushed a new version, [v0.3.4rc1.dev3](https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev3), which contains all of the fixes made in this and other closed threads since the last release.  

You can find a new feedback thread for this release [here](https://github.com/FormerLurker/Octolapse/issues/242).

FYI, I still have not fixed whatever is breaking Octolapse in Safari because I've not been able to acquire an apple device to test.  I'm working on it though, and hope to have a fix asap.

Thanks to everyone who participated in this thread!  Some pretty substantial issues were detected and fixed thanks to you!  I think we're getting close to a maintenance release candidate!

Closed!

closed 2018-09-15 by FormerLurker