FormerLurker/Octolapse #233: X: None Y: None Z: None errors

blind-coder, 2018-09-10:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES - LIVE Print

#### What were you doing when the problem occurred

1. Start a print sliced by Cura 2.7.0
2. Have OctoLapse enabled
3. G28/G29 is done OUTSIDE the print as G29 would collide with my camera - need to move it in between.

#### What should have happened?
Timelapse should have been recorded.

#### What happened instead?
Tons and tons and tons of error messages like in https://github.com/FormerLurker/Octolapse/issues/144 - GCode contains G90 at the start and absolute coordinates in G1 / G0 commands.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version:      OctoPrint 1.3.9 running on OctoPi 0.13.0 

#### Printer model & used firmware incl. version

Printer Model: Anet A8
Printer Firmware Version: Marlin

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://oc.anderdonau.de/index.php/s/af0MhleBaQmXgBr

#### Link to settings.json

Link to settings.json with all passwords removed: Can't find export option....

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://oc.anderdonau.de/index.php/s/MEpMHzvB3ncuy9A

#### Link to octoprint.log

Link to octoprint.log:  https://oc.anderdonau.de/index.php/s/NjlK7XrwaUfCga5

FormerLurker, 2018-09-10:
I'm guessing you are running the latest stable release (0.3.1), correct?  I've corrected the 'position error bomb' you are probably experiencing in a newer version (working on release now if I can get some additional testing).  

Unfortunately I still haven't dealt with the issue of having no home command in the startup script.  I plan to add a dummy gcode to accomplish this, but haven't gotten around to it yet.  You could try the following (if your printer supports the M226 command, that is):

1. Put a G28/G29 in your start gcode at the appropriate place.
2. Add an M226 after the G28/G29
3. Start your print
4. Reposition your camera after the printer pauses due to M226 (Octoprint includes this command in the list of 'long running commands' so it shouldn't timeout while you are moving your camera around).
5. Unpause the print from your printer (not Octoprint) and it should continue, and hopefully Octolapse will now work as expected.

If you would be willing to help me out a bit, perhaps you would consider upgrading to the most recent pre-release of Octolapse?  There are some great new features in there, and lots of bug fixes.  You will lose any customizations you've made to your snapshot profiles, but they will be replaced with new and better defaults.  Also, you'll have to edit your printer profile, select your slicer (another new feature) and copy the requested settings over.  Here is a link to the latest rc/devel release:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev2.zip

[Here is a brand new tutorial video](https://www.youtube.com/watch?v=sDyg9lMqMG8) that explains the basics of how to configure the new release.  You can leave feedback or report issues relating to this release [here](https://github.com/FormerLurker/Octolapse/issues/232).

Thanks for posting!  Let me know if you need any additional assistance.

blind-coder, 2018-09-10:
I'll be happy to try the newer version. How do I go about installing it? I used OctoPrint's Plugin Manager to install it and it installed v0.3.1.

FormerLurker, 2018-09-10:
Excellent, thank you!  

You can install via the plugin manager by using the '... From URL' box.  Simply paste in this URL:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev2.zip

Be sure to check out the video I linked to above.  It should make things a bit easier :)

blind-coder, 2018-09-11:
I just finished a print with v0.3.4rc1.dev2 and the results can be seen here: https://youtu.be/1BoTlYzUIXc

What I did was add the following GCode startcode:
```
M190 S60
M104 S210
M109 S210
G90
M82
G28
M0 Remove cam
G29
M0 Position cam
G1 F3000 X110 Y110
G1 Z10
G92 E0
G91
G1 F200 E6.5
G1 F200 E3
G90
G92 E0

```
With that, the result came out fine.
One thing I noticed, though, was that the first snapshot was taken halfway through the skirt being printed, not sure where that originated from, though.

FormerLurker, 2018-09-11:
OK, I don't know much about the A8, but your explanation makes sense, thanks!

I don't see anything obvious in the gcode, so I'll need your settings.json file to do an accurate debug.  You can export it by going into the OctoPrint settings (wrench icon), clicking on Octolapse, then clicking on 'Download Settings'.  Once I have it I'll run it through the debugger.  Maybe we caught a bug, which is great!

Thanks again!

blind-coder, 2018-09-11:
Here you go: https://oc.anderdonau.de/index.php/s/9hF7ae8c68eH5FR

FormerLurker, 2018-09-11:
Well, there is some issue with G92 when using absolute extrusion for sure.  I haven't pinned it down 100%, but essentially Octolapse thinks there is a retraction when there isn't.

The other issue I see is for some reason your snapshot profiles did not update during the settings migration!  That is not good.  I would recommend restoring the defaults for Octolapse (Octoprint Settings->Octolapse->Restore all default settings.  Unfortunately you'll have to re-configure your printer profile... Sorry about that.

I'll look into both of these issues and will get back with you when I have figured it out.  Thanks again for your help!

blind-coder, 2018-09-11:
I didn't really do an upgrade. I uninstalled the v0.3.1 version of the plugin and then installed the v0.3.4rc1.dev2 version. I wasn't sure how OctoPrint would handle that and didn't want to end up with two intsallations of the same plugin.

Printer profile reconfiguration is not that hard, don't worry. Will try that tomorrow.

FormerLurker, 2018-09-11:
I've never tried uninstalling and reinstalling the builds that include the settings migration before!  Here's what I think happened:  The settings.json file was not removed during uninstall (not sure why that is, but I've never looked into it and don't control the uninstall procedure).  When you installed the new version, the migration routine did not run since it was a 'new' install.  However, a settings file already existed, so it was detected and used.  I need to look into that and figure out a solution.  Thanks for helping me get to the bottom of that.

I figured out what was going on with the 'snapshot is taken halfway through the skirt' issue, but I'm not sure what to do about it.  I do have a question about your A8 that you might be able to answer.  In this gcode, it looks like you are doing a prime and retract

```
G91
G1 F200 E6.5
G1 F200 E3
G90
```

However, I'm confused about the G91/G90.  Does G90/G91 change the E axis mode on the Anet A8?  If so that means that these codes would push 9.5 MM of filament through the nozzle and could be re-written like so:
```
G91
G1 F200 E9.5
G90
```

If G90/G91 does NOT influence the extruder on the Anet A8, that means your E axis remains in absolute mode and is first pushing 6.5mm of filament through the exturder and then retracting by 3.5mm.  If this is true (and I suspect it is) that means that Octolapse would detect a retraction of 3.5mm, and (depending on your retraction settings) would not take a snapshot until 3.5mm is extruded, which should be after about 25% of your skirt is printed.

This is my dilemma (I'm assuming that all of the extrusion is in absolute, not relative, but I could be wrong):  

Given this gcode:
```
G91; We can ignore this line if G91 does not influence the extruder
G1 F200 E6.5
G1 F200 E3
G90 ; we can ignore this line if G91 does not influence the extruder
G92 E0
```
when the G92 E0 command is executed, should Octolapse 'forget' about the previous retraction?  Octolapse seems to correctly be adjusting the internal E axis offset, but is maintaining it's 'retraction length' of -3mm after the G92, which is actually true, though perhaps somewhat confusing.

One final issue I noticed is that in your snapshot profile, extruder trigger requirements are disabled, which means Octolapse should ignore the extruder state I mentioned above, but it doesn't.  I suspect this is a bug with the enable/disable check box since that is brand new.  That should be an easy fix.  What great progress!

FormerLurker, 2018-09-11:
Well, I've verified that the extruder trigger requirements are actually working as expected and that the problem is instead related to how Octolapse detects layer changes.  It does this by looking for extrusion at a height greater than the previous highest extrusion.  In your case it's not detecting extrusion until 3.5mm has been pushed out due to the retraction gcodes I mentioned earlier.  The initial prime is not being counted because it occurred above the priming height (at z=10mm).  I'll think about this and will determine if any changes need to be made.

blind-coder, 2018-09-12:
I'm not sure where that double priming came from, I made a lot of tests with dual and triple extruders in the recent past, so it might be a leftover from that.
To make sure what happen, I ran the GCode manually. Here's the output after running G28:

```
Send: M114
Recv: X:125.00 Y:180.00 Z:100.00 E:0.00 Count X:12500 Y:18000 Z:40000
Recv: ok
Send: G90
Recv: ok
Send: M82
Recv: ok
Send: G1 F3000 X110 Y110
Recv: ok
Send: G92 E0
Recv: X:110.00 Y:110.00 Z:100.00 E:0.00 Count X:11000 Y:11000 Z:40000
Recv: ok
Send: G91
Recv: ok
Send: G1 F200 E6.5
Recv: ok
Send: M114
Recv: X:110.00 Y:110.00 Z:100.00 E:6.50 Count X:11000 Y:11000 Z:40000
Recv: ok
Send: G1 F200 E3
Recv: ok
Send: M114
Recv: X:110.00 Y:110.00 Z:100.00 E:9.50 Count X:11000 Y:11000 Z:40000
Recv: ok
Send: G90
Recv: ok
Send: G92 E0
Recv: X:110.00 Y:110.00 Z:100.00 E:0.00 Count X:11000 Y:11000 Z:40000
Recv: ok
```
So one thing to note is that G91 puts all motors (X/Y/Z/E) into relative mode (see also http://marlinfw.org/docs/gcode/G091.html). That means that after G1 F200 E3 there is 9.5 units of filament extruded.

Personally, I'd say it's safe to forget the retraction at G92 E0. There's really just two scenarios where that happens:
- priming at the start of a print
- priming at extruder switch

In both cases, the extruded length is set to 0 to not interfere with the actual print.

blind-coder, 2018-09-12:
I have reset the Octolapse config and configured my slicer again. Here's the new settings.json: https://oc.anderdonau.de/index.php/s/lHxclKUutgkkJOx

And here's a result print: https://youtu.be/GDV_FeOoqEM

FormerLurker, 2018-09-12:
@blind-coder, that terminal output was VERY helpful.  Thanks for that.  I wouldn't have thought to ask you to do that, so I'll have to add it to my list of debugging steps!

So, G91 DOES influence your extruder (as I sort of suspected above).  I just adjusted the default A8 profile to match this!  I only learn what ppl tell me about other printers, so thanks!.  I wish I had one of each to test on :)

Regarding the G92 and removing any existing retraction length (stored by octolapse to keep tabs on how much retraction currently exists), If Octolapse decides it's time to take a snapshot RIGHT AFTER a g92, would you expect it to retract before moving?  If that happened your already retracted filament would be retracted even more, which I think could cause a jam (at least it would on my mmu, but it's a known constant jammer, and not in a good way).

Maybe I could switch the layer tracking to detect a layer change if the E axis advances at all on a given layer (even if it is retracted a bit) as a compromise?  I suspect most EndGcode scripts would trigger snapshots in that case, but will need to try it to see.  I guess I could add a gcode that would end the timelapse that could be put in the end gcode script that would take care of that though.  Or, maybe I use the method above detect just the start of the first layer?  I'll keep pondering.

Also, It's a little hard to see what's going on in that video.  Maybe take a look at this link and see if your camera quality can be improved:  https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-image-preferences

Pay special attention to the parts about editing Octopi.txt, especially the security hazards involved if you are using port forwarding!  Also, not all cameras support all settings.

Lastly, when you ran the latest print, did you re-slice the file to match your new slicer settings, or did you not change the slicer settings?  I'm wondering if I should re-run your code to see what's going on, or if things are basically solved (except for the issue where Octolapse takes a pic partway through the skirt)?

FormerLurker, 2018-09-12:
@blind-coder, I took a look at your settings and noticed a few things that you could do to improve your quality.

1.  Select the slicer you are using from the drop down box within the printer profiles.
2.  Copy your slicer settings in exactly (all of them if you can).
3.  Did you watch the video I linked to above?  It explains how to adjust both your slicer speeds and those in Octolapse so that they are all unique.  This will come into play on the next step.
4.  Switch to the 'Layer Change - High Quality' trigger.  If your speeds are unique that should really boost your print quality.

blind-coder, 2018-09-13:
Don't bother with the video quality, I have a very cheap 10 EUR USB camera on there which never produced good pictures, ever. I might give the C920 I have here a shot, but I'll have to print an M8 mount adapter first...

I'll try the unique speeds later today, gotta have to reslice everything then...

About the G92 E0, I really don't expect a picture to be taken there at all. Usually, there will have been a retraction before that, and a priming afterwards, possibly even a prime tower printed. Think of it as accompanying an M600 GCode if you do a manual filament change for multicolor prints.

FormerLurker, 2018-09-13:
Let me know how the high quality snapshot profile works out for you!  I'm really excited about that feature.

regarding the G92, it wouldn't take a pic there.  If I reset the retraction detected by Octolapse, it would try to take a pic at the beginning of your skirt instead of in the middle.  If you watch, Octolapse SHOULD (perhaps it doesn't) snap a pic right when filament start to come out while printing the skirt, and should then retract and lift (if those settings are enabled).

If you use M600, Octolapse won't detect any retraction there at all and will assume your extruder is in the exact same state after the M600 as before.

Also, you mention a prime tower.  'Feature Detection' can restrict snapshots to your prime tower.  I think this is a simplify only feature though, but it sounds like that's what you are using :)

FormerLurker, 2019-10-12:
So, this has been opened for a while, and a LOT of enhancements have been made to Octolapse in the meanwhile, so I'm going to go ahead and close this issue.  Please let me know if you want it reopened, and thank you for your post!!!

closed 2019-10-12 by FormerLurker