FormerLurker/Octolapse #234: Problems installing Pillow (PIL)

FormerLurker, 2018-09-10:
Some users are having trouble installing Octolapse because a required library, Pillow, fails to install.  If you experience this issue, or if you have found a working solution, please post it here.  Thanks!

SpectrumFRC3847, 2018-09-14:
I've dad this issues a couple times. I've tried a few different things to get pillow installed the latest caused a boot loop on my pi, so I'm reflashing now.

SpectrumFRC3847, 2018-09-14:
I haven't been able to fix it.

I just tried again on the new flash, here is the install log from the package manager.

https://gist.github.com/SpectrumFRC3847/a19e19944ba74279ba53ff59091355fc

FormerLurker, 2018-09-14:
Have you tried upgrading pip?
```
pi@octopi:~ $ source oprint/bin/activate
(oprint)pi@octopi:~ $ pip install --upgrade pip
```

FormerLurker, 2018-09-14:
Sorry, for some reason like half of your message was cut off (using a GitHub app and it's got a few bugs I think).  I really did read everything I got, lol..  ignore my previous message since you obviously tried that.  Let me research some and get back with you tomorrow

FormerLurker, 2018-09-14:
Actually...  When you updated pip did you activate the venv first like so:

```source oprint/bin/activate```

SpectrumFRC3847, 2018-09-14:
No I don't think I did.

FormerLurker, 2018-09-14:
Ok, try that right after you ssh in, upgrade pip, and try installing octolapse again.  Send me your log if it fails again.

If it works I'll submit a ticket for octopi.  Maybe they can upgrade pip and use that in the next release.

SpectrumFRC3847, 2018-09-14:
Hey I think that did it. 

`Successfully installed Octolapse-0.3.1 Pillow-5.2.0
Done!`

FormerLurker, 2018-09-14:
Cool!  Fyi, a new release will drop soon.  Nearly ready for release candidate status.  Let me know if you want to try it out.  I'm looking for more testers!

Bmarquez1997, 2018-09-25:
> @Bmarquez1997 <https://github.com/Bmarquez1997>, when you say running
> setup.py caused the pi to crash, do you mean within the plugin manager, or
> did you try to manually install Octolapse?
>
> Also, when you upgraded pip did you activate the virtual environment first
> like so (from the /home/pi directory):
> source oprint/bin/activate
>
> then
>
> pip install --upgrade pip
>
> and finally try to install octolapse from the plugin manager.
>
> Let me know one way or another and we can continue debugging!

When I upgraded pip I did it by first activating the virtual environment
using "source oprint/bin/activate".  When I run the octolapse install
through the plugin manager it stops at the line that says it is running
setup.py for Pillow, but I can just click continue/cancel on that message
and continue using octolapse.  When I try to manually install Pillow
(running "pip install Pillow" in the virtual environment) that's when the
pi will completely crash/shut down when it hits the step for running
setup.py

Bmarquez1997, 2018-09-26:
The first thing I normally do when first starting a clean linux environment is run an update and upgrade, just to make sure I'm not missing anything.  Whenever I run the manual install the most I can get before the pi restarts is shown in "Manual_Pillow_Install_Error.png".  When trying to install from the Plugin Manager in octoprint I get the error message shown in "Octopi_Octolapse_Install_Error.png".  I attempted to download octoprint.log, but for some reason it would not let me.  The closest I could get for you is a screenshot of the contents of the file.  You can see those in the files octoprint.log_part(1-4).  If there's anything else I can do to prove more information please let me know!

Thanks!
[image]
[image]
[image]
[image]
[image]
[image]

FormerLurker, 2018-09-26:
Ok, I'm going to try to recreate your install here on a spare pi.  Let me know if these are the exact steps you took:

1.  Flash the latest OctoPi
2.  Update the environment (update + dist-upgrade)
3.  Upgrade pip
4.  Attempt to install pillow (this should fail)

If I can figure out a way to reliable recreate this issue I think we'll be that much closer to getting this solved.  FYI, here's what I did last time I installed a new instance:

1. Flash the latest OctoPi
2. Upgrade Pip
3. Install Octolapse via the plugin manager
4. Update the environment (update + dist-upgrade)

If you want you could see if those steps work for you.

Also, just to verify, what pi are you using?  What version of OctoPi did you flash?  Did you update Octoprint after install?

Thanks!

Bmarquez1997, 2018-09-26:
I am using a raspberry pi 3, I believe the version I flashed was 0.15.1
(from the octoprint website), and after running the
update/upgrade/dist-upgrade I updated octoprint from the settings menu in
octoprint.  Once my current print finishes I'll try re-flashing and follow
the steps you mentioned.  Is there a place that I can get the newest octopi
img, since it seems the one on the octoprint website is a couple versions
behind?

FormerLurker, 2018-09-26:
@Bmarquez1997, it looks like 0.15.1 is the latest release version:  https://github.com/guysoft/OctoPi/releases

Also, I have a spare pi 3 to work with here, so that's good!

Bmarquez1997, 2018-09-26:
So if 0.15.1 is the current release version, then what are the additional
updates that octoprint has to install?

FormerLurker, 2018-09-26:
Octopi and Octoprint are separate entities.  There have apparently been a few updates to Octoprint since OctoPi 0.15.1 was released.

Bmarquez1997, 2018-09-26:
Ah, okay.  That makes much more sense.  Thanks!  I'll try the steps you
mentioned once i get home, and let you know what it does

Bmarquez1997, 2018-10-05:
Hi!  Sorry about the long gap between messages.  I followed the new steps you mentioned above, and I got the same error message.  Is there something I'm missing/forgetting to install?

DeclanCSL, 2018-12-05:
@FormerLurker i need your advice to guide me ..
i did what you told us but still cant
1) source oprint/bin/activate
2) pip install --upgrade pip
[image]
[image]
[image]

FormerLurker, 2018-12-05:
@Bmarquez1997 - I'm SO SORRY about the delay in responding!  This got lost in the blur that is year end :)

I see this error pop up from time to time, but have never been able to figure it out.  I have to say I've never seen it from a fresh install, though sometimes pip needs to be updated for whatever reason.  It's so irritating I'm thinking about switching to a different library altogether.

@Bmarquez1997, I know you've tried some of this, but try the second code block below.

@DeclanCSL, try installing PIP manually via:
```
source oprint/bin/activate
pip install pillow
```
If that doesn't work, try this:
```
source oprint/bin/activate
pip install pillow --no-cache-dir
```
Please save the console output and post here.  Also, since I'm having trouble reproducing this issue, perhaps one of you can post an issue [here](https://github.com/python-pillow/Pillow/issues/new) to the pillow github page?  If not I'll create the issue.  It's time to get this solved.

Also, I'll be able to respond more quickly after the new year.  This is my busy time, unfortunately.  Thanks for being patient!

FormerLurker, 2018-12-05:
One last thing to try.  You could be missing dependencies (though I can't figure out how since we're all using the same image).  Try running this command outside (not within) the virtual environment:

```
sudo apt-get install libjpeg-dev zlib1g-dev
```

Then retry the pillow install via
```
source oprint/bin/activate
pip install pillow --no-cache-dir
```

Bmarquez1997, 2018-12-06:
> @FormerLurker No worries!  I got caught up in my classes this semester so
> I completely forgot about this being an issue.  When I ran the command to
> install those two dependencies, it looked as if it had files to
> install/download, so I'm guessing that they either weren't there or needed
> updated.  I ran the pillow install (with the output being piped to a file),
> and my pi crashed when it said "running setup.py".  I checked the output
> file and it was empty, so I possibly didn't have it writing to the file
> correctly.  I'm going to do another clean install and try running those
> commands again, to see if it was something I changed in this install (that
> has been running since probably october).  Random question, but the
> software I use to write the image to the sd card wouldn't have any affect
> on something like this, would it?
>
> Thanks,
> -Bmarquez1997

Below is the output from trying to run the pillow install manually, and
then the output from running the octolapse install through the OctoPi UI.
I've also attached the octoprint.log referenced in the octolapse install,
in case that helps.  Like you mentioned above, I can open a ticket on the
Pillow page, since this almost seems like a Pillow issue and less of an
octolapse specific one.

[image: image.png]

Installing plugin "Octolapse" from
https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip.../home/pi/oprint/bin/python2
-m pip install https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
--no-cache-dirLooking in indexes: https://pypi.org/simple,
https://www.piwheels.org/simpleCollecting
https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zipDownloading
https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zipCollecting
pillow (from Octolapse==0.3.4)Downloading
https://files.pythonhosted.org/packages/1b/e1/1118d60e9946e4e77872b69c58bc2f28448ec02c99a2ce456cd1a272c5fd/Pillow-5.3.0.tar.gz
(15.6MB)Requirement already satisfied: sarge in
/home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4)
(0.1.4)Requirement already satisfied: six in
/home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4)
(1.11.0)Requirement already satisfied: OctoPrint>1.3.8 in
/home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4)
(1.3.9)Requirement already satisfied: psutil in
/home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4)
(5.4.8)Requirement already satisfied: flask<0.11,>=0.10.1 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.1)Requirement already
satisfied: Jinja2<2.9,>=2.8.1 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (2.8.1)Requirement already
satisfied: werkzeug<0.9,>=0.8.3 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)Requirement already
satisfied: tornado==4.5.3 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (4.5.3)Requirement already
satisfied: PyYAML<3.11,>=3.10 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (3.10)Requirement already
satisfied: Flask-Login<0.3,>=0.2.11 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.2.11)Requirement already
satisfied: Flask-Principal<0.4,>=0.3.5 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.3.5)Requirement already
satisfied: Flask-Babel<0.10,>=0.9 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.9)Requirement already satisfied:
Flask-Assets<0.11,>=0.10 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.10)Requirement already
satisfied: markdown<2.7,>=2.6.9 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.11)Requirement already
satisfied: pyserial<3.5,>=3.4 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (3.4)Requirement already satisfied:
netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages
(from OctoPrint>1.3.8->Octolapse==0.3.4) (0.7.19)Requirement already
satisfied: watchdog<0.9,>=0.8.3 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)Requirement already
satisfied: netifaces<0.11,>=0.10.6 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.7)Requirement already
satisfied: pylru<1.1,>=1.0.9 in
/home/pi/oprint/lib/python2.7/site-packages (from
OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.9)Requirement already
[... 85 lines cut ...]
rsa<3.3,>=3.2.3->OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.2)Requirement
already satisfied: urllib3<1.23,>=1.21.1 in
/home/pi/oprint/lib/python2.7/site-packages (from
requests<3,>=2.18.4->OctoPrint>1.3.8->Octolapse==0.3.4)
(1.22)Requirement already satisfied: idna<2.7,>=2.5 in
/home/pi/oprint/lib/python2.7/site-packages (from
requests<3,>=2.18.4->OctoPrint>1.3.8->Octolapse==0.3.4)
(2.6)Requirement already satisfied: chardet<3.1.0,>=3.0.2 in
/home/pi/oprint/lib/python2.7/site-packages (from
requests<3,>=2.18.4->OctoPrint>1.3.8->Octolapse==0.3.4)
(3.0.4)Requirement already satisfied: Unidecode<0.05,>=0.04.14 in
/home/pi/oprint/lib/python2.7/site-packages (from
awesome-slugify<1.7,>=1.6.5->OctoPrint>1.3.8->Octolapse==0.3.4)
(0.4.21)Requirement already satisfied: regex in
/home/pi/oprint/lib/python2.7/site-packages (from
awesome-slugify<1.7,>=1.6.5->OctoPrint>1.3.8->Octolapse==0.3.4)
(2018.2.21)Requirement already satisfied: pytz>=0a in
/home/pi/oprint/lib/python2.7/site-packages (from
Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint>1.3.8->Octolapse==0.3.4)
(2018.4)Installing collected packages: pillow, OctolapseRunning
setup.py install for pillow: startedError!Could not install plugin,
unknown error, please consult octoprint.log for details

I was able to install a couple other plugins (ipOnConnect, LED Strip
Control, and Filament Sensor), but no luck on octolapse...

Bmarquez1997, 2018-12-06:
Apparently I can't attach things through email... Here is the file mentioned above
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/2651214/octoprint.log)

FormerLurker, 2018-12-06:
@DeclanCSL, can you post the console output for the following command:
```
pip install pillow --no-cache-dir
```
Also, did you run this command:
```
sudo apt-get install libjpeg-dev zlib1g-dev
```
if so, did it complete?  Thanks!

DeclanCSL, 2018-12-06:
@FormerLurker ..
pip install pillow --no-cache-dir
[image]
nothing moving

DeclanCSL, 2018-12-06:
[octoprint (2).log](https://github.com/FormerLurker/Octolapse/files/2653474/octoprint.2.log)

FormerLurker, 2018-12-06:
@Bmarquez1997 and @DeclanCSL, can you post plugin_pluginmanager_console.log?  That is the file that contains information regarding plugin installs.

@DeclanCSL, maybe you can try the pillow install again?  Hanging like that on download could be some server issue that has since been resolved.  I've personally never seen that before.  Also, I'm assuming the `sudo apt-get install libjpeg-dev zlib1g-dev` command completed with success?

DeclanCSL, 2018-12-07:
@FormerLurker 
yup done sudo apt-get install libjpeg-dev zlib1g-dev
[image]
still stuck pillow install , wont move
[image]
[image]
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/Octolapse/files/2655440/plugin_pluginmanager_console.log)

Bmarquez1997, 2018-12-07:
Here's my pluginManager log
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/Octolapse/files/2655449/plugin_pluginmanager_console.log)

The only thing I can really see is that it goes to download octolapse, and then pillow, and then that's the end of the file.

DeclanCSL, 2018-12-09:
@FormerLurker Pls help me Sir..
im excited to used octolapse..
haha

FormerLurker, 2018-12-10:
@DeclanCSL, 

The command above should be
```
sudo apt-get install libjpeg-dev zlib1g-dev
```
not
```
sudo apt-get install libjpeg-dev zliblg-dev
```
Notice that the difference is that zlib1g-dev has a ONE as the fifth character.  Try that again please.

DeclanCSL, 2018-12-10:
@FormerLurker 
hi sir thank for your reply
[image]
but still can't ...

DeclanCSL, 2018-12-17:
@FormerLurker i used another printer but also cant install octolapse..

FormerLurker, 2018-12-17:
@DeclanCSL, the printer shouldn't affect this at all.  Since there is no error message after 'running setup.py install for pillow ...', it will be very difficult to debug your issue :(

Can you try updating your OS via the following commands:
```
sudo apt-get update
sudo apt-get dist-upgrade
```

Then try to install pillow manually with the following commands and let me know what the output is (hopefully it installs or provides some kind of error message):
```
source oprint/bin/activate
pip install pillow
```

if that fails try to run pip with the --no-cache-dir option again
```
pip install pillow --no-cache-dir
```

fingers crossed!

DeclanCSL, 2018-12-17:
@FormerLurker 
install the pillow  ..the screen just blink 2 time and that it...
[plugin_pluginmanager_console (1).log](https://github.com/FormerLurker/Octolapse/files/2686696/plugin_pluginmanager_console.1.log)
[octoprint (3).log](https://github.com/FormerLurker/Octolapse/files/2686698/octoprint.3.log)
please help to look

FormerLurker, 2018-12-17:
@DeclanCSL, I think we need to bring in the big guns now.  Can you post an issue [here](https://github.com/python-pillow/Pillow/issues/new)?  Be sure to include your OS and link to this issue if you can.  I'll chime in when possible.

DeclanCSL, 2018-12-18:
@FormerLurker it done now..
The problem is the current not enough..
after see the sign and i change to 5V 2.1A then okey ..
haha

Bmarquez1997, 2018-12-18:
@DeclanCSL so the problem is fixed for you by using a power source with
more current? Or am I misunderstanding your last message?

DeclanCSL, 2018-12-18:
@Bmarquez1997  the voltage 
on the octoprint corner will shown a sign of tunder logo means not enough voltage and current..
previous im using 5v 1A..then i change to 2A then okey edi ..
i think when on pillow setup need enough V/A to setup

Bmarquez1997, 2018-12-18:
@DeclanCSL I don't see a thunder image on the web interface, do you mean on
a physical display connected to the raspberry pi?  I'll give it a shot
later today once I can check my power supply's rating

Bmarquez1997, 2018-12-18:
> @Bmarquez1997 <https://github.com/Bmarquez1997>, I believe that
> 'thunderbolt' image is only available on Octoprint 1.3.10, so make sure
> that's installed. Since your pi is crashing during the install, I'd say
> that sounds a lot like the issue @DeclanCSL <https://github.com/DeclanCSL>
> was seeing! Please let me know if you get this resolved. If so I'll update
> the installation instructions.

@FormerLurker ah, okay.  I'm still on 1.3.09, so that's probably why I
don't see it.  I went to update to 1.3.10, and it seems the install failed
and broke something else (that is now preventing the server from
starting).  I'm going to just flash a fresh image of octopi tonight, and
give that power supply a try and see if that fixes it.  Thanks!

Billd24, 2018-12-26:
Was having issues installing OctoLapse on 1.3.10, and can confirm that the power supply was the issue - very weird !!!

Bmarquez1997, 2019-01-14:
Sorry for the late response, I ran into some hardware issues with my
raspberry pi and had to fix those first.  Switching to a different power
supply indeed fixed the issue, and I was able to install octolapse
successfully!  Thank you so much for your help!!

FormerLurker, 2019-01-14:
@Bmarquez1997 and @Billd24,

Excellent!  I will leave the issue open for a bit in case others have issues.  I may add a wiki article on the subject since there are lots of ppl with underpowered PSUs

FormerLurker, 2019-02-04:
So glad @junglerhino37!  Underpowered PSUs are the root of many an issue, or so I've learned.  This was a tough one for me to track down, and I wouldn't have been able to find it without all of the feedback here.

yyh1002, 2019-04-13:
I had the same problem and finally had it installed. 
I'm running Octoprint on OrangePi Lite with Armbian. 

My error:
> 2019-04-13 02:28:59,452 ! Command "/home/octoprint/OctoPrint/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-SJYaRU/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-9Inw6D/install-record.txt --single-version-externally-managed --compile --install-headers /home/octoprint/OctoPrint/venv/include/site/python2.7/pillow" failed with error code 1 in /tmp/pip-install-SJYaRU/pillow/

Attemps I tried that didn't work:
> sudo apt-get install libtiff5-dev libjpeg62-turbo-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev libharfbuzz-dev libfribidi-dev tcl8.6-dev tk8.6-dev python-tk
> source ~/OctoPrint/venv/bin/activate
pip install --upgrade pip

The code itself worked but Octolapse still won't install afterwards.
I then tried directly install pip in terminal. Failed to install, same error as the plugin manager.

### The command that worked ([found in this link](https://github.com/python-pillow/Pillow/issues/793#issuecomment-48724868))
> source ~/OctoPrint/venv/bin/activate
pip install Pillow==2.5.0

It then told me pillow installed succsessfully. I went back to Octoprint Plugin Manager to install Octolapse and succeeded. 

Hope it's helpful.

Hydrosys4, 2019-08-12:
Had exactly same issue and I solved with a more powerful power supply, as weird as it seems I can confirm it.

FormerLurker, 2019-10-12:
I'm going to go ahead and close this issue since it's now a will known problem that has even been handled within Octoprint.  @Foosel added some features to prevent plugin installation if there are undervoltage or overheating errors, which should make it obvious what's happening (awesome feature, thank you!).  If anyone has any additional problems installing Pillow, please open a new issue and I'll take a look.

Thanks to everyone who has contributed here!

closed 2019-10-12 by FormerLurker