FormerLurker/Octolapse #236: Unable to Change, find or add printer profile

pr0xyfl00d3r, 2018-09-12:
#### Version of Octolapse
 
Octolapse Version: 0.3.4rc1.dev2

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Installing 0.3.4rc1.dev2 
2. when trying to access a current profile i get the following error 
3. In Octolapse Printer settings page clicking add profile does nothing. 

#### What should have happened?
adding printer profile

#### What happened instead?
unable to add printer profile

#### Link to settings.json
Link to settings.json with all passwords removed: https://pastebin.com/nHQBts1Y

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/pr0xyfl00d3r/354371a7ad30bda8e1814c836e5532ef

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]

[image]

FormerLurker, 2018-09-12:
Thanks a lot for posting!  I'll dig into this after work and will let you know what I find.

pr0xyfl00d3r, 2018-09-12:
That’s was quick 😊 I will try this shortly and get back to you

pr0xyfl00d3r, 2018-09-12:
i have tried installing from https://github.com/FormerLurker/Octolapse/archive/devel.zip but is still the same

FormerLurker, 2018-09-12:
Can you restart OctoPrint and then post your octoprint.log and plugin.octolapse.log (if it exists)?

pr0xyfl00d3r, 2018-09-12:
added the logs to this chat
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/2376146/octoprint.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2376147/plugin_octolapse.log)

FormerLurker, 2018-09-12:
No exceptions, but I do notice one interesting thing:  Octolapse is now reporting the newer version (previously it was showing 0.3.1).  Can you re-post your settings.json file?  I'd like to see if it was migrated, and if so try to tell why it still appears to be broken.  I think we're close.

pr0xyfl00d3r, 2018-09-12:
added the setting file zipped [settings.json.zip](https://github.com/FormerLurker/Octolapse/files/2376220/settings.json.zip)

FormerLurker, 2018-09-12:
Thank you for being so patient and for posting everything that's needed to debug this!  I know it's sort of a pain in the butt.

So, oddly enough, those settings worked just fine on my development instance.  Either this is a problem with Safari (I assume you are using a mac based on the files I've received) or it's some problem with my development instance (in that it works there but not in a live environment).  Are you using OctoPi?  Also, and I hate to ask, but did you try clearing your browser cache (Cmd + Shift + R I believe)?

If that doesn't work, maybe there are javascript errors?  Can you look at this link and see if you can extract any javascript errors from the console: https://codex.wordpress.org/Using_Your_Browser_to_Diagnose_JavaScript_Errors#Safari

The next step for me will be to try all of this on my Pi with the latest instance.  I'll try this ASAP after I hear back from you to make sure you are using OctoPi and that the Cmd Shift R didn't work, and to see what javascript errors you might have noticed in the console.

pr0xyfl00d3r, 2018-09-12:
i see these errors with safari [image]

i fired up firefox and it works ok

pr0xyfl00d3r, 2018-09-12:
and yes its octopi

FormerLurker, 2018-09-12:
I haven't done any debugging on Safari, sorry.  I need to figure out a good way of doing that short of actually buying a Mac (which I would like to do at some point if I have some $$ to burn).

I might create a new issue for the Safari things since it looks like the original problem was solved (migrations crash).  I'm pretty sure it used to work just fine in Safari, so it must be something I've done recently.

pr0xyfl00d3r, 2018-09-12:
0.3.1 functioned fine but the setting was a pain and i saw the updates you made which would make my life easier... but if you need help debugging on a mac i can help you with that, and i suppose for now i can use firefox or chrome

pr0xyfl00d3r, 2018-09-12:
you can put macos on a vm also

FormerLurker, 2018-09-12:
I phoned a friend who has a macbook, but it might be a week or so before I can get hold of it.  I'll let you know when I've had a chance to debug.

I did look into getting macos on a VM and there is a lot of contradictory information.  I also found a way to remote debug via an ipad or iphone, but I don't have one of those either :(

However, if you are familiar with javascript (and maybe knockout), I'd love to have some help debugging :)

FormerLurker, 2018-09-18:
@TheLion and @pr0xyfl00d3r,

It's possible that I found and fixed this issue.  Apparently the 'toFixed' function in safari only supports 20 digits.  That seems reasonable, except this error doesn't seem to happen on any other browsers, and I was unaware of the limitation.  I now detect and correct these issues before applying toFixed.

I also saw some issues in the rendering profile surrounding the watermark capability.  There was some javascript syntax that doesn't seem to be supported by safari (let and lambda), so I replaced them with a more traditional syntax.

I only had 30 minutes to test during an evaluation of BrowserStack, so there might be more that I missed.  It's way too expensive for me to purchase a license, but they have a program where you can get unlimited testing for open source projects if you put their logo on your github page.  I'll look into this so that I might be able to debug future issues.

Please test the corrections out by installing with this url within the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/125338f788950e50d2e00a4187b5d60b49f62a63.zip

Install that, reboot and let me know if the problem is resolved or not.  You MIGHT have to clear your cache after rebooting to get the new javascript working.  Please let me know if you need to clear your cache or not (try it first without clearing your cache).  I'm getting ready to submit an issue to the Octoprint github page regarding javascript refresh and I'm trying to verify that I'm not the only one having the issue (could be related to my dev instance??).

Thanks!

pr0xyfl00d3r, 2018-10-06:
hey, trying now

FormerLurker, 2018-10-06:
Awesome, thanks!  I'm working on Octolapse today, so it's a good time for me!

TheLion, 2018-10-06:
I thought I also replied to this, but apparently I haven't.

The changes you made work fine now on all my devices.

FormerLurker, 2018-10-06:
Thanks for responding @TheLion!  I'm glad it's working for you now.

FormerLurker, 2019-10-12:
OK, I think this bug is solved!!!  Closed!

closed 2019-10-12 by FormerLurker