FormerLurker/Octolapse #242: Feedback for release v0.3.4rc1.dev3

FormerLurker, 2018-09-15:
This issue is for any general feedback regarding the new rc/devel release of Octolapse.  Let me know if you have any bugs, issues, or if things are working just great!  [See the release notes here.](https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev3)

If you want to keep tracking the latest rc/devel branch to be notified right away when new releases are available and you are using Octolapse v0.3.4rc1.dev0 (a few versions earlier might work too) or above, go to Octoprint Settings(wrench)->Software Update->Click the settings wrench in the upper left corner.  Then switch the OctoPrint version tracking to 'Release' and the 'OctoPrint Release Channel' to 'Devel RCs'.   If you aren't notified right away, you can click 'Force Check for Update' under 'Advanced Settings' in the software update plugin. 

Note:  I've received reports of problems running Octolapse on Safari, specifically involving opening, adding, and editing printer profiles.  If you are able to test with Safari and send me any console error logs, I'd be appreciative, Thanks!

FormerLurker, 2018-09-17:
@SystemDisc, sorry to hear that!  

When I swapped out my settings with the settings you posted I noticed that there was still an error with the current snapshot guid, and it was the same error I saw before (current_snapshot_profile_guid does not exist).  Did you follow this step that I sent in the last thread:

>Once Octoprint comes back up, make sure to select a different snapshot profile manually (any old profile will do), then select the one you want to use. That will fix the broken (missing) guid issue.

I'm hoping you DIDN'T follow the above step, else something different is happening.  I've gotten one report of issues with Safari (which are very difficult for me to test), so hopefully there is no overlap.

Try changing the snapshot profile to 'layer change - compatibility', then immediately download your settings file and send it to me.  If you notice that it reverts to the 'Every 01:00 - Compatibility' profile at some point (the first item in the list) without you changing it, please let me know.

Thanks!

SystemDisc, 2018-09-17:
Oh, sorry. I didn't follow that step. I'm trying again.

SystemDisc, 2018-09-18:
When I switched snapshot profiles, I got an error about the GUID missing. The same error came up (with a different GUID) when switching back. After switching back and forth between some snapshot profiles a few times, the error eventually listed the GUID as null, and on the next switch immediately after that there was no longer an error message.

I'm running a print now and Octolapse *is* being triggered. I'll let you know what the results are after the print finishes.

ryanneufeld, 2018-10-07:
I'm trying the rc as well, and even though my gcode has a G90 comamnd I'm getting the position error.

Further to this, it seems like it starts recording many layers after the first.

[X-motor strain releif v8.gcode.gz](https://github.com/FormerLurker/Octolapse/files/2453556/X-motor.strain.releif.v8.gcode.gz)

[image]
[plugin_octolapse.log.gz](https://github.com/FormerLurker/Octolapse/files/2453554/plugin_octolapse.log.gz)

FormerLurker, 2018-10-08:
@ryanneufeld,

This is happening because there are movement commands (G1 in this case) before your XYZ axis mode is set:
```
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```
In my opinion this is not a safe operation.  

Fortunately it is quite easy to fix!  Just move the G90 to the very top of your start gcode.

Also, I'm not sure why that G92 command is there (G92 E0.0) since M83 is set above that.  There could be some reason.

Is this the default start gcode for the Mk3 Slic3r PE profile?

ryanneufeld, 2018-10-09:
Yeah, this is what prusa ships as the default start Gcode for the MK3

They also have G92 E0.0 on layer change. I'm assuming due to an old marlin
bug that isn't fixed in their fork of the firmware.

--Ryan Neufeld BSc CIS
ryan@neufeldmail.com

FormerLurker, 2018-10-09:
@ryanneufeld, thanks for your response.  Perhaps someone should post an issue on their github page regarding the those G1 commands.  If for some reason one sends a G91 and then starts a print without a power cycle, it's possible to literally run into trouble :)

Did moving G90 to the top fix your issue, by the way?

ryanneufeld, 2018-10-09:
Yes it did!  I want to experiment with multiple cameras now!

--Ryan Neufeld BSc CIS
ryan@neufeldmail.com

smplfy, 2018-10-12:
Hi there, just noticed when running an external DSLR that I had a communication timeout. Seems to function mostly normally but every time the snapshot is taken I get this:

"Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves."

Sometimes I get that message twice in a row.

FormerLurker, 2018-10-17:
@smplfy, so sorry for the late response, but I've been unable to work on Octolapse for the last week or so.

Check your camera profile and look at the 'Timeout' setting.  I think it defaults to 5000MS (5 seconds).  That's a really long time IMO, but if you are taking very high resolution images it takes a long time to both save the image to your camera's RAM, and then even longer to copy them to the PI via USB (probably usb 2.0).  Though increasing the timeout will (hopefully) eliminate the error message, you have two options to reduce the amount of time it takes to acquire an image from the DSLR in the first place:

1.  Reduce the resolution - this will have a big impact.  If you dont need 4000x6000 or higher resolution for video editing (or some other purpose) you should consider reducing it somewhat.
2.  Prevent the camera from downloading the image to the pi until right before the timelapse renders.  This has the unfortunate side effect of preventing snapshot preview (the picture won't update on the Octolaspe tab), preventing snapshot transposition (flip, mirror, transpose, etc), and preventing text overlays (rendering settings).

Please take a look at this reply I made on the Octoprint discourse site: https://discourse.octoprint.org/t/octolapse-how-to-turn-off-saving-and-rendering-to-the-pi/4341/5?u=formerlurker

It explains how to configure Octolapse so that it:

1.  Deletes all images on your camera SD card recursively (this stinks, but it's necessary.  backup your family photos first!)
2.  Set the capture target to SD instead of internal RAM.
3.  Capture an image from the DSLR, but return IMMEDIATELY after the shutter closes, before the image is saved to SD.
4.  Right before rendering download all of the images from the camera and rename/place them in the appropriate directory so that the timelapse can be rendered normally.

Note: Rendering a 4K+ video on an RPI takes a really long time and takes a lot of space.  

I'm working on a guide for this process, as well as another setup video.

Let me know if this helps.

smplfy, 2018-10-18:
Thanks so much for your help. 

I’m working through this right now, but I’m having issues getting the images to delete recursively. Gphoto is telling me that operation is not supported. I’m going to try removing that section and just manually clearing the camera before a shoot. I also was having issues using the trigger command, the camera would just hang and say “connecting to pc” and not ever save the photo. Changed that to capture-photo, hoping the capture target still works with that command. I know with capture-photo we have to wait for it to be stored to the SD, but the delay is still much shorter than it was before. Still learning and not much experience with gphoto2!

BTW thanks for your comments on youtube, I’m publishing these under the Tiny Planet logo. Just so you know who you’re talking to!

I actually tested having my timeout set to the max of 25000ms and was still getting this error. I am using an older DSLR (Nikon D40X) and it’s using, at best, USB 2.0, so the capture times were super long. The store on camera / download at the end approach seems like a good workaround. Fingers crossed!

Thanks again and thanks for your awesome work!

-Daniel

HerrRossi-cgn, 2019-01-03:
Hello FormerLuker
I have the problem that my Canon 550D goes into sleep mode as soon as the intervals between the photos are longer than 30sec.

Then I get the error message:
snapshot_script_error: The snapshot script failed with the following error message: *** Error: No camera found. **

how can I wake up the camera befor the photo will taken?
do I have to do that in the script or is there a command in Octolapse?
Am a beginner in Linux.
Thank you for your help and the great Plug In