FormerLurker/Octolapse #243: [feature request] Import slicer settings

MimbaMonkeyHouse, 2018-09-19:
I believe we would all benefit from being able to import all slicer settings directly or via a setting file, say .curaprofile or .fff!

FormerLurker, 2018-09-19:
Hi @mimbamonkeyhouse!  This is very high on my list of priorities, actually, so thanks for mentioning it.  I hope to read slicer settings directly from the gcode file soon after the next release.  

What version of octolapse are you using, if I might ask?

FormerLurker, 2019-10-12:
This has been completed at long last, but is even better than what was suggested.  Octolapse will now import slicer settings directly from the gcode file!  Cura requires a small script to be added to get this working, but it's easy peesy.  Let me know if you want to try it out.  This feature is included in V0.4 (unreleased), but it's been under testing for a good while, and most things are pretty stable now.

Closed!

closed 2019-10-12 by FormerLurker