FormerLurker/Octolapse #244: Synchronization failed error, no frames were captured

4Z1ZV, 2018-09-19:
Printer: CR10 Mini

Latest version of Octoprint and Octolaps, installed today.

Default Octolaps setting.

RPI 3 Model B.

No errors nor related messages during print.

Error pips up at the print end just before greem message - Saved to SD Card..

No file saved.

4Z1ZV, 2018-09-19:
[SD CARD CASES PLA.gcode.txt](https://github.com/FormerLurker/Octolapse/files/2398791/SD.CARD.CASES.PLA.gcode.txt)

Attached please find a sample file with my current start and end G-Code.

4Z1ZV, 2018-09-19:
Many thanks for you ultra fast response.

I will test it on my next job.

Will take longer this time...  :-)

4Z1ZV, 2018-09-30:
OK. The solution does the job.

:-).

Hoever it raised new issue - The current height is constantly reported as 0.2mm.

When I disabled the Octolaps, the report is OK.

4Z1ZV, 2018-09-30:
Just to be more accurate, the reported height show the real value when thete is a layer change and the print head is licated a the shot position.

FormerLurker, 2018-09-30:
Are you talking about the [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) plugin?  If not, maybe you could send me a snapshot of what you're seeing?

4Z1ZV, 2018-09-30:
====================
Zvi Segal
zvisegal@yahoo.com
Mobile: +972-546-666651
4Z1ZV
====================

Sorry for typos, sent from my mobile....

4Z1ZV, 2018-10-01:
Here is a copy of the printer state.

Once I enable the Octolaps,  Current Height remain unchanged except for layer change (when printer head goes to the defined position).

File: CR-10_Dial_Indicator_PETG.gcode 
User: zvika
Timelapse: On Z Change
Filament (Tool 0): 9.28m
Approx. Total Print Time: 1 hour
Print Time: 00:14:08
Print Time Left: 1 hour ●
Printed: 309.3KB / 1.9MB
Current Height: -0.20mm
Current Height: 0.20 / 30.52mm 
Layer: 10 / 108

FormerLurker, 2018-10-01:
@4Z1ZV, 

I don't have a 'Current Height' on my instance.  I still think that looks like the [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) plugin.  Can you verify that you have the DisplayLayerProgress plugin installed?  If so, you may want to post an issue on the DisplayLayerProgress plugin page [here](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/new).

Also, if you look in the Octolapse tab, open the current settings by clicking the link, and enable the position and position state info panels (sliders at the bottom) while a print and Octolapse is running, you should see the correct height/layer numbers.

Please let me know one way or another.

4Z1ZV, 2018-10-01:
I enabled the position and position state.
It does open a window and show the correct height.

However the reported height on the main tub is still incorrect.

When I disable the Octolaps, event during print, on the next layer it shows the correct value.

FormerLurker, 2018-10-01:
@4Z1ZV, sorry to ask again, but you didn't answer my question about the [DisplayLayerProgress plugin](https://plugins.octoprint.org/plugins/DisplayLayerProgress/).  Did you install this plugin?  I'm just trying to figure out what is going on, and this will help.

4Z1ZV, 2018-10-01:
The DisplayLayerProgress was installed.

I also tried to disable it. SAME ISSUE.

When I disable the Octolaps - The Height value is OK.

When I disable the Octolaps - It shows the following:

On layer change it momentary shows the real height of the head.

Than it shows the relative value - related to the last lap position.

FormerLurker, 2018-10-01:
OK, thanks!  Now try to disable the 'DisplayLayerProgress' plugin, then REBOOT Octoprint, then try running a print (rebooting is probably required to completely disable the plugin).  I'm pretty sure the height/layer display will vanish from OctoPrint.  If that's the case (and I'm pretty sure it is) then the problem is with the DisplayLayerProgress plugin and not Octolapse.

My guess would be the DisplayLayerProgress plugin isn't able to correctly account for relative Z movements (G91), and you should probably post this issue to the DisplayLayerProgress plugin home page here:  https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/new

FormerLurker, 2018-10-02:
Thanks for going back and forth with me, I know debugging can be frustrating.  

So here is what you say you are seeing:
```
File: CR-10_Dial_Indicator_PETG.gcode
User: zvika
Timelapse: On Z Change
Filament (Tool 0): 9.28m
Approx. Total Print Time: 1 hour
Print Time: 00:14:08
Print Time Left: 1 hour ●
Printed: 309.3KB / 1.9MB
Current Height: -0.20mm
Current Height: 0.20 / 30.52mm
Layer: 10 / 108
```

Here is what I'm seeing:

[image]

I don't even see the 'Current Height' or 'Layer' information on my printer state panel, so I'm really confused how/why it's there!  Any thoughts?

4Z1ZV, 2018-10-02:
[
[image]
](url)

This is my screen at Offline state.

4Z1ZV, 2018-10-02:
It is time to update to latest release :-)

Shadowen, 2018-10-13:
Not sure if this was mentioned, but @4Z1ZV 's screenshot is obviously taken on mobile... with the TouchUI plugin: https://plugins.octoprint.org/plugins/touchui/. I stopped using that plugin a while ago because it wasn't working well for me, but your experience may vary.

FormerLurker, 2019-10-12:
So, this is now a pretty old feature, and it's very possible that if this was a bug it's been solved in one of the many updates over the last year.  I'm going to close this down, but if you would like me to reopen, please let me know, or just post a new issue.  Thanks for your post!!!

closed 2019-10-12 by FormerLurker