FormerLurker/Octolapse #245: Unable to install Octolapse

JohnnyBBravo, 2018-09-21:
I've done a manual installation of OctoPrint and it seems i'm not able to install the Octolapse plugin.
When using the plugin manager, i select "get more", then find Octolapse and click install.
Everything appears to install OK, i even get the message "Successfully installed Octolapse-0.3.1
Done!", but the little pop-up at the right side says "Plugin management log, install unknown" with a little check mark next to it. When restarting OctoPrint (and clearing browser cache) Octolapse is nowhere to be found (as if it never installed, even though it said it did).

On a side note: FFMPEG is installed and Pillow installation went fine. I've also tried installing Octolapse rc/maintennce release, but the result is exactly the same. Pip is version 18.0, Pillow is version 5.2.0

#### Version of Octolapse
Octolapse Version: 0.3.1

#### Version of OctoPrint
OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO - Unable as Octolapse doesn't install

#### What were you doing when the problem occurred
Attempting to install Octolapse via the OctoPrint's plugin manager.

#### What should have happened?
Octolapse should have installed

#### What happened instead?
Message said Octolapse successfully installed, but after OctoPrint restart it is nowhere to be found and seems as if it didn't install

#### Operating System running OctoPrint and Octolapse
OS Name: Raspbian Jessie (On a BananaPi Zero)
Os Version: 3.4.113-BPI-M2Z-Kernel

#### Browser and version of browser, operating system running browser
Browser: Chrome 69.0.3497.100
Browser OS: Windows 10

#### Link to settings.json
I don't know how to export settings.json

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://pastebin.com/i58fadNr

#### Link to octoprint.log
Link to octoprint.log: https://pastebin.com/rLZe1H8x

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: https://imagebin.ca/v/4GSCoWsG5pXw

FormerLurker, 2018-09-21:
That is very strange.  I've never seen that before!  Octolapse is clearly installed since I see a log file was generated.  Can you snap a pic of your Octoprint page for me?  I'd like to see if the title bar add on loaded (it would say octolapse in the title bar).  Also, can you look in the octoprint settings page and look under 'plugins' (bottom left corner of the settings screen) for an octolapse link?

We'll get to the bottom of this!

JohnnyBBravo, 2018-09-21:
I don't get it either...
The title bar add-on doesn't load - Screenshot here: https://imagebin.ca/v/4GTILzurkhCM
The plugins page also doesn't show Octolapse - Screenshot here: https://imagebin.ca/v/4GTIp2AJrWMk

I've looked at this thread https://discourse.octoprint.org/t/installing-a-plugin-shows-me-unknown-in-the-popup-i-cant-uninstall-plugins-through-the-plugin-manager-octoprint-up-to-1-3-8-octopi-0-15-0-or-pip-10-x/1746 but downgrading pip to 9.0.3 doesn't make any difference.

Thanks for your help!

JohnnyBBravo, 2018-09-21:
The only hint might be in octoprint.log which states:
`2018-09-21 10:47:51,730 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 847, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 41, in <module>
    import octoprint_octolapse.camera as camera
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 41, in <module>
    import octoprint_octolapse.camera as camera
AttributeError: 'module' object has no attribute 'camera'`

Although I have no idea what all that means...

FormerLurker, 2018-09-21:
Forgot to look at octoprint.log, whoops.  There is definitely some error loading the installed plugin.  Can you maybe try installing the latest devel/RC via the plug-in manager via the plug-in manager from this link: https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev3.zip

I suspect a missing prerequisite, though I don't know what it could be yet.  After you reboot can you link to the latest octoprint.log (give me the whole file if possible please).  

Thanks!

JohnnyBBravo, 2018-09-21:
The result of trying to install the latest devel/RC as you suggested gives the same results.
octoprint.log is here: https://filebin.ca/4GTPLfVfZ3j4/octoprint.log

Thanks!

JohnnyBBravo, 2018-09-22:
SUCCESS!!

These two lines in octoprint.log caught my attentions:

`ImportError: /usr/lib/arm-linux-gnueabihf/libc.so: invalid ELF header
2018-09-22 16:00:56,884 - octoprint.plugin.core - ERROR - Error loading plugin octolapse`

The error loading octolapse came immediately after the error about libc.so, therefore it seemed that libc.so was potentially a culprit and i did the following:

`mv /usr/lib/arm-linux-gnueabihf/libc.so /usr/lib/arm-linux-gnueabihf/libc.so.old
cp /usr/lib/arm-linux-gnueabihf/libc.so6 /usr/lib/arm-linux-gnueabihf/libc.so
service octoprint restart`

And presto, all is working fine :-)

closed 2018-09-24 by JohnnyBBravo

JohnnyBBravo, 2018-10-01:
There are some issues with this solution though.
I wanted to compile some other software and the C-compiler threw an error...

Reverting the libc.so to the original file solved this and allowed me to compile, however afterwards i had to do the same trick as mentioned above to prevent octoprint from throwing an error.

I'm not sure if there is a permanent solution to this, as this is all a bit tedious...

FormerLurker, 2018-10-01:
Yeah, what an odd issue.  Please keep me updated.  If I can ever replicate this I'll definitely dig in and see what I can find.  I may have to grab a BananaPi and see what I can figure out.