FormerLurker/Octolapse #246: Octolapse with Cura and Simplify3D - getting mixed results

Travman1970, 2018-09-23:
#### If this is a feature request describe it here

On my Ender 3, Octolapse functions fine with time lapse videos when object is sliced using Cura, but when the same object is sliced in S3D, the video is not good.  Both slicers use similar slicing parameter and both used identical start & end gcodes.

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Print item on Ender 3 after sliced in Cura = Octolapse video looks very good (at correct stabilization settings).
2. Print same item after sliced in S3D = Octolapse video the bed and extruder is all over the place in video not where the setting states where it should be (kept extruder at center)
3. Both slicers use the identical start & end gcode.

#### What should have happened?
I should have the video with correct bed and extruder placement according to the stabilization settings.

#### What happened instead?
Not getting correct stabilization during time lapse.  One thing I noticed is that when object printing is sliced in Cura, as soon as I put "print' the extruder goes to home and waits till bed and extruder is at temp.  When sliced in S3D, the extruder does not move until bed & extruder is at temp, then goes to home and starts the print.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: Marlin 1.1.8 customed from TinyMachines3D

#### Browser and version of browser, operating system running browser

Browser: Chrome 69
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-09-24:
My guess is you're running into a bug (fixed in the latest development rc release) that causes snapshots to be taken too early when using 'Absolute Extrusion'.  It doesn't always happen, but some combinations of G92 when they appear close to a layer change can cause this.  You have two options to fix this:

1.  Switch to relative extrusion in S3D.
2.  Upgrade to the latest development release candidate via the plugin manager using this url:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev3.zip

If you use options number 2, you might want to check out this video that explains how to configure your printer profile in the new version:  https://youtu.be/sDyg9lMqMG8

Let me know how it goes!

FormerLurker, 2018-09-25:
Check this guide out for configuring your camera:  https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-image-preferences

Pay close attention to the required edits to octopi.txt (as well as the security ramifications if your pi is accessible via the internet).

Let me know how it goes!

Travman1970, 2018-09-25:
Yeah seems like it.

Chris
Christopher M. Travis
8810 42nd Ave
Kenosha, WI 53142
Tel: 262-652-0030  Cel: 262-484-8835
email: ctravis13@gmail.com

"*Si fractum non sit, noli id reficere* - *If it ain't broke, don't fix it*"

FormerLurker, 2018-09-25:
@Travman1970, looks like you may have inadvertently doxed yourself :)  If you didn't intend to do that, maybe you should edit your comment and remove the identifying info.

Travman1970, 2018-09-25:
yeah did the reply in the email.

FormerLurker, 2018-10-01:
@Travman1970, since we resolved your initial issue I'm going to close this out.  Let me know if you want me to reopen it, or you can submit a new issue if you run into any additional problems.

Thanks for posting!

closed 2018-10-01 by FormerLurker