FormerLurker/Octolapse #249: Error: invalid literal for int() with base 10

ChrisWLeon, 2018-09-30:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Sent a print job from Cura
2. Octoprint Fails Error: invalid literal for int() with base 10
3. STEP_...

#### What should have happened?
no crash

#### What happened instead?
crashed

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi on Raspberry 3 B+
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: TH3D Unified Firmware U1 R2.2

#### Browser and version of browser, operating system running browser

Browser: Safari
Browser OS: MAC
[CCR10_wildrosebuilds_puzzlecube.gcode-3.txt](https://github.com/FormerLurker/Octolapse/files/2431519/CCR10_wildrosebuilds_puzzlecube.gcode-3.txt)
[octoprint-2.log](https://github.com/FormerLurker/Octolapse/files/2431520/octoprint-2.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2431522/plugin_octolapse.log)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-09-30:
Hi Chris, thanks for posting!

I see the issue and will try to patch it for the next release.  Four M117 lines have the ellipsis (…) character, and for some unknown reason this is causing problems.  If you remove them, Octolapse should continue working.

They are the very last character on line 11, 14, 20 and 29 of your gcode file (M117 command).

I'll let you know when this is patched.  Thanks again for reporting this!

FormerLurker, 2018-09-30:
@ChrisWLeon,

I thought I already sent this response, but something must have happened.  If you get some of this info twice, I apologize.

I found what I believe to be an error in your gcode file on line 28:

`Gi z1.0 F3000 ; move Z up 1mm at 3000mm/min`

I believe Gi should be G0 (or G1).  This was causing an error that I also patched .  Octolapse would ignore this command now, and I believe your printer probably is too.  If you could test that for me by running the command through the OctoPrint terminal tab I'd be appreciative.  This code should work if you run it line at a time:
```
G90; set XYZ to absolute coordinates
G28; Home XYZ
M114; Get the current position, let me know what this returns please
Gi z1.0 F3000 ; should not do anything
M114; Gets the current position.  It should not have changed from the previous m114(hopefully)
```

I believe I have the other item fixed (Issue #252).  I have pushed a patched version that you can install via the plugin manager using this URL in the '... from URL' box:

`https://github.com/FormerLurker/Octolapse/archive/bdd085b8fb74a0bdeb3818865a4b3fa9c5f5517e.zip`

This version is quite a bit different than the one you are using now.  If you decide to install the update vs removing the ellipses, be sure to watch [this](https://www.youtube.com/watch?v=sDyg9lMqMG8&t=3s) video which should explain setup in the new version.  There are a few things in your printer profile you'll want to change before using the new version.

FormerLurker, 2018-09-30:
One last thing:  I noticed you are using OctoPrint 1.3.9, which technically isn't compatible with Octolapse 0.3.1 (though it does sometimes work, which is odd).  It may be a good idea to upgrade.

ChrisWLeon, 2018-10-01:
Thanks for your help, and your fast response. I have made most of the changes but have to wait for a print to finish before i can test it. You mention below upgrading. I am under the impression that 1.3.9 and 0.3.1 are both the latest versions? Which version are you suggesting?
Regards and thank you again
Chris

FormerLurker, 2018-10-01:
@ChrisWLeon, 

The version of octoprint you are running is fine as long as you upgrade Octolapse.  0.3.1 is the latest 'Stable' release.  However, 0.3.1 is now 149 updates behind the latest 'pre-release' version.

With that in mind, I'm copying text from a previous comment I made here (though I changed the URL since I made a few more updates):

>I believe I have the other item fixed (Issue #252). I have pushed a patched version that you can install via the plugin manager using this URL in the '... from URL' box:

>https://github.com/FormerLurker/Octolapse/archive/64f75d9c8226ca5629508500ad03fa1694828496.zip

The link above is for v0.3.4.rc1.dev3, which is the most recent 'pre-release' that includes fixes for the unicode (ellipsis) problems you mentioned.  Make sure you watch [this video](https://www.youtube.com/watch?v=sDyg9lMqMG8&t=3s) which explains how to configure the new version.

Let me know how it goes!

FormerLurker, 2019-10-12:
So, this issue has been stale for about a year.  I'm guessing the problem has been solved.   Also, Octolapse now handles unicode characters within m117 much better, so this problem isn't likely to exist on the newest version (V0.4, which is in testing).

Thanks for your submission, and let me know if you'd like this issue to be reopened!

closed 2019-10-12 by FormerLurker