FormerLurker/Octolapse #254: slicer settings overriding problem

rafaa1988, 2018-10-02:
I really like Octolapse but don't like the way it overrides my slicer settings. I was sending Gcode from my slicer to my printer without problems. Is there a way to keep my slicer's settings not effected by octolapse?

FormerLurker, 2018-10-02:
Hi @rafaa1988!  Octolapse doesn't change your slicer settings, but it does need to know some of them in order to operate correctly.  Specifically it needs to know how much to retract, how fast to retract/detract, what (if any) zhop height you are using and how fast to move XY and Z axis.  Octolapse injects its own gcode to move the printhead, and can optionally lift and retract before doing so, thus it needs some knowledge about your slicer settings.

Maybe I'm not understanding you correctly.  Can you let me know if I am, and if not can you explain a bit more?

Thanks!

rafaa1988, 2018-10-02:
Thanks! Maybe I don't understand Octolapse well enough. I think that Octolapse takes the Gcode, analyzes only the G and M codes. Any other information (like as said retract, detract and so on) that were provided with the slicer won't be analyzed. Instead they should be entered in the octolapse slicer settings. I think this is not the case when using Octoprint only. Because Octoprint will just send all lines line by line to the printer.

FormerLurker, 2018-10-02:
OctoLapse works this way (basically) when NOT in test mode

1.  Watch each line of gcode that comes and update the current position, position state, and extruder state (need to know some slicer settings to do this properly)
2.  Look for a 'Trigger' like a layer change, snapshot gcode, or timer (depends on your snapshot settings) (also need to know some slicer settings to do this)
3.  If a trigger is found, stop sending gcode from the file and stabilize the print head (needs to know some slicer and printer information to do this)
4.  Continue sending gcode from the file and repeat at step 1

Does that clear some things up?

Step 3 is the only time that gcode is modified.  If Octolapse didn't do this it couldn't stabilize the print!

Now, if you are running in 'Test Mode' Octolapse will strip all E axis movements (and some other things like bed/extruder warm up), but that's just for testing purposes.

rafaa1988, 2018-10-02:
Big thanks! It's clear now

closed 2018-10-02 by rafaa1988

FormerLurker, 2018-10-02:
No problem, thanks for posting!  Come back anytime!