FormerLurker/Octolapse #255: Unable to install

ManuGithubSteam, 2018-10-07:
#### Version of Octolapse

Octolapse Version: 

latest dev version.
and
master-zip

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

1. install it via URL as i can not find it in the plugins manager of octoprint.

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE
Interface of Octolapse appear somewhere.

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

Noting, I can not start/ configure octoalpe as there is no interface for it.

#### Operating System running OctoPrint and Octolapse

OS Name: raspiab lastest update was yesterday.
Os Version: OS_VERSION_GOES_HERE

FormerLurker, 2018-10-08:
@ManuGithubSteam,

Thanks for posting this issue!  I have a few questions:

1.  Can you post your octoprint.log file somewhere and send me a link?  

2.  Does the Octolapse install fail completely, or does it succeed, then after rebooting you don't see Octolapse anywhere?

3.  When you say you tried the latest dev version, can you post the link you used?

I must say that I'm really confused why you can't find Octolapse within the plugin manager.  When I search for 'Octolapse' it shows up just fine.  Can you post a screenshot of your search?  It's not that I don't believe you, but I'd like to see exactly what you are seeing.

ManuGithubSteam, 2018-10-08:
Hi
Thanks for looking into it :-)
> @ManuGithubSteam,
> 
> Thanks for posting this issue! I have a few questions:
> 
> 1. Can you post your octoprint.log file somewhere and send me a link?

[octoprint.log.2018-10-07.zip](https://github.com/FormerLurker/Octolapse/files/2456871/octoprint.log.2018-10-07.zip)

> 2. Does the Octolapse install fail completely, or does it succeed, then after rebooting you don't see Octolapse anywhere?

It succeeds, like octoprint is telling me that the install went ok. But after reboot i see nothing.

> 3. When you say you tried the latest dev version, can you post the link you used?
> 
I used these links:

https://github.com/FormerLurker/Octolapse/archive/master.zip
https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev3

> I must say that I'm really confused why you can't find Octolapse within the plugin manager. When I search for 'Octolapse' it shows up just fine. Can you post a screenshot of your search? It's not that I don't believe you, but I'd like to see exactly what you are seeing.

[image]

Hope this helps :)

FormerLurker, 2018-10-08:
It does help, thanks!  It says in the logs that Octolapse 0.3.1 is installed but disabled.  It gives a similar message about 0.3.4.rc1.dev3.  Can you try the following please:

1.  Scroll through your plugins and uninstall all versions of Octolapse that appear.  Your log indicates that there may be multiple versions for some reason (probably something I did, lol)
2.  Reboot the pi (a real reboot, not just restarting octoprint)
3.  Try searching for Octolapse again (it should appear now), but don't install it!  Let me know if it does NOT appear.
4.  If step 3 succeeded, try installing 0.3.4.rc1.dev3 again from this link:  https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev3

Let me know how it goes!

ManuGithubSteam, 2018-10-08:
> It does help, thanks! It says in the logs that Octolapse 0.3.1 is installed but disabled. It gives a similar message about 0.3.4.rc1.dev3. Can you try the following please:
>
> 1. Scroll through your [...]
Done, Found one instance.
> 2. Reboot the pi (a real reboot, not just restarting octoprint)
done
> 3. Try searching for Octolapse again (it should appear now), but don't install it!  Let me know if it does NOT appear.
it appears
> 4. If step 3 succeeded, try installing 0.3.4.rc1.dev3 again from this link:  https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev3
in the works right now
> 
> Let me know how it goes!

FormerLurker, 2018-10-08:
whoops, I gave you the wrong link!  Try this link instead:

```
https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev3.zip
```

ManuGithubSteam, 2018-10-08:
Ok i managed to install it.

But:

- Setting: Layer change High Quality is not available.
- Despite adding the G90 Gcode into my starter Gcode sequence Octolapse complains that it does not find it.

I own a Prusa MK3 so what would be the settings for that ?

Thanks a lot!

FormerLurker, 2018-10-08:
Progress!

Can you export and attach your settings.json file from Octoprint Settings (wrench icon)->Octolapse->Download Settings?

Also, can you please attach your full gcode file and I'll run it through the debugger.  I've got a few other requests to deal with today, so I'm not 100% sure I'll be able to get to it until tomorrow, but I promise I'll try!

ManuGithubSteam, 2018-10-08:
Ok cool. 
I was able to fix the G90 problem by putting it in the very top of the start gcode and also in the filament specific code on the very top.

[Test_old.zip](https://github.com/FormerLurker/Octolapse/files/2457430/Test_old.zip)
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/2457432/Settings.zip)

Now i get this:
[image]

FormerLurker, 2018-10-08:
Awesome!  However, from looking at your original start gcode, I can tell that there is a bug somewhere.  That should have worked.

Now this final issue is actually pretty easy to fix, fortunately!  I'd like to fix the MK3 profile so this doesn't happen to other users.  Here's what you should do:

Go into your Octolapse Printer Profile and click the following check box:
```
Override Octoprint Printer Profile
```
That will show some new controls.  Enter the following values
XMin:0
XMax:250
YMin:-3
YMax:210
ZMin:0
ZMax:210

Notice that Ymin is negative!  That's because the MK3 (and the Mk2) prime outside of the regular print area.  Hopefully that fixes your final issues.  Let me know if you have more trouble!

ManuGithubSteam, 2018-10-08:
Ok i put that in there but now this happens:

Position - Coordinates (X:None,Y:None,Z:None,E:0.00000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

This hapens if the extruder has to home x first i belive. Wich he will do every time...

FormerLurker, 2018-10-08:
I looked in Test_old.gcode (the file you sent) and found one line that will cause an error in your start gcode.  You have this line:

G1 X0 Y200 Z50; home X axis

Which is definitely not safe to run before homing XZ and Z.  I can't see any reason to have this line. If I were you I'd remove it completely.

Second, the file you sent me has only a single layer, and on my debugger when running with your settings.json file (I did have to update the print volume) it takes a picture as expected but does not render a timelapse since it only captured a single frame.  If you are printing a different gcode file (not the test_old.gcode file you sent), please send it to me so I can try it out.

Thanks!

ManuGithubSteam, 2018-10-10:
Hi
i actually tied to print this in the picture above.: 

[Test_2.zip](https://github.com/FormerLurker/Octolapse/files/2465867/Test_2.zip)

As you can see it has many layers but octolapse just stopped after two snapshots. 

[image]

The whole point of that line of gcode is:
The mk3 sometimes takres the shortest route to home the x axis. This is sometimen on the right side. Because the right Endstop ist just closer. So in this circumstance. The Printer thinks that right is the home position wich is obviously false and all the homeing and bed level will fail afterward.

So this line tries to make sure that the left side is always nearer and it will home to that direction.

FormerLurker, 2018-10-10:
Thanks for clearing that up.  I'll take a look after work.  If you've changed your Octolapse settings at all over the last two days, send me a new copy and then I'll have the latest version.

Octolapse should query the printer for a position after the home command is issued using the default settings, but I'll double check to make sure that's happening in this case.  I might be able to edit the virtual printer to home to a different location to test.

ManuGithubSteam, 2018-10-10:
Hi

No i did not change the settings others that you said here. 

I assue the problem is this:

The extruder is somewhere in the the middle or side but not definded region.

Then this happens:

Position - Coordinates (X:None,Y:None,Z:None,E:0.00000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

As The printer did not home yet and does not know its position.
And as far as i can see as long this message is displayed there is nothing happen in the snap picture department. 
Im not sure if the bed level prcedure does not set something arwy too and then octolapse is on a toal loss and does not know where it is etc.

I think it would be better to not check for position at all as you know from the gcode line who is printed what you print like infill and then you can desicide: now take a pic....

FormerLurker, 2018-10-10:
@ManuGithubSteam, something went wrong when your snapshot profiles were upgraded from v0.3.1 :(

Can you try restoring the default settings (Octoprint Settings (wrench)->Octolapse->restore default settings), reconfigure your printer profile (sorry about that) and try again?  I think that should fix it.

The problem is the 'feature detection enabled' setting seems to be missing from your snapshot profile.  The 'Layer Change' profile should have vanished when you upgraded, but for some reason it was not.  

Let me know how it works!

ManuGithubSteam, 2018-10-11:
Hi 
i reset everything as you said but its not working as you can see here:
Printed the same gcode.
Still same error at the beginning and then nothing happens anymore.

[image]
[image]

FormerLurker, 2018-10-11:
Try switching to the 'Layer Change - Compatibility' profile.  It's best to get that one working before using the high quality profile.  You can see in the 'layer trigger' information that it's waiting for features to trigger, which probably means you are running into the 'minimum layer time' (see your slicer settings).  The high quality trigger won't work if the slicer slows down a layer because Octolapse can't predict what the speeds will be.

ManuGithubSteam, 2018-10-12:
Hi
Ok this seems to have worked now i have 700 snapshots.

But: Where can i download the timelapse. Where is it stored?

FormerLurker, 2018-10-12:
Check the built-in timelapse tab.  It should be there.

ManuGithubSteam, 2018-10-12:
Its not :-(
Could it be that it is still rendering ? How would i check that ?

[image]

FormerLurker, 2018-10-12:
If you are rendering with one thread (see the rendering advanced settings), it can take a few hours.  Look in the octolapse tab and hover over the rendering icon (film strip).  However, from your picture earlier it doesn't look like it :( send me logs please.

If you use a sftp client you could browse to the folder and you might be able to grab the images directly.  Check /home/pi/. octoprint/data/octolapse/snapshots.  There should be subdirectories in there (2 levels deep) that look strange (guids).  If your images haven't been purged yet they should be there.

Ug..  sorry for the issues.  It's always best to test on a very small print before trying something larger.  Maybe do that for your next test?

FormerLurker, 2018-10-12:
Oh, check your free space too!  If ffmpeg errors out it might not report the issue (need to fix that).

ManuGithubSteam, 2018-10-14:
> If you are rendering with one thread (see the rendering advanced settings), it can take a few hours. 

Its not happening :-(
In the rendering Settings I can NOT select how many thread should be used. There is no option for it.

Look in the octolapse tab and hover over the rendering icon (film strip). However, from your picture earlier it doesn't look like it :( send me logs please.
> 
Nothing did happen...

> If you use a sftp client you could browse to the folder and you might be able to grab the images directly. Check /home/pi/. octoprint/data/octolapse/snapshots. There should be subdirectories in [...]
It looks like this:

toni@octo:~/.octoprint/data/octolapse/snapshots $ ls
0cc99eaa-69a3-496a-bdb5-97a83f8e613f  9791a8e6-0392-4c33-80ab-38b22bef0890  c1df1765-614a-49a0-aa7a-a73d64cceab5              latest_snapshot_thumbnail_300px.jpeg
126cc8c0-dd11-4568-80ba-ec6d2bd21c9e  ab0a5003-bfe4-4990-b1c9-9662a8999456  dc7b11b3-b0f2-4196-829a-56c7ade09f9f              latest_thumb_354def78-9eea-409a-ad23-ee966dfff4ba.jpeg
44cd1e74-273b-4977-988d-32605d267205  ba97da05-d80d-4ea8-8205-4e0540a50f85  latest_354def78-9eea-409a-ad23-ee966dfff4ba.jpeg
4b405e31-73b5-4d8e-bcd8-323dfe0038ea  bea6ad62-c6d3-4731-aa85-3e0712210a4c  latest_snapshot.jpeg
toni@octo:~/.octoprint/data/octolapse/snapshots $ cd dc7b11b3-b0f2-4196-829a-56c7ade09f9f
toni@octo:~/.octoprint/data/octolapse/snapshots/dc7b11b3-b0f2-4196-829a-56c7ade09f9f $ ls
354def78-9eea-409a-ad23-ee966dfff4ba
toni@octo:~/.octoprint/data/octolapse/snapshots/dc7b11b3-b0f2-4196-829a-56c7ade09f9f $ cd 354def78-9eea-409a-ad23-ee966dfff4ba/
toni@octo:~/.octoprint/data/octolapse/snapshots/dc7b11b3-b0f2-4196-829a-56c7ade09f9f/354def78-9eea-409a-ad23-ee966dfff4ba $ ls
metadata.csv  Test_2000000.jpg
toni@octo:~/.octoprint/data/octolapse/snapshots/dc7b11b3-b0f2-4196-829a-56c7ade09f9f/354def78-9eea-409a-ad23-ee966dfff4ba $ 

> Ug.. sorry for the issues. It's always best to test on a very small print before trying something larger. Maybe do that for your next test?

[Ukulele.tar.gz](https://github.com/FormerLurker/Octolapse/files/2476184/Ukulele.tar.gz)

Ufotunatelly i did not use live diganostic. Will do with the next print.

ManuGithubSteam, 2018-10-14:
Just got this:
[image]

ManuGithubSteam, 2018-10-14:
Another error:
[image]

Full log:
[plugin_octolapse.tar.gz](https://github.com/FormerLurker/Octolapse/files/2476770/plugin_octolapse.tar.gz)

FormerLurker, 2018-10-17:
@ManuGithubSteam 

Sorry for the delay!  I've been unable to work on Octolapse due to life in general :)

Here is the relevant log entry:

```
2018-10-14 18:51:35,325 Running ffmpeg with command string: /home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG%06d.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"
2018-10-14 18:51:35,502 return-code: Could not render movie, got return code 255: ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstFileSink:filesink0: Could not open file "mp4" for writing.
Additional debug info:
gstfilesink.c(423): gst_file_sink_open_file (): /GstPipeline:pipeline0/GstFileSink:filesink0:
system error: Permission denied
```
Depending on your snapshot profile settings (specifically the 'Clean Snapshots After Rendering Failure' setting), you might be able to run this command via ssh:
```
/home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG%06d.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"
```

See if that works first.  My guess is this you installed avconv (gpu_avconv???  never heard of it) manually and there are some permission issues.  I'm not sure, but it's a good place to start.

FormerLurker, 2018-10-17:
@ManuGithubSteam,

Whoops, I just realized that this command uses a temp directory:
```
/home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG%06d.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"
```
The snapshot files should still be on your pi depending on the 'Clean Snapshots After Rendering Failure' setting.  Here is an example directory from my pi containing some snapshots:

```
/home/pi/.octoprint/data/octolapse/snapshots/e41b117d-9c6b-46e7-ac1c-89e91b05255d/9c1b7f43-a6a7-4eb1-8b33-0fb62ed59cf6
```
Here is the form:  {octoprint install directory}/data/octolapse/snapshots/{print_guid}/{camer_guid}

Now my direcory will always be different than yours since the Guids are essentially random, but here is the avconv with MY snapshot directory entered:

```
/home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/home/pi/.octoprint/data/octolapse/snapshots/e41b117d-9c6b-46e7-ac1c-89e91b05255d/9c1b7f43-a6a7-4eb1-8b33-0fb62ed59cf6/new_v09_0.2mm_PETG%06d.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"
```

I recommend using filezilla, or another sftp client, which makes browsing the folders very easy.  Please let me know if you need any additional assistance, and good luck!

ManuGithubSteam, 2018-10-25:
> @ManuGithubSteam
>
> Sorry for the delay! I've been unable to work on Octolapse due to life in general :)
>
> Here is the relevant log entry:
>
> ```
> 2018-10-14 18:51:35,325 Running ffmpeg with command [...]

Sadly iam not able to run the command successfully:

toni@octo:~ $ ls
1            hub-ctrl.c                 ledband_tween_light.conf       master.zip         package-selections       rc.local               test.txt
fipbox       ledband_sz.conf            ledband_tween_light_new2.conf  mjpg-streamer      prusa3d_fw_3_0_12        scripts                venv
haproxy.cfg  ledband_tween_light2.conf  ledband_tween_light_new.conf   oauth2_creds.json  prusa3d_fw_3_0_12.zip    Slic3r-version_1.36.2  version_1.36.2.tar.gz
hub-ctrl     ledband_tween_light3.conf  __MACOSX                       OctoPrint          raspian2fipbox_moded.sh  test.py
toni@octo:~ $ cd 1
toni@octo:~/1 $ ls
cpumpg.sh  gpu_avconv  gpuavi.sh  gpubench.sh  gpumkv.sh  gpump4.sh  scripts.zip
toni@octo:~/1 $ cd ..
toni@octo:~ $ /home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG%06d.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"
ERROR: from element /GstPipeline:pipeline0/GstMultiFileSrc:multifilesrc0: Error while reading from file "/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG000000.jpg".
Additional debug info:
gstmultifilesrc.c(467): gst_multi_file_src_create (): /GstPipeline:pipeline0/GstMultiFileSrc:multifilesrc0:
Failed to open file '/tmp/octolapse_renderRxqWdD/new_v09_0.2mm_PETG000000.jpg': No such file or directory
ERROR: pipeline doesn't want to preroll.
toni@octo:~ $

ManuGithubSteam, 2018-10-25:
> @ManuGithubSteam,
>
> Whoops, I just realized that this command uses a temp directory:
>
> ```
> /home/toni/1/gpu_avconv -framerate 10 -loglevel error -i [...]

Ok this is the directory:

toni@octo:~/.octoprint/data/octolapse/snapshots/a30f7df3-8ca2-473f-a901-000f220fefc8/354def78-9eea-409a-ad23-ee966dfff4ba $ ls
metadata.csv                  new_v09_0.2mm_PETG000008.jpg  new_v09_0.2mm_PETG000017.jpg  new_v09_0.2mm_PETG000026.jpg  new_v09_0.2mm_PETG000035.jpg  new_v09_0.2mm_PETG000044.jpg
new_v09_0.2mm_PETG000000.jpg  new_v09_0.2mm_PETG000009.jpg  new_v09_0.2mm_PETG000018.jpg  new_v09_0.2mm_PETG000027.jpg  new_v09_0.2mm_PETG000036.jpg  new_v09_0.2mm_PETG000045.jpg
new_v09_0.2mm_PETG000001.jpg  new_v09_0.2mm_PETG000010.jpg  new_v09_0.2mm_PETG000019.jpg  new_v09_0.2mm_PETG000028.jpg  new_v09_0.2mm_PETG000037.jpg  new_v09_0.2mm_PETG000046.jpg
new_v09_0.2mm_PETG000002.jpg  new_v09_0.2mm_PETG000011.jpg  new_v09_0.2mm_PETG000020.jpg  new_v09_0.2mm_PETG000029.jpg  new_v09_0.2mm_PETG000038.jpg  new_v09_0.2mm_PETG000047.jpg
new_v09_0.2mm_PETG000003.jpg  new_v09_0.2mm_PETG000012.jpg  new_v09_0.2mm_PETG000021.jpg  new_v09_0.2mm_PETG000030.jpg  new_v09_0.2mm_PETG000039.jpg  new_v09_0.2mm_PETG000048.jpg
new_v09_0.2mm_PETG000004.jpg  new_v09_0.2mm_PETG000013.jpg  new_v09_0.2mm_PETG000022.jpg  new_v09_0.2mm_PETG000031.jpg  new_v09_0.2mm_PETG000040.jpg  new_v09_0.2mm_PETG000049.jpg
new_v09_0.2mm_PETG000005.jpg  new_v09_0.2mm_PETG000014.jpg  new_v09_0.2mm_PETG000023.jpg  new_v09_0.2mm_PETG000032.jpg  new_v09_0.2mm_PETG000041.jpg  new_v09_0.2mm_PETG000050.jpg
new_v09_0.2mm_PETG000006.jpg  new_v09_0.2mm_PETG000015.jpg  new_v09_0.2mm_PETG000024.jpg  new_v09_0.2mm_PETG000033.jpg  new_v09_0.2mm_PETG000042.jpg  new_v09_0.2mm_PETG000051.jpg
new_v09_0.2mm_PETG000007.jpg  new_v09_0.2mm_PETG000016.jpg  new_v09_0.2mm_PETG000025.jpg  new_v09_0.2mm_PETG000034.jpg  new_v09_0.2mm_PETG000043.jpg

This is the command i try:

/home/toni/1/gpu_avconv -framerate 10 -loglevel error -i "/home/toni/.octoprint/data/octolapse/snapshots/a30f7df3-8ca2-473f-a901-000f220fefc8/354def78-9eea-409a-ad23-ee966dfff4ba/new_v09_0.2mm_PETG000000.jpg" -threads 1 -r 25 -y -b 125000k -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/toni/.octoprint/data/octolapse/timelapse/new_v09_0.2mm_PETG_20181014185123.mp4"

This does do something but there is no output file yet.

So question:
How do i get that into the octolapse config that i dont need to do that by hand ?

If it works that is. I will report back :)

ManuGithubSteam, 2018-10-26:
Ok the command i used above does indeed work.
I got my timelapse.
However while that timelapse was rendering, i was printing some more and there was no timelapse for that.

So my question:

Where do i have to hack code or script to automathe that process like it should be:
- No manual selection of the to render picture folder
- Autostart of the render after last pic is taken 

On a informative note:
Why is this so complex and not inlcuded in the code already ? Or is it just "me" ? 

Thanks

ManuGithubSteam, 2018-11-01:
@FormerLurker Any news on an update ?
Thanks

FormerLurker, 2018-11-01:
Sorry, I'm out of town so it's been difficult to look at any complex issues.

>Why is this so complex and not inlcuded in the code already ? Or is it just "me" ?

Unfortunately, it is only you as far as I know.  At least nobody else has reported any rendering errors like this before.  Octolapse is supposed to automatically render, but according to the logs I have seen from you there is a permission error (Could not open file "mp4" for writing.).  

So, from your logs I see the path to avconv is /home/toni/1/gpu_avconv.  I've never heard of gpu_avconv, did you install that yourself?  Also, can you try something for me?  Please try to use the built-in timelapse functionality (timelapse tab) and see if that has the same error?  If it does, then we will know for sure that it is a simple permission issue.

Again, sorry about the delay.  I should be back in town by Sunday (hopefully).

ManuGithubSteam, 2018-11-03:
Hello FormerLurker,

No worries i just wanted to know if you still work on it.

I changed the path back to the normal avconv (/usr/bin/avconv) and will retry a print. See how this goes.
GPU_avconv is a patched version of avconv and it should distrubute rendering load also to the gpu of the raspi. 
Yes i tried that out. 

It works with the normal timeplase function without problems (assuming the normal timelapse function does use the path provided in the Octoprint config..)

I will report back how the normal avconv is doing  :-)

ManuGithubSteam, 2018-11-04:
Ok i tired with the normal timelapse and the files get created (build in timelapse function) but they are empty.
I belive this could be beacuse the camera does not point onto the headbed because my cam holder broke and it is filming the same pice of white wall all the time ?

Also i get this errors:

Octolapse Rendering Failed
return-code: Could not render movie, got return code 1: [mpeg4 @ 0x16ef820] bitrate tolerance too small for bitrate
Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

Rendering timelapse failed
Rendering of timelapse holder_peg_pi_v1_0.2mm_PETG_20181104084340 failed with return code 1

More 
[mpeg2video @ 0x161e6e0] bitrate tolerance too small for bitrate
Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

ManuGithubSteam, 2018-11-06:
Ok there is a new kid on the block.
Same setup as before:
Octolapse Startup Failed
Unable to start the timelapse. Cancelling print. Error: An unknown exception occurred while testing the 'F100 Wide' camera profile. Check plugin_octolapse.log for details.

LOG:

2018-11-06 18:52:44,911 Octolapse - loaded and active.
2018-11-06 18:52:48,855 Octolapse - is loading template configurations.
2018-11-06 18:53:06,445 Octolapse - is loading template configurations.
2018-11-06 19:09:47,044 Octolapse - loaded and active.
2018-11-06 19:09:48,191 Octolapse - is loading template configurations.
                                                                        2018-11-06 19:17:56,356 Octolapse - loaded and active.
2018-11-06 19:17:57,498 Octolapse - is loading template configurations.
2018-11-06 19:18:12,919 Octolapse - is loading template configurations.
2018-11-06 19:19:06,562 Traceback (most recent call last):
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_print_start
    camera.test_web_camera(current_camera)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/camera.py", line 54, in test_web_camera
    url, verify=not camera_profile.ignore_ssl_error, timeout=float(timeout_seconds))
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/api.py", line 72, in get
    return request('get', url, params=params, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/api.py", line 58, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/adapters.py", line 508, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x690b5d70>: Failed to establish a new connection: [Errno 111] Connection refused',))

2018-11-06 19:19:06,653 Traceback (most recent call last):
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1299, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 566, in on_gcode_queuing
    self.detect_timelapse_start(command_string, tags)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 719, in detect_timelapse_start
    self.on_print_start(tags)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 556, in on_print_start
    self.OnPrintStartCallback(tags)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 968, in on_print_start
    raise e
ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x690b5d70>: Failed to establish a new connection: [Errno 111] Connection refused',))

2018-11-06 19:19:35,301 Traceback (most recent call last):
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_print_start
    camera.test_web_camera(current_camera)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/camera.py", line 54, in test_web_camera
    url, verify=not camera_profile.ignore_ssl_error, timeout=float(timeout_seconds))
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/api.py", line 72, in get
    return request('get', url, params=params, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/api.py", line 58, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg/requests/adapters.py", line 508, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x690b2ad0>: Failed to establish a new connection: [Errno 111] Connection refused',))

2018-11-06 19:19:35,323 Traceback (most recent call last):
[... cut ...]
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 556, in on_print_start
    self.OnPrintStartCallback(tags)
  File "/home/toni/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 968, in on_print_start
    raise e
ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x690b2ad0>: Failed to establish a new connection: [Errno 111] Connection refused',))

FormerLurker, 2018-11-06:
This looks like a camera URL issue.  Can you send me your camera settings for 'Base Address' and 'Snapshot Address Template'?  

Here is the default: 
Base Address: http://127.0.0.1:8080/
Snapshot Address Template:  {camera_address}?action=snapshot

Also, regarding this comment:
> Ok i tired with the normal timelapse and the files get created (build in timelapse function) but they are empty.

Can you elaborate?  I wonder if there was an error in the OctoPrint logs, or if your suspicion was correct about the camera holder.  Regarding the bitrate error (bitrate tolerance too small for bitrate), can you post your rendering setting for bitrate and fps?

Thanks for sticking with the debug process!  I'm looking forward to getting things working for you... eventually :)

ManuGithubSteam, 2018-11-06:
> This looks like a camera URL issue. Can you send me your camera settings for 'Base Address' and 'Snapshot Address Template'?
> 
Base adress: http://127.0.0.1:8080/ 

Strean url from normal timelapse (I can see a picture there when i press test)

/webcam/?action=stream

Snapschot Adr. Template: {camera_address}?action=snapshot

From normal timeplase function:

http://127.0.0.1:8080/?action=snapshot

Also when i press test in Octolapse it reports, that it is fining a camera and snapshot came back ok.

> Here is the default:
> Base Address: http://127.0.0.1:8080/
> Snapshot Address Template: {camera_address}?action=snapshot
> 
> Also, regarding this comment:
> 
> > Ok i tired with the normal timelapse and the files get created (build in timelapse function) but they are empty.
> 
> Can you elaborate? 
The size is 0 byte.

I wonder if there was an error in the OctoPrint logs, or if your suspicion was correct about the camera holder. Regarding the bitrate error (bitrate tolerance too small for bitrate), can you post your rendering setting for bitrate and fps?
> 
[image]

> Thanks for sticking with the debug process! I'm looking forward to getting things working for you... eventually :)
Thanks i can say the same to you :-)
I really want this to work :-) Normal timelapses are boring.....

FormerLurker, 2018-11-06:
Hmm.. The default is set to 6000K for each rendering profile.  Perhaps your cursor got stuck somewhere and it accidentally got changed?  If not that I'm going to blame file corruption :)

Can you take a snapshot of your Octolapse tab showing the Camera profiles?  I'm wondering if maybe you have more than one camera enabled.  Here's what I'm wanting to see:

[image]

ManuGithubSteam, 2018-11-06:
> Hmm.. The default is set to 6000K for each rendering profile. Perhaps your cursor got stuck somewhere and it accidentally got changed? If not that I'm going to blame file corruption :)
> 
hmm ok.
> Can you take a snapshot of your Octolapse tab showing the Camera profiles? I'm wondering if maybe you have more than one camera enabled. Here's what I'm wanting to see:
> 
[image]

F100 is a copy of default i just changed the name.

> [image]

FormerLurker, 2018-11-06:
Thanks for posting that, but it doesn't illuminate the issue sadly.

Are you maybe running behind a proxy or something?  I've been searching that error, and proxies seem to be a common cause.  I'll keep digging.

ManuGithubSteam, 2018-12-02:
Hi,
Just to inform you: My webcsam has broken down. Till i have a new one and a new octiprint insall i will not be able to report back...

FormerLurker, 2019-10-12:
So, this issue has been stale for a while, and there have been a TON of changes to the camera profiles and settings since our last communication.  I'm going to go ahead and close this issue, but if you'd like me to reopen it, please let me know!  Also, you could easily create another issue too, since the initial problem is different from the last issue we were talking about.  

Thank you for your post!

closed 2019-10-12 by FormerLurker