FormerLurker/Octolapse #258: [Request] Gif Checkbox as adition to mp4 rendering

R0dri, 2018-10-16:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE
I saw an option to export the video as a new feature for the last 5 months. I was wondering if it could output the normal vido mp4 for "HQ sharing" as well as the GIF for a quick preview while sharing. For example I could share the GIF to my friends as soon as it is over and then edit the MP4 and make a cool video for YouTube.... Maybe there would be helpful a button to keep original images as well just in case. This could be done with 2 checkboxes on the interface.

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE
0.3.1

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE
1.3.9 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-10-17:
Hi, thanks for posting!  You can already keep the image files if you want.  Look in the snapshot profile for 'Clean Snapshots After Rendering'.  Depending on what version you are using it might be called something slightly different.

Perhaps when I add a file browser I could make it easier than a check box.  Maybe we could add an 'export video' icon on the file browser that could use avconv or ffmpeg to convert the video to another format?  That would mean we don't need any additional settings at all.  What do you think?

seltix5, 2019-07-29:
Hello,
That wold be a nice option, my current processos is to use a online tool to convert the video into GIF. The option to generate booth the video and the GIF wold be even better =p

FormerLurker, 2019-10-12:
I've correctly categorized this as an enhancement, and will leave it open since it is a pretty cool ides :)

FormerLurker, 2020-05-09:
So, I have added a different feature that may negate the need for this option.  Now you can create an archive of the snapshot images and re-render them as many times as you like.  There is a dialog popup that shows the archives (even allows you to upload/download zip files containing images).  You can render the archive using any rendering profile you want.  This is available in 0.4.0rc1+

Thoughts?