FormerLurker/Octolapse #260: [Request] Disable initial octolapse movement/capture when nozzle Z = 0

madmax2, 2018-10-22:
#### If this is a feature request describe it here

I have the setting stabilization setting
**Fixed Extruder at Center**

I notice that when octoprint starts it normally homes all
Then when you press print the nozzle would make a violent really fast movement towards the center (due to the above setting) before returning to home all position and then begin the Gcode startup script 
and then start the print.

The problem is this violent fast movement should not be done at all when nozzle is at home all position
since Z=0.
I hear bad sound when it does this initial movement.
Mostly likely the nozzle probably scraping the heat bed + my glass bed.
Also the movement is way too fast and should be slower since the nozzle is so close to the bed.

I suggest the stabilization movement should not be done at such an early stage when the nozzle is still at Z=0.
It should be omitted until at least layer 1 etc or when Z is more than 0 to avoid this problem
OR the nozzle should move up by e.g. 3mm before doing this initial movement.
.
The speed of this first movement should also be reduced to protect our nozzle from scratching the heat bed etc..

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-10-23:
Thanks for posting this feature request!

Currently this setting doesn't exist, because for all printers that I've used (a very limited selection indeed), the z-height after a home well above 0 (0.15 for my machine as it is currently configured).  I was not aware that some printers actually touch the bed with the nozzle after a home command.

I could add a setting within the layer/height trigger within snapshot profile that prevents a snapshot for the first N layers.  Would that suffice?

dun3, 2018-12-09:
I would certainly appreciate that feature, not just because of the danger to the heat bed, but much more because I don't want to mess with first (and second) layer adhesion.

FormerLurker, 2018-12-10:
@dun3, in that case I'll take a look at implementation.  Bug me about it if you don't hear back in a couple weeks, thanks!

madmax2, 2018-12-28:
@FormerLurker Yeah it is good to be cautious and consider that not everyone's printers will have the same Z homing distance. There are many different factors that determine how close their nozzle is to the bed including:
-how close people have levelled their bed
-firmware settings
-slicer settings..

Yes, that setting might make it more safer...
Please update this thread if you have implemented this 
so I can test to give you some feedback.

It would also be good to have a setting so that we can control the speed of the extruder
when it makes movement to center or any location that is used for taking the snapshot.

FormerLurker, 2018-12-30:
@madmax2, there is a setting to control the extruder speed within the printer profile settings.  The exact name varies according to the slicer type selection.  For Slic3r it's called 'Travel Speed'.  Octolapse will always use this speed when moving to the snapshot location.  Note that this setting should probably match your slicer settings.

@dun3, good point regarding adhesion.  I hadn't thought about that.  I've personally not had any issues, but can see how some might.

Regarding the bed safety issue, if you need a workaround in the meanwhile, I would bet that a small start gcode modification could prevent scratching both by Octolapse and probably in several other situations.  You could simply raise your nozzle right after the home command(s) to the first layer height (assuming 0.2mm layer height here):
```
; add this line after G28
G0 Z0.2 ; move Z axis to 0.2mm
```
There must be some kind of priming gcode right after your home command.  The 'Layer Trigger' setting won't take a snapshot until it detects extrusion BELOW the 'priming height' which is set within the printer profile.  Your slicer should add a similar G0/G1 command right before the first layer starts to move the Z axis to the proper height for printing the first layer of your object.  If you have problems, post your start gcode and I'll take a look.

FormerLurker, 2019-10-12:
So, I've added support for creating snapshot position restrictions per printer, that should achieve this quite easily.  You can define now a custom bounded box, and Octolapse will only take snapshots when it is within that box.  This could be used to prevent snapshots on the first N layers, or when close to the top of the gantry, or around edges of the bed that are warped and have poor adhesion.  This is available starting with V0.4 (in testing now).  Let me know if you'd like to test this.  I'm going to go ahead and close the issue since I'm doing maintenance today (at long last).  Thanks for posting!!!

closed 2019-10-12 by FormerLurker