FormerLurker/Octolapse #263: Octolapse not taking picture at correct time

shabbirun, 2018-10-30:
#### If this is a feature request describe it here

N/A

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Configured printer settings(including forcing absolute XY) and set extruder to be left, back
2. Enabled Octolapse and started print
3. The stills were not taken at the correct time

#### What should have happened?
The camera should have clicked when the nozzle moved to the left and back, as it did.

#### What happened instead?
Most of the photos were clicked while the nozzle was still on top of the print

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Tevo Tarantula
Printer Firmware Version: Marlin 2.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://files.fm/u/m3gpey7e

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://files.fm/u/xbh36c57

Thank you!

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-10-30:
Hi there, thanks for posting!  This is a known issue that has been fixed in the latest release candidate, which you can install via the plugin manager with this URL:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.zip

Things have changed in this version!  Check out this video for a step by step setup guide: https://www.youtube.com/watch?v=sDyg9lMqMG8

Let me know if that helps!

shabbirun, 2018-10-31:
Hi there - it's working now, thank you so much for the quick reply!

closed 2018-10-31 by shabbirun