FormerLurker/Octolapse #264: Excessive Error Messages when Using Exclude Region Plugin

bradcozine, 2018-11-01:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Installed Octolapse and Exclude Region Plugins.
2. Enabled Octolapse for every print.
3. Began print with several regions excluded.

#### What should have happened?
Print should have continued as normal with no indication of error.

#### What happened instead?
OctoPrint UI flooded with Octolapse Position Error Messages.

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3
Printer Firmware Version: 3.4.1

#### Browser and version of browser, operating system running browser

Browser: Chrome Version 70.0.3538.77 (Official Build) (64-bit)
Browser OS: Linux Mint 18 Cinnamon 64-bit 4.4.0-127-generic

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: All GCODE.

#### Link to settings.json

Link to settings.json with all passwords removed: Will post if requested.

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  Unavailable, will regenerate if requested.

#### Link to octoprint.log

Link to octoprint.log:  Unavailable, will regenerate if requested.

#### Link to contents of Javascript console in the browser

Link to javascript console output: N/A

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imagebin.ca/v/49YO8ej3Hedv

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-11-01:
I think this has been fixed in the latest pre release (0.3.4rc1).  The errors have been rate limited.  Let me know if you need help installing the new version for testing.  I'm traveling and have limited access, but will help however I can

bradcozine, 2018-11-01:
Octolapse Version: 0.3.1
Tried it with a different file. 
Printed several layers with no exclude and no errors.
Added an exclude and immediately got errors.
As soon as the layer changed, the errors stopped. 

I'm going to hold of on going to 0.3.4rc1 until you let me know whether you think this is something that hasn't been addressed yet so I can help you reproduce/troubleshoot. I was a software engineer in a previous life so I could probably ferret out something useful if you point me where you need.

Otherwise, let me know if this all sounds familiar and I'll go ahead and get those new goodies you been cooking up in 0.3.4rc1!

Also, I watched the video, very well presented!! Kept thinking to myself that it would be nice if down the line the configurations could just be parsed from uploaded config.ini files and ultimately with a synced/git'd slicer config folder.

FormerLurker, 2018-11-01:
I'm working on extracting config from the slicers right now!  I am pretty much finished with slic3r, but need to finish the code for cura and simplify.  It will be in v0.4 (I hope).  

Let me try the exclude region plugin myself (it looks awesome) and see if I can replicate the issue.  I am shooting to be back home on Sunday, so probably early next week.  

Thanks for confirming that the issue is easily replicated in 0.3.1.  I'll try with the newest version when I test.

bradcozine, 2018-11-01:
EXCLUDE REGION ENDORSEMENT: If you don't active use the Exclude Region plugin for your normal printing activity, do. It is probably second only on my list to Octolapse. It has saved me from cancelling hundreds of hours on jubs where an object got botched and allowed me to reuse gcode where I only needed to reprint a portion. I'm sure others will back me up on this and someone would be hard pressed to name a downside.

bradcozine, 2018-11-01:
Also, when you check add a use case for when the print head moves through an newly excluded region and one where it doesn't move through a newly created region.

bradcozine, 2018-11-02:
Out of curiosity, has disabling the error message altogether been considered since there is little to nothing the user would be capable of doing to address it anyway? Or is there a use case where the error is of value? (A few word response is fine, just floating it out there. Don't need to take up time on this.)

FormerLurker, 2019-10-12:
So it has been a LONG time since I last posted in this issue.  There is another issue in this repo dealing with exclude region, and I've not gotten around to testing it with the most recent version.  However, I've done a LOT to the error message handling (mostly rate limiting/eliminating entirely), so this is likely mitigated somewhat.   I'm going to leave this open since irritating error messages need to be removed, and I haven't verified 100% that this is the case.

FormerLurker, 2020-05-09:
Those error messages are GONE!  finally..  closed :)

closed 2020-05-09 by FormerLurker