FormerLurker/Octolapse #265: Feature Request: Switch on lights before taking a photo

dushyantahuja, 2018-11-01:
Hi,

I'm looking to setup some lights on my printer. I can easily connect the lights using a pin on the controller (that can be controlled using gcodes), or a pin on the raspi (3B+). These lights can be switched on / off using existing plugins on Octoprint.

However, I want the lights to switch on automatically a couple of seconds before a photo is taken and then switched off after a couple of seconds. Want to create good looking timelapses, but don't want the lights to be on all the time.

Any suggestions would be great. Thanks.

FormerLurker, 2018-11-01:
You might want to check out the latest pre-release version (0.3.4rc1), which contains the ability to execute custom scripts at various points.  Specifically you'll probably need to use the before and after snapshot scripts (see [this](https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-camera-scripts) wiki page).

You can install that version via the plugin manager using this URL:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.zip

Let me know if that helps, and good luck!

dushyantahuja, 2018-11-03:
Looks like a great idea. Will try and let you know if I see any issues. Thanks.

closed 2018-11-03 by dushyantahuja