FormerLurker/Octolapse #266: Frame Width & Height Settings? & Where are the raw image files?

ZhuDaHai, 2018-11-04:
#### If this is a feature request describe it here
How to set the Video Width and Height? 
It default to 640x480 and looks terrible. 
I also set a framerate of 30fps, but it rendered at 25
And where can I find the raw jpg image files so I can render it myself?

#### Version of Octolapse
 
Octolapse Version: v0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

FormerLurker, 2018-11-04:
The resolution of your timelapse will be the same resolution of your camera stream.  If you are using OctoPi, you'll need to edit octopi.txt.  Check out [this](https://discourse.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203) guide.

Not sure what's going on with the FPS.  Can you send me a link to your timelapse, and post your settings.json file?  You can download it from Octoprint Settings->Octolapse.  There is a button there to download the file.

Thanks for posting!

ZhuDaHai, 2018-11-05:
Here's a screen capture of the timelapse windows properties showing 25fps. I'll upload the file to my Google Drive if you still need to view it.
And attached is the Settings.json file
I found the setting for keeping the individual jpg images, unfortunatly it was after I did this timelapse, so they were all deleted. So I turned that feature off.
Thanks!
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/2546665/Settings.zip)

[image]

FormerLurker, 2018-11-06:
Please post the video (not on youtube, which will re-encode) so I can take a look.  Also, were you able to change your resolution?

ZhuDaHai, 2018-11-07:
I believe I was able to change the resolution. I did a quick snapshot using the URL and it came back at the correct 4K resolution. Thanks!
Here's a link to the video, but I put it on my Google Drive and it says 'We are processing this video, plesae check back later'. Hope that doesn't mean they're reformatting it.
: https://drive.google.com/file/d/1WUe4f1Ef5gRDS6bh3BIjDJ9DtXti3_w8/view?usp=sharing

FormerLurker, 2019-10-12:
So, I fixed the framerate issue.  Turns out it was a hard to spot error in the avconv/ffmpeg commands.  This fix will be included starting with V0.4.  I think all of the issues talked about in this thread are now fixed!  Let me know if you want me to reopen this, or feel free to post a new issue if you are having other issues.  Thanks!

closed 2019-10-12 by FormerLurker