FormerLurker/Octolapse #267: Causes print to fail to start with PRUSA MK3 with MMU2

MStylesMS, 2018-11-04:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

Attempted to print a single color object using extruder #5.  

#### What should have happened?
It should have printed using the filament in tool #5.

#### What happened instead?
It got through the initialization process of bed leveling and going home, but when it was supposed to load the filament from #5 it attempted to load the filament from #2.  

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: ???

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3 with MMU2
Printer Firmware Version: 3.4.1

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-11-06:
@MStylesMS, check out [this link](https://discourse.octoprint.org/t/prusa-mmu-t-not-working-via-octoprint/4726).  It looks like something is going on with the T ...  pseudogcode.  In the meanwhile I'm going to mark this as 'under observation'.

szafran81, 2019-01-01:
I can somewhat confirm that.
My setup:
MK3 FW: 3.5.1
MMU2 FW: 1.0.3
OctoPrint: 1.3.10
Octolapse: 0.3.4
Slic3r PE: 1.41.2
Cameras: 3 x c930e & 1 x c920 (currently only one of those is configured in OL)

OctoPrint running on an x86 Ubuntu MATE v18... but this shouldn't have anything to do with it.

Now... I'll try to describe what's going on on my config.
When the cameras are connected and OL is enabled everything is fine - the prints start and go on their way as they should.
When the cameras are connected and OL is disabled everything is also fine.
When the cameras are disconnected and OL is disabled everything is also fine.
When the cameras are disconnected and OL is enabled then the things go wrong. I'm getting a popup error from OL that the camera is not present (tried disabling the camera and everything goes the same but only the text in the popup is different), and the print doesn't start. This is what I'm getting in terminal:

`Changing monitoring state from "Operational" to "Printing"
Changing monitoring state from "Printing" to "Cancelling"
Send: N6 M108*44
Send: N7 M104 S0*98
Recv: Unknown M code: $6 M108
Recv: ok
Send: N0 M110 N0*125
Send: N1 M140 S0*100
Recv: ok
Send: N2 M107*39
Recv: ok
Recv: ok
Recv: ok
Send: N3 G0 Z0.2 E-1.2 F9000*118
Send: N4 G0 X0. Y120. F9000*64
Recv: echo: cold extrusion prevented
Recv: ok
Recv: ok
Send: N5 M84*26
Changing monitoring state from "Cancelling" to "Operational"`

This is the start g-code generated by the Slic3r PE for MMU Single profile:
`M73 P0 R301
M73 Q0 S303
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M107
M115 U3.5.0 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S255 ; set extruder temp
M140 S100 ; set bed temp
Tx
M190 S100 ; wait for bed temp
M109 S255 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

G21 ; set units to millimeters

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
Tc
; purge line
G1 X55.0 E8.0 F2000.0
M73 Q0 S302
M73 P0 R301
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0  F2200.0
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
G1 X5.0 E4.0 F1000.0

M221 S95
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G92 E0.0
M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion`

NOTE: There is no M108 in there (actually there is no M108 anywhere in the g-code file).

From what it seems to me OL is not forwarding the Tx (runs function which shows a menu to select which filament to use on the LCD) & Tc (physically selects the chosen filament with the MMU2 selector and loads the filament up to extruders bondtech gears) codes properly when it's enabled and the cameras are disconnected.

FormerLurker, 2019-01-02:
@szafran81, I'm repeating and elaborating on this issue from the [OctoPrint discourse forums](https://discourse.octoprint.org/t/prusa-mmu-t-not-working-via-octoprint/4726/62?u=formerlurker) for anyone who is following along.

Tx is working for me just fine, and previously with MMU1.0 I was using T?.  I think the camera initialization may indeed be your issue. Octolapse will cancel your print by default if no camera can be detected (can be disabled within the Octolapse main settings at OctoPrint Settings (wrench/spanner icon)->Octolapse->Main tab-> Cancel if startup errors are detected).  Not sure why you are getting cold extrusion errors, except that maybe there is some issue within the cancel script that is allowing some gcode to get to your printer that should have been stopped by the cancel action (I'll review this sometime this week).

Octolapse cancels these prints by default because it wouldn't be able to create a timelapse, and that really confuses new users when they don't end up with a timelapse at the end of their print. Once the user knows what's going on, it's a simple matter to disable the option.

Let me know if that helps.

szafran81, 2019-01-02:
I've disabled that option. Will give it a try after my current printer finishes.
That still leaves the cold extrision and M108 problem - I've searched and can't find anything that will point me to what injects this unsupported gcode.

FormerLurker, 2019-10-12:
So, Octolapse now correctly handles all T commands in both the MMU1 and MMU2 styles starting with V0.4 (in testing now).  I'm calling this one closed!  Let me know if anyone wants to test, or to reopen this issue.  Thanks!

closed 2019-10-12 by FormerLurker