FormerLurker/Octolapse #268: Clean Snapshots After Rendering Enabled not saving

ZhuDaHai, 2018-11-05:
Diagnostic Logging was Enabled: NO

1. Snapshot Settting
2. Advanced Options
3. Disable 'Clean Snapshots After Rendering'
4. Save
5. Close window
6. Snapshot Setting
7. Advanced Options
8. 'Clean Snapshots After Rendering' still enabled
9. Say 'Grrrr'

#### What should have happened?
'Clean Snapshots After Rendering' should be unchecked

#### What happened instead?
'Clean Snapshots After Rendering' was still checked
Said 'Grrrr'

OS Name: OctoPi
Os Version: OctoPrint 1.3.9 running on OctoPi 0.14.0
Octolapse Version: v0.3.1

FormerLurker, 2018-11-05:
This issue has been fixed in the latest pre-release, sorry about that.

ZhuDaHai, 2018-11-05:
I look forward to that update. Thanks!

closed 2018-11-05 by ZhuDaHai

FormerLurker, 2018-11-05:
No problem!  Let me know if you want to try it out and I'll send instructions for installation.