FormerLurker/Octolapse #269: "After Snapshot Script" image processing with imagemagic causes Camera error

Maelstrom2014, 2018-11-08:
"After Snapshot Script" image processing with imagemagic causes Camera error. Can I process It in backround wthout increasing snapshot time? The next layer printing time is enough to process one last snapshot.
Thanx

FormerLurker, 2018-11-08:
@Maelstrom2014, what host OS are you using for Octoprint?  Essentially you will need to spawn a new process within the 'After Snapshot Script' to process the image, then immediately exit the script (without throwing an error).  This process differs from OS to OS.  For windows I believe you use the 'Start' command.  I'd have to do some research to figure this out for linux and macOS.  Let me know if you get stuck.

You could also use the pre-render script for this, though the script will require some looping.  No need to worry about how long that script takes.  I believe it will happily run forever, while the before/after snapshot scripts have a timeout to prevent the script from stalling the print for an inappropriate length of time.

Maelstrom2014, 2018-11-09:
I'm using Octopi, linux. So I need to add "&" the start of the line in script to process it in backgroud and rest camera time to 125 ms?
like this? 
`& sudo convert -normalize -auto-gamma -auto-level -sigmoidal-contrast 3,50% -sharpen 2.5 "${SNAPSHOT_FULL_PATH}" "${SNAPSHOT_FULL_PATH}"
`

FormerLurker, 2018-11-09:
Looks pretty good.  Try it out with a small test print.  Test mode (debug profile in octolapse) is perfect for this.

FormerLurker, 2018-11-09:
Oh, I'm assuming that convert is is configured to run without a password when using sudo, right?

Maelstrom2014, 2018-11-11:
So we do not need 'sudo' or any other sudousers permission changes. everything works ok without sudo.
Thanx.

closed 2018-11-11 by Maelstrom2014

FormerLurker, 2018-11-11:
@Maelstrom2014, FYI, I did some testing regarding scripts with child processes, and it looks like my current code waits until all child processes are finished, which will cause some delays when running your image processing script.  I think you will need to use the 'pre render' script for this.  I wrote one out, but did NOT test it since I'm in the middle of a test print at the moment.  Let me know if this makes sense:
```
#!/bin/sh
# Camera Pre-Render 'Convert' script
# Written by: Formerlurker@pm.me

# Put the arguments sent by Octolapse into variables for easy use
CAMERA_NAME=$1
SNAPSHOT_DIRECTORY=$2
SNAPSHOT_FILENAME_TEMPLATE=$3
SNAPSHOT_FULL_PATH_TEMPLATE=$4

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "The snapshot directory does not exist: '${SNAPSHOT_DIRECTORY}'." >&2
  exit 1
fi

# Loop through each image, constructing each from the snapshot_full_path_template
a=0
while :
do
  CURRENT_IMAGE=$(printf "${SNAPSHOT_FULL_PATH_TEMPLATE}" "${a}")
  if [ ! -d "${CURRENT_IMAGE}" ]; 
  then
    # The current image does not exist, we are done!
    break
  fi
  convert -normalize -auto-gamma -auto-level -sigmoidal-contrast 3,50% -sharpen 2.5 "${CURRENT_IMAGE}" "${CURRENT_IMAGE}"
  a=$((a+1))
done
```

Hopefully I didn't goof :)