FormerLurker/Octolapse #270: Camera Settings After snapshot Script Error:  The process does not exist

Maelstrom2014, 2018-11-09:
works but not, something changed and it does not work :(

Octolapse - Camera Settings Error
There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: The process does not exist

Octolapse v0.3.4rc1.dev3
OctoPrint 1.3.9 running on OctoPi 0.14.0

tried, but no luck :
1) "sudo" before command.

FormerLurker, 2018-11-11:
Can you please send me two things:

1.  The full bash script you are trying to execute.
2.  Your settings.json file.  [Here are instructions for doing this](https://github.com/FormerLurker/Octolapse/wiki/Settings#export-your-settings).

If I had to guess I'd say you put the snapshot script in the wrong place.  That error message is saying the initialization script is failing, which only runs ONCE before your print starts.  You probably need to adjust your camera profile by setting your 'Camera Type' dropdown box to 'External Camera - Script'.  Then put the script path in the 'Snapshot Acquire Script' text box that appears.  Remove the script path from the 'Before Print Start Script' box if anything is there and try again.

Also, you will need sudo probably.  Have you watched [this video](https://youtu.be/J5SskYwP5JQ) explaining external DSLR setup?  If not, please do!

Thanks for posting!

Maelstrom2014, 2018-11-11:
I found the answer!
If script MUST contain first line this
`#!/bin/sh`
or this
`#!/bin/bash`

closed 2018-11-11 by Maelstrom2014

FormerLurker, 2018-11-11:
Oh, yeah, I should have mentioned that.  Thanks for letting me know things are working for you!  Happy Octolapsing ;)