FormerLurker/Octolapse #272: Octolapse error on print start - Missing G90/91

tech-rat, 2018-11-12:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.3.4rc1

#### Version of OctoPrint

OctoPrint Version: 1.3.10rc1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Upgraded octoprint to 1.3.10rc1
2. Printed like usual
3. Got octolapse error: 
[image]
4. Octolapse works as expected after the error. Just have to re-enable it before each print.

#### What should have happened?
No error message and the plugin would stay enabled

#### What happened instead?
Warning message

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian GNU/Linux 8
Os Version: Raspbian GNU/Linux 8

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3
Printer Firmware Version: 3.4.1-1356

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE 
[Object.zip](https://github.com/FormerLurker/Octolapse/files/2573421/Object.zip)

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-11-12:
HI!  To fix your problem change this start gcode:
```
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.2.1 ; tell printer latest fw version
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M83  ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S95
M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion

```

to this:

```
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.2.1 ; tell printer latest fw version
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83  ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S95
M900 K30; Filament gcode
```

Notice that I removed 1 redundant M83 command and moved the G90 and G21 BEFORE any G1 or G0 commands.

I'm REALLY working hard to eliminate the need to adjust start gcode to get Octolapse working, so I appreciate your willingness to submit an issue.  Please let me know if this change works for you.  If you continue to have trouble I'll be here to help!

tech-rat, 2018-11-12:
I just upload directly from Slic3r to Octoprint.
I never modify the GCODE by hand.
Where in Slic3r would I make those changes?

Thanks
Tech-Rat

tech-rat, 2018-11-14:
I am still really new to GCODE so I want to understand what I'm changing.

My current top of any GCODE file looks like this:
M115 U3.4.0 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
M73 Q0 S74
M73 P0 R74
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S95
M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

Am I correct that because its printing the intro line before issuing the G21 and G90 commands that is whats causing the Octolapse error? And by adding the G21/G90 commands as part of the Start G-Code I will end up with two of those commands in the top of my gcode output but the new set will be before the intro line.

Thanks
Tech-Rat

FormerLurker, 2018-11-15:
Your suspicion is correct.  Adding the G90 to the top makes it obvious to Octolapse where your print head is moving when you print those priming lines.  Here's an alternative to the above fix:

Go into Octolapse printer profile settings and change 'X/Y/Z Axis Mode' to 'Default to Absolute'.  Now Octolapse will assume that your printer is in Absolute mode when the print begins.  Your gcode assumes this is true (even though manually sending a G91 before a print will break this assumption), so it's not an additional risk.  In fact I'm considering setting this as the default for some printer profiles.  My intent with the current setting (don't assume anything about the axis mode) was to prevent Octolapse from making an out-of-bounds movement in all cases.  It appears as though most default start gcode isn't nearly as careful or strict as Octolapse.

FYI, I have requested that a feature be added to Marlin, and will soon request this of other firmware folks, to add the ability to query the current axis mode from the printer.  If this were true none of this would be necessary.  In your case the error message goes away because at some point Octolapse sees the G90, then it updates its position based on the next G0/G1 command it sees.  The error message is really telling you that your start gcode is ambiguous, and depending on what happened before you start your print (like sending a G91 for some reason), the priming line code may not do what it should.

tech-rat, 2018-11-15:
I set that mode and the error went away.

Thanks for all your detailed explanation and the great plugin.

Tech-Rat

closed 2018-11-15 by tech-rat

FormerLurker, 2018-11-15:
Anytime!  I was happy to see you on the octoprint discourse site, btw!  It's great how our little community interacts so often.