FormerLurker/Octolapse #275: Error when opening mjpegstreamer control.htm

BastR, 2018-11-18:
#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

I updated the plugin and tried to configure this new version as best I could. I have read the wiki and the paragraph to enable mjpegstreamer control page changing: 

_#camera_http_webroot="./www-octopi"_
_#camera_http_options="-n"_
to this:
_camera_http_webroot="./www"_
_camera_http_options=""_

but when I open "http://my_octopi_adress:8080/control.htm" I have no options available just "(null) HTTP output plugin" . As a result, I can't enable the custom settings 

#### What should have happened?
Enable Custom Image Preferences

#### What happened instead?
Got and error 

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi

#### Printer model & used firmware incl. version

Printer Model: Homemade
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Browser: Last Safari
Browser OS: macOs

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

https://drive.google.com/open?id=1OH60LI5hnIoFPRF0wbZeGyu2PW6ymTYf

https://drive.google.com/file/d/1MPDaQ0d9mxQxx0dgGsPC-kI0ZAQF51LV/view?usp=sharing

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2018-11-18:
@BastR, just to be certain, you rebooted your  pi after making that change, correct?  I've never seen control.htm look like that before, though, so thanks a bunch for reporting it!  Can you maybe try this address and see if anything different happens:

http://my_octopi_adress/webcam/control.htm

Is your instance of Octoprint running behind a proxy by chance?  I'd like to replicate your setup to see if I can see the error you are witnessing myself if possible.  That would be the easiest way for me to get to the bottom of this.

Thanks again for your submission!

FormerLurker, 2018-11-19:
Can you post your boot/octopi.txt settings (remove anything sensitive in there before posting please)?  Also, can you let me know what version of OctoPi you are using?  Your camera model may help, since the custom settings aren't available when using all cameras (though I have never seen the control.htm screen look like that screenshot you posted).  Lastly, are you running any proxy software?

BastR, 2018-11-19:
I have a RPI camera V2, my boot/octopi.txt is here [https://drive.google.com/open?id=1Zhy7HZsyoiorQY_h8IpiVNKDl7Tp29U7](url), I'm not running a proxy software :(

OctoPrint version : 1.3.9
OctoPi version    : 0.15.1

FormerLurker, 2018-11-19:
@BastR, I'm not sure if the raspberry pi camera supports the control.htm page unfortunately.  I don't have an RPI camera right now, but I'll look into getting one so I can test.

You might want to post on the OctoPrint discourse page at https://discourse.octoprint.org/

It's likely that someone there knows if control.htm should work for the raspi cam.

BastR, 2018-11-20:
It's a pity, the rpi camera is not expensive and I find the quality acceptable.  The negative point in my opinion is to have to modify the file /boot/octopi.txt and systematically restart the rpi to change the camera settings... it's very annoying.

Maybe I'm not doing it right.

I’ll post on https://discourse.octoprint.org/.

Thanks you

FormerLurker, 2018-11-22:
Thank you so much for letting me know!  Now I can advise future users.  Good luck and happy holidays!

ZhuXuanyi, 2019-06-22:
@FormerLurker add bcm2835-v4l2 to the end of file /etc/modules works for me. 
Thanks to this post (https://forum.astroprint.com/t/raspberry-pi-camera-not-working/2691/5)

FormerLurker, 2019-10-12:
So, I think all of these issues have been solved, so I'm going to go ahead and close this issue.  I did want to mention that the camera controls have been GREATLY enhanced in V0.4 (currently in testing), so keep your eyes peeled.  Let me know if I should reopen this issue, or feel free to create a new one if anyone has other related issues.  Thanks!

closed 2019-10-12 by FormerLurker