FormerLurker/Octolapse #276: [Request]

Jnesselr, 2018-11-19:
#### If this is a feature request describe it here

I believe this is a request, but it might just be me not understanding octolapse. I have an existing octoprint printer profile and a bunch of slicer profiles already set up for my hackerspace. A few members were asking about Octolapse, so I tried to install it but it would appear I have to completely add a new profile as opposed to pointing it to the existing profiles. Can we get a feature to just import the existing printer profile. I see #243 exists for importing the slicer profiles.

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.9

FormerLurker, 2018-11-19:
@Jnesselr, many of the print profile parameters can be used from the current default OctoPrint profile, like the build volume.  The rest of the printer profile settings require some info from the slicer, which varies pretty dramatically from printer to printer.  It also requires updating if the slicer values change, unfortunately.

I am working on a feature that will pull settings directly from the gcode file if they are available (supported in Slic3r and Simplify3d, but not cura unfortunately).  I plan to make this available in the next release.

Thanks for your post, and let me know if you have other questions!

FormerLurker, 2019-10-12:
So, I think this issue can be closed (let me know if I am mistaken).  Octolapse V4 can now extract settings directly from your gcode file for Simplify 3d and Slicer, and will even work with Cura after adding a small script to your start/end gcode!  Thanks for your post!

closed 2019-10-12 by FormerLurker