FormerLurker/Octolapse #278: Double Purge

z1nfinity, 2018-11-23:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: 0.3.4

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Input Cura Slicer settings into the Printer Profile.
2. Loaded g-code to Octoprint
3. Started print.

#### What should have happened?
When initially starting a print the default start code provided by Prusa for their Cura profile should only produce a single purge/prime line then continue to print. 

#### What happened instead?
When starting my print using Octolapse it will start the purge line, print half of it, move to the center of the bed, then go back to the purge line and finish the other half, then go back to the start point of the print and begin printing.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3
Printer Firmware Version: 3.4.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: [](https://www.reddit.com/r/3Dprinting/comments/9zjbec/octolapse_purge_issue/)

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2018-11-23:
This is strange, but not unexpected behavior.  Can you check what layer height your prime line is printing at?  If it is greater than your layer height, there is a setting that can be used to prevent this.  If you include your start gcode, I'll take a look.  Unfortunately I do not have an mk3 to test :(

z1nfinity, 2018-11-23:
I'm not familiar enough with 3D printer g-code yet, but I can post the start code used in Cura.

> ;FLAVOR:Marlin
;TIME:16446
;Filament used: 2.03734m
;Layer height: 0.08
;Generated with Cura_SteamEngine 3.6.0
M82 ;absolute extrusion mode
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S215 ; set extruder temp
M140 S60.0 ; set bed temp
M190 S60.0 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G92 E0.0 ; reset extruder distance position
G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
G92 E0
G1 F6000 E-0.8
;LAYER_COUNT:518
;LAYER:0
M107
G0 F1500 X113.297 Y92.39 Z0.15
;TYPE:SKIRT
G1 F6000 E0
G1 F750 X113.838 Y91.804 E0.01532
G1 X114.402 Y91.252 E0.03048
G1 X115.247 Y90.602 E0.05095
G1 X115.591 Y90.373 E0.05889
G1 X116.269 Y89.969 E0.07405
G1 X116.985 Y89.636 E0.08922

z1nfinity, 2018-11-24:
Here are my Cura settings.
https://imgur.com/Fmj8va8

MisterGlass, 2018-12-27:
Having a similar issue as well. I actually get several, between each step of the initial purge (bubble, line, bigger line) Here is the relevant portion of my starting gcode (from the [community pretty PLA profile](https://github.com/eoprede/prusa_profiles/blob/master/MK3/Slic3r/Pretty%20PLA%20V3.ini))
```
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 E8 ; Purge Bubble
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
```
@FormerLurker you mentioned a setting that might help? Is there a way to disable for the first few layers?

FormerLurker, 2018-12-27:
Hey!  Im working on a setting to prevent snapshots on the first X layers.  Will continue after the holidays.  Hound me if you dont hear back please!  I have quite the backlog and dont want this to get lost.  Thanks!

MisterGlass, 2018-12-27:
@FormerLurker Thank you! Enjoy your holiday!

FormerLurker, 2019-10-12:
I'm going to close this issue since it's been idle for a long while.  Please feel free to post a new issue if you are still seeing this problem.  Thanks!

closed 2019-10-12 by FormerLurker