FormerLurker/Octolapse #279: Octolapse - Camera Error ERROR: Could not trigger image capture. ERROR: Could not capture.

3dnewb, 2018-11-26:
#### If this is a feature request describe it here

I followed the installation tutorial found here: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

This is not a feature request:

I have no issue running the script from SSH, the output file is saved 
as seen here: pi@octopi:~ $ /home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"
New file is in location /capt0000.jpg on the camera
File /home/pi/scripts/test.jpg exists. Overwrite? [y|n] y
Saving file as /home/pi/scripts/test.jpg
Deleting file /capt0000.jpg on the camera
Deleting 'capt0000.jpg' from folder '/'...

When I try to run a print in octolapse I receive the following error: 

Octolapse - Camera Error
snapshot_script_error: The snapshot script failed with the following error message: ERROR: Could not trigger image capture.
ERROR: Could not capture.
The snapshot was not found in the expected directory: '/home/pi/.octoprint/data/octolapse/snapshots/a622bd1b-770c-40c0-b279-c924cbba9039/d6e71adc-f819-4b45-8666-59505e2e5852/PI3MK2_SmartTemperatureTower_PETG_200-230000000.jpg'

I can hear the camera "click" and the display shows a message stating connecting to PC

#### Version of Octolapse
 
Octolapse Version: Octolapse v0.3.4

#### Version of OctoPrint 

Profile | Name |  
-- | -- | --
Printer |   |  
Stabilization |   |  
Snapshot |   |  
Rendering |   |  
Cameras | Default Webcam                                                                                                                                                                            DSLR
Debug |   |  

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

Profile | Name |  
-- | -- | --
Printer |   |  
Stabilization |   |  
Snapshot |   |  
Rendering |   |  
Cameras | Default Webcam                                                                                                                                                                            DSLR
Debug |   |  

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO YES

#### What were you doing when the problem occurred

1. Started a print with Full Diagnostic - Test Mode
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
A time lapse start

#### What happened instead?
Received the following Error:
Octolapse - Camera Error
snapshot_script_error: The snapshot script failed with the following error message: ERROR: Could not trigger image capture.
ERROR: Could not capture.
The snapshot was not found in the expected directory: '/home/pi/.octoprint/data/octolapse/snapshots/a622bd1b-770c-40c0-b279-c924cbba9039/d6e71adc-f819-4b45-8666-59505e2e5852/PI3MK2_SmartTemperatureTower_PETG_200-230000000.jpg'.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://github.com/3dnewb/Octolapse-Troubleshooting/blob/master/PI3MK2_SmartTemperatureTower_PETG_200-230.gcode

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://github.com/3dnewb/Octolapse-Troubleshooting/blob/master/octoprint(1).log

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://github.com/3dnewb/Octolapse-Troubleshooting/blob/master/Settings.json

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2018-11-27:
Hi, thanks for submitting your issue!  Can you post your script?  I'm guessing it's probably the same as the example script from the tutorial, but it never hurts to check.

3dnewb, 2018-11-29:
Sorry for the delay!

Yes, I am using the script straight from the tutorial.

FormerLurker, 2018-11-29:
Can you turn on full diagnostic test mode, print a few layers with your DSLR attached, then download plugin_octolapse.log and send me a copy please?  Hopefully there is some info in there. 

Also, turn your snapshot timeout way up, say 30 seconds, just to rule that out.

3dnewb, 2018-11-29:
Sure thing, I will do it later tonight

FormerLurker, 2018-11-29:
Fyi, the snapshot timeout is in the camera profile.  Lastly, did you watch the YouTube tutorial on getting DSLR working (just curious)?

FormerLurker, 2019-10-12:
So, this issue has been idle for a while.  Before I close it I wanted to let you know that the 3 DSLR tutorials have been enhanced over the last several months, so you might want to give them a look if you are still having problems.  Feel free to create a new issue if you are still having problems, and thanks for your post!

closed 2019-10-12 by FormerLurker

ItsTalhaKhan, 2021-08-09:
> #### If this is a feature request describe it here
> I followed the installation tutorial found here: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera
>
> This is not a [...]

Hey @FormerLurker , 
I'm also facing the same error.
The script works good on SSH. 

But this error appears in the log when testing the script "ERROR - Error output (stderr) for 'New Camera433 - Snapshot Camera Script': ERROR: Could not get image."

The camera appears in auto detect. 
Sudo is bypassed for gphoto2. 
The script runs from octolapse. (Tested)

But I guess it doesnt run "sudo gphoto2 --capture-image-and-download --filename  "${SNAPSHOT_FULL_PATH}"

ItsTalhaKhan, 2021-08-09:
Its working now. 
gave permission to the script and pi files.

sudo chmod +x /home/pi
sudo chmod +x scripts

And its working beautifully.

Btw, @FormerLurker great work bro.
Keep it up.