FormerLurker/Octolapse #281: Access Forbidden

Xlaits, 2018-11-27:
#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Install OctoLapse
2. Change Printer Profile settings
3. Save

#### What should have happened?
Printer profile should have saved.

#### What happened instead?
Error: Forbidden dialogue

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian
Os Version: Stretch

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: Unknown

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: None

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log: https://hastebin.com/ovokupotim.php

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://hastebin.com/iruxohumuh.php

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]
[image]
[image]
[image]

Xlaits, 2018-11-27:
I apologize now for some of the missing info, as I am working on getting that solely from an SSH session.

FormerLurker, 2018-11-27:
@Xlaits, this is a new one, thanks for reporting it!  Have you tried rebooting the pi to see if that fixes things?  Also, have you added any security to your OctoPi (proxy software, etc.)?

If none of the above applies, perhaps you can turn on the developer console (F12 for firefox) then go to the console and try editing your printer profile.  Any errors in the console?  If not, switch to the network tab, reload the page (f5), and see if there are any error there.  Post anything you can see that looks odd.

Xlaits, 2018-11-27:
[Firefox Console Output](https://hastebin.com/muyoqasoxi.sql)

So, I also can't log out, yet I have to log back in each time I reload.

FormerLurker, 2018-11-28:
So this is a fresh install of octolapse?  Something seems to be going wrong pulling the profile data down.  Can you try restoring the default settings in Octoprint settings->Octolapse), then restart Octoprint.

Also, there are lots of 404 and 403 errors in your log file, most aren't related to Octolapse.

You might try uninstalling octolapse and seeing if things start working again.

I will have more time this weekend to look more into this, but will spend a bit more time with the log file tomorrow after work.

Xlaits, 2018-11-28:
Uninstalling OctoLapse solves all of my problems, until I install it again.

Using the "restore defaults" doesn't work, either. It literally just sits there, doing nothing.

FormerLurker, 2018-11-28:
Ok, thanks for letting me know.  Can you try disabling all your other plugins temporarily, except octolapse?  Perhaps it is a conflict of some type with a particular plug-in.

FormerLurker, 2018-11-28:
@Xlaits, I just looked a bit more at your logs, and it looks like you're being attacked.  Are you using port forwarding to access your pi over the internet?  If you, there are MUCH better and SAFER alternatives you should look into.  It's just a matter of time before your system is compromised, if it isn't already.  If I were you I'd start over with a clean install (I know that is a painful option for some), and discontinue port forwarding immediately.  Who knows what could be going on there.

Also, are you running TouchUI by chance?  If so, try uninstalling that plugin temporarily, then see if Octolapse still gives you fits.

FormerLurker, 2018-11-28:
Let me know how big the disk image is, and let me know if you'd be willing to send me a copy somehow.  If so, maybe I can restore it on my rpi.  That would (hopefully) let me see this error in action.  I'd really like to figure out what in the heck is going on.

Also, as an aside, have you considered opening up ONLY the mjpegstream to the internet?  This would be much safer, and would let you embed the image in a webpage somewhere.

Xlaits, 2018-11-29:
It's about 8GB, and I have no problems sending you a copy, however I would need to make you an admin account before sending it to you.

Other than that, No I haven't, actually. I have no idea how I'd go about that.

FormerLurker, 2018-11-29:
Cool!  I'll look into ways of only allowing external access to the cam stream.  I'd like to do that myself.  I currently use VPN.

Xlaits, 2018-11-29:
So, with a fresh install of OctoPi and Octolapse, I have _no_ issues.

Xlaits, 2018-11-29:
https://mega.nz/#!NZJV0CSI!msfQo3w0ixwt9cigLwrsBzuVtj_gQsfSdH_vuZ1FJTs

FormerLurker, 2019-10-12:
So this issue has been idle for a while, and I think the original problem has been solved. I never got around to looking at your image... So sorry about that, I was really sick when you posted it and got buried in work after.  I know that took a lot of effort, and I really appreciate it.

Anyway, I"m going to go ahead and close this issue now.  Thank you so much for your post, and feel free to create another issue if you ever have more problems, questions, or suggestions!

closed 2019-10-12 by FormerLurker