FormerLurker/Octolapse #282: External camera script triggering twice

KalebClark, 2018-11-28:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Setup Octolapse for external script
2. Configured per Octolapse wiki instructions
3. Put Octolapse in full diagnostic test mode.
4. Ran test print

#### What should have happened?
Run through test print, octolapse capture images and render.

#### What happened instead?
Ocotlapse triggers the camera no problem. But it gives me the IO error claiming that gphoto2 is already running, or one of its other components (I could not get that message back for this report, but it does say "Camera Error" every time it takes a snapshot). While the print was running, I ssh'd into the raspi and did a ps -eaf | grep gphoto and I see that two of the take-snapshot.sh scripts are running at the same time. If I try to run two at the same time, I get the same error as reported by octolapse.

#### Operating System running OctoPrint and Octolapse

OS Name: raspbien
Os Version: Jesse

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/KalebClark/2c5833d18c4e86b38f31fcebc988de75

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

KalebClark, 2018-11-28:
Sorry there is no log. I have full diagnostic test mode on, but it did not generate the log file.

FormerLurker, 2018-11-28:
Double check for the log file (Octoprint settings->logs->plugin_octolapse.log) please.  The logger is probably the best tested feature (lol), but if you still don't see it, I think that's the first problem that needs to be solved.

Maybe take a look in the settings for the debug profile.  Maybe logging got disabled somehow.

KalebClark, 2018-11-28:
Got the log!

here it is: https://gist.github.com/KalebClark/1ee19452d44d51233bd8a480920ca171

Thanks for the replay FL!

FormerLurker, 2018-11-28:
Ok, thanks for the log!  Can you also let me know what version of Octolapse and OctoPrint you are using?  Also, are you running OctoPi (if so which version), or did you install OctoPrint manually?  Also, if you can please post your snapshot script too so I can proofread it.

Lastly, I'm assuming you don't get any error on the first snapshot, right?  If not, maybe you can also post your test gcode?  It could be that the camera doesn't have enough time to save the image then move it to the pi before the next snapshot is triggered.  If this is the case you might try testing that by lowering the camera resolution to the minimum, then re-run your test print and see if the error vanishes.  If it goes away I'll think about how to add a feature to detect this case and to report a sensible error message.

KalebClark, 2018-11-28:
Thanks for the prompt reply! (Fixed?? See bottom)

**More Info**: Octoprint and Octolapse have been running perfectly with previous release of octolapse and raspi cam. Updated octolapse to 0.3.4 for external script functionality. 

**More Info**: Right when the print head goes to back left (as it should), I do a ps -eaf on the raspi to see the gphoto command running. On every snapshot I see two instances running. PID's are different, so its not hanging a process. (I also verified by running `ps -eaf` between snapshots, and there are no gphoto processes running) .Also seems to not be incrementing the counter for the snapshots. (Gist here:) https://gist.github.com/KalebClark/66d0ea19275046573a7ca2025092d743

Yes, here is everything you asked for:

**Octolapse version**: 0.3.4
**Octoprint Version**: 1.3.9
**Install Mode**: Installed Octoprint manually over raspbien Jessie. 
**Snapshot Script**: https://gist.github.com/KalebClark/afe02e2da72bb422f3cee2b993965af8
**Error on first snapshot**: It does. Throws error on every snapshot.
**Saving Time**: I have upped it to (20000 snapshot timeout config) 20s. I tested by copy/pasting command from `ps -eaf | grep gphoto2` output while octoprint was running (ran tests while octolapse was NOT running)). Used the 'time' command to get real timing info.
**GCODE Test**: https://gist.github.com/KalebClark/0b97bdbb77c1c2bfdee7fe6e8e1aa5ba

Sooo.... While I was gathering all of this info, I tested with the suggestion of upping the timeout. I had it at 16s, which seemed REALLY long. But I guess it wasnt long enough because it worked after upping it to 20. Going to run a real print and see now. Ill post after to let you know. 

But here is a bunch of info! =)

FormerLurker, 2018-11-28:
Thanks for all of that!  I will wait to review until I hear back regarding your real print :)  My guess is that you are seeing the script process as well as the child process (from `sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"`).  I'm guessing your printhead stays put the entire time the snapshot is being downloaded, right?  If so, it's possible to trigger a snapshot and wait ONLY for the snapshot itself to be acquired.  This can pretty dramatically reduce the time it takes to capture an image (over half, maybe more), but you do lose some functionality (snapshot image preview and transpose to be exact).

The idea is that we can capture the image and save it to SD.  Octolapse continues the print while the camera is saving the image to the internal SD (use fast sd memory!).  After the print is completed a second script runs to download and rename all the images, then the rendering process kicks in.

I'm still testing this process, but several folks are using it.  The biggest downside is that all photos on the SD card must be deleted before the print starts (can be accomplished via the camera initialization script) due to some limitations in gphoto2 (or the dslr interface, not sure exactly which).  It may still be possible to work around this, but I haven't discovered a way to do this yet.  

After you get your live print working let me know if you want to explore this option, and thanks for all the info!

KalebClark, 2018-11-28:
Yep, Printhead is staying still for 20 seconds =) Really slowing down this print. haha.

I would be interested in testing that! Is the functionality built in to this version? Do I just do --capture-image, and then setup the pre/post scripts to wipe SD (CF in my case... Its a 300D) and download images?

KalebClark, 2018-11-28:
More fun info, print failed due to "too many consecutive timeouts". Terminal complaining about all the timeouts from waiting for the image to download.

It ran about 15 snapshots, then choked on the timeout. So its not consistent, some frames work, some dont apparently.

KalebClark, 2018-11-28:
Ill check out those timeouts.

Raspi 3 B+.

Its the camera that is the bottleneck. Its *the first* digital rebel. Bought it like 17 years ago! I was trying to use it as a dedicated camera. I plugged in my Canon 70D and it took picture, downloaded (at high resolution) within 4 seconds. 

I also checked that the old camera does not support bulk download, or deleting files from the CF card. Im going to have to either suck it up and deal with LONG timeouts and LONG prints, or figure out a different camera to use. 

Thanks for all of your help, sorry it ended up being something silly. Your software works great! Hopefully you learned something from the debugging info I sent you.

And by the way, thanks for the time you put into this. Really good stuff.

FormerLurker, 2018-11-28:
No problem at all!  I'm more than happy to help, especially someone who is willing to post all the necessary info and is willing to put up with the back and forth :)  

You might still be able to use that camera though.  You might be able to use the '--trigger-capture' flag, then download the resulting images from your camera manually.  That would be an extra step for you to perform, but it might just give you acceptable performance!  Here is a copy of the acquire image script (extra vars there for brevity):
```
#!/bin/sh
# Camera Capture Script - Leave on Camera, Don't download
# Requires a camera initialization script with the following command:  gphoto2 --capture-image --set-config capturetarget=2
# Written by: Formerlurker@pm.me
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# trigger the camera and exit immediately
gphoto2 --trigger-capture
```

And an initialization script to set the capture target to the SD card:
```
#!/bin/sh
# Camera Initialization Script
# Sets the capture target to the SD card
# Written by: Formerlurker@pm.me

# Put the arguments sent by Octolapse into variables for easy use
CAMERA_NAME=$1
# Set camera to save images to flash memory
gphoto2 --set-config capturetarget=2
```

Don't give up hope yet!

KalebClark, 2018-11-28:
Awesome, Trying it now. 

That old camera does not support "--trigger-capture". Had to use "--capture-image". Docs say its the same thing. Captures to internal memory. Camera also does not support the init script for capture target, nor any other config commands. Or erasing images, or even downloading images... 

**Going to try this process**:
1. Format card manually.
2. Let octolapse capture all images to camera internal memory
3. Download images (can use faster computer than raspi for rendering)
4. Render with FFMPEG. 

Ive done timelapse renders with ffmpeg before, just gotta remember how to do it right =)

KalebClark, 2018-11-29:
Alright, Futzed around with this for a bit. That procedure was seeming to work, but I ran into something new. 

When using Octolapse, prints get stringy. I turned off octolapse, and prints are fine. I have verified that all my slic3r PE settings match on the Octolapse printer settings page. Very familiar with that, and I am positive they are right. I cancelled some of the prints, thats why they dont look finished. Was intentional. 

Examples:
[image]
[image]
[image]

FormerLurker, 2018-11-29:
Did you select your slicer type in the printer profile, then copy the settings out?  If not, try that first.  There are lots of ways to deal with this part.

KalebClark, 2018-11-29:
Yeah, In the Octolapse printer config, I selected Slic3r Prusa Edition and copied all the settings from slic3r to the Octolapse printer config.

FormerLurker, 2018-11-29:
Double check retraction length and speeds.  Don't use 0 for either speed!

KalebClark, 2018-11-29:
The MK3 profile default with Slic3r PE has detraction speed at 0. Works fine for all other prints. 

I can change it and run my quick test again to see what happens.

FormerLurker, 2018-11-29:
Sorry that's a known issue.  Put in the retraction speed (that's what 0 means in slicer, but it means something different in others.  Working on that).  See if that helps.

KalebClark, 2018-11-29:
I set retraction and detraction to 35 in slic3r and in octolapse.

Same result. Stringy fun!

FormerLurker, 2018-11-29:
Boo!  Is that all that's changed since you posted your settings?  I'll take a look at them tomorrow if so.  That amount of stringing is not acceptable :)

KalebClark, 2018-11-30:
Hey man, sorry for delay! The only things I have changed are logging levels (Test mode -> Real -> Test etc.) and printer settings in octolapse for the few items you asked to change. So settings.json should still be relevant.

FormerLurker, 2018-11-30:
Excellent!  I'll take a look tomorrow.  I have an mk2.5 upgrade to do first :)

FormerLurker, 2019-10-12:
Hey, so sorry I never got back to this.  I ended up getting sick for a LONG time right after this post, and just lost track.  I'm really embarrassed honestly.

Anyway, since this has been idle for a good long while, and I'm finally dealing with the issue back log, I'm marking this 'Waiting for Response' to give you time to let me know if you are still having issues.  Once again, so sorry for the delay.. :(

FormerLurker, 2019-11-09:
So, it's been nearly a month since this thread was active, so I'm going to go ahead and close it down.  @KalebClark, if you want me to reopen, let me know.  Or, you can always create a new ticket.  Thanks!

closed 2019-11-09 by FormerLurker