FormerLurker/Octolapse #283: Octoprint 1.3.9 [Running setup.py install for pillow: started]

angryjelly24, 2018-11-30:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. Downloaded Octopi image v0.15.1 and installed on raspberry pi
2. Updated Octopi to v1.3.9 and rebooted system
3. Opened plugin manager to try and install octolapse 

#### What should have happened?
Had just freshly installed and updated  Octopi to v1.3.9 and tried to install Octolapse from the plugin manager.

#### What happened instead?
Installation failed constantly at the "Running setup.py install for pillow: started" line. 
Also tried manual installation but got this error "Command "python setup.py egg_info" failed with error code 255 in /tmp/pip-req-build-_9pc644v/"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: v1.3.9

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: Marlin v1.1.9

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/e4kPj1Pw

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2018-11-30:
Try manually installing pillow into the Virtual environment.  Be sure you upgrade pip first.  Let me know if you need help with that.

ManicalEngineer, 2018-12-23:
Which virtual env does pillow need to get installed in to?

FormerLurker, 2019-10-12:
Since this was solved, I'm going to finally close the issue :)  As an aside, I noticed that most problems installing pillow are actually PSU related!   It sounds crazy, but it's been verified over and over again.  Installing through the console appears to work more often than through Octoprint, but now Octoprint has a feature that prevents installation if there are undervoltage/overheating issues exactly because of things like this.  I thought you might enjoy hearing that.

So, thanks for posting here!  I appreciate it.  Closed :)

closed 2019-10-12 by FormerLurker