FormerLurker/Octolapse #285: Could not claim the USB device

gigiukas, 2018-12-08:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

gigiukas, 2018-12-14:
Is here somebody who is willing to help me?

FormerLurker, 2018-12-14:
You didn't fill out the ticket above, so I don't know what your problem is even.  Please fill in the missing info, and be as specific and detailed as possible.  Thanks!

FormerLurker, 2018-12-14:
I'm not familiar with that camera.  Does it operate as a webcam, or are you using a custom script to acquire photos (maybe with gphoto2 or digicam control)?

FormerLurker, 2018-12-18:
Have a look at [this](https://github.com/gphoto/gphoto2/issues/181) issue within the gphoto2 github page.  It looks very similar.  It looks like the PI is connecting to the SD card as a mass storage device, which prevents gphoto2 from communicating with it.  You may need to kill the offending processes.

FormerLurker, 2019-10-12:
So, this issue has been idle for a long time, and I'm finally getting around to dealing with the issue backlog.  I'm going to go ahead and close this, but if you want it reopened, just let me know!  Thanks for your post!!!

closed 2019-10-12 by FormerLurker