FormerLurker/Octolapse #286: excessive snapshot delay after updating to 0.3.4

OutlawAndy, 2018-12-09:
#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

***
I've been using Octolapse with a raspicam on my mk3 for sometime with great success.  After updating to 0.3.4 and resolving several initial error codes at the start of printing, I was able to get printing again.  Except that there is now a ~5second delay during snapshot.

Have I missed a new setting somewhere? BTW - I'm using the gcode method for taking snapshots with "Snap" after each layer change in the gcode, incase that is relavent.

Thanks for your help!
***

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 mk3
Printer Firmware Version: 3.4.1

gege2b, 2019-01-24:
I have a similar issue (but with a bit less delay - about 1sec)
I'm using Octoprint 1.3.10 and Octolapse 0.3.4

Here is a snippet of the full diag log when a snapshot is triggered : 
```
2019-01-24 12:42:56,330 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X80.268 Y104.760 E3.09163, tags: set(['source:file', 'filepos:702709', 'fileline:25385'])
2019-01-24 12:42:56,352 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X80.026 Y104.900 E3.09654, tags: set(['source:file', 'fileline:25387', 'filepos:702769'])
2019-01-24 12:42:56,354 Received G1
2019-01-24 12:42:56,356 Position Change - G1 X80.026 Y104.900 E3.09654 - Absolute Move From(X:80.176,Y:104.88,Z:27.75,E:853.06682) - To(X:80.026,Y:104.9,Z:27.75,E:853.06927)
2019-01-24 12:42:56,363 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X80.176 Y104.880 E3.09409, tags: set(['filepos:702739', 'source:file', 'fileline:25386'])
2019-01-24 12:42:56,384 Queuing Command: Command Type:None, gcode:M118, cmd: M118, tags: set(['filepos:702775', 'source:file', 'fileline:25388'])
2019-01-24 12:42:56,386 GcodeTrigger - Waiting for extruder to trigger.
2019-01-24 12:42:56,388 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X80.026 Y104.900 E3.09654, tags: set(['source:file', 'fileline:25387', 'filepos:702769'])
2019-01-24 12:42:56,393 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X74.974 Y104.900 E3.17844, tags: set(['source:file', 'fileline:25389', 'filepos:702805'])
2019-01-24 12:42:56,401 Received G1
2019-01-24 12:42:56,402 Position Change - G1 X74.974 Y104.900 E3.17844 - Absolute Move From(X:80.026,Y:104.9,Z:27.75,E:853.06927) - To(X:74.974,Y:104.9,Z:27.75,E:853.15117)
2019-01-24 12:42:56,404 Extruder Changed: E:0.0819, Retraction:0 IsExtruding:False-True, IsExtrudingStart:False-False, IsPrimed:True-False, IsRetractingStart:False-False, IsRetracting:False-False, IsPartiallyRetracted:False-False, IsRetracted:False-False, IsDetractingStart:False-False, IsDetracting:False-False, IsDetracted:False-False
2019-01-24 12:42:56,407 A snapshot is triggering
2019-01-24 12:42:56,408 About to take a snapshot.  Triggering Command: G1
2019-01-24 12:42:56,414 Snapshot Gcode - SnapshotCommandIndex:0, EndIndex:3, Triggering Command:G1 X74.974 Y104.900 E3.17844
2019-01-24 12:42:56,416     G1 X80.026 Y104.900 F7200.000
2019-01-24 12:42:56,421     G1 X80.026 Y104.900
2019-01-24 12:42:56,423     G1 F1200.000
2019-01-24 12:42:56,435     G1 X74.974 Y104.900 E3.17844
2019-01-24 12:42:56,436 Sending snapshot start gcode and snapshot commands.
2019-01-24 12:42:56,446 Octolapse is requesting a position.
2019-01-24 12:42:56,448 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X80.026 Y104.900 F7200.000, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-01-24 12:42:56,449 Received G1
2019-01-24 12:42:56,451 Position Change - G1 X80.026 Y104.900 F7200.000 - Absolute Move From(X:80.026,Y:104.9,Z:27.75,E:853.06927) - To(X:80.026,Y:104.9,Z:27.75,E:853.06927)
2019-01-24 12:42:56,457 Queuing Command: Command Type:None, gcode:M400, cmd: M400, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-01-24 12:42:56,464 Queuing Command: Command Type:None, gcode:M114, cmd: M114, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-01-24 12:42:56,474 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X80.026 Y104.900 F7200.000, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'snapshot_gcode', 'source:plugin'])
2019-01-24 12:42:56,491 Sent to printer: Command Type:None, gcode:M400, cmd: M400, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'snapshot_gcode', 'source:plugin'])
2019-01-24 12:42:57,696 The position request is being sent
2019-01-24 12:42:57,702 Sent to printer: Command Type:None, gcode:M114, cmd: M114, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'snapshot_gcode', 'source:plugin'])
2019-01-24 12:42:57,724 Octolapse has received a position request response.
2019-01-24 12:42:57,728 Printer event received:PositionUpdate.
2019-01-24 12:42:57,749 Taking a snapshot.
2019-01-24 12:42:57,751 Starting Snapshot Download Job Immediately.
2019-01-24 12:42:57,753 Snapshot - downloading from http://127.0.0.1:8080/?action=snapshot to /home/pi/.octoprint/data/octolapse/snapshots/264257de-30c0-474f-96d9-b576d8c68fb2/354def78-9eea-409a-ad23-ee966dfff4ba.
2019-01-24 12:42:57,799 Snapshot - Snapshot saved to disk at /home/pi/.octoprint/data/octolapse/snapshots/264257de-30c0-474f-96d9-b576d8c68fb2/354def78-9eea-409a-ad23-ee966dfff4ba/20190124-115108-Superman_Crystal_Console_Pen_Holder_0.2-0-000139.jpg
2019-01-24 12:42:58,014 Snapshot Download Job completed, signaling task queue.
2019-01-24 12:42:58,023 Sending snapshot return and end gcode.
2019-01-24 12:42:58,031 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X80.026 Y104.900, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-01-24 12:42:58,033 Received G1
2019-01-24 12:42:58,037 Position Change - G1 X80.026 Y104.900 - Absolute Move From(X:80.026,Y:104.9,Z:27.75,E:853.06927) - To(X:80.026,Y:104.9,Z:27.75,E:853.06927)
2019-01-24 12:42:58,044 Queuing Command: Command Type:None, gcode:G1, cmd: G1 F1200.000, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-01-24 12:42:58,045 Received G1
[... cut ...]
2019-01-24 12:42:58,058 Sent to printer: Command Type:None, gcode:G1, cmd: G1 F1200.000, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'snapshot_gcode', 'source:plugin'])
2019-01-24 12:42:58,063 Received G1
```

As you can see, there is almost 2sec on this snapshot between the "A snapshot is triggering" and the "Snapshot Download Job completed, signaling task queue."

I'm not using any stabilization settings, I just want to take a snap when the gcode trigger is parsed.
As I can see, there are a lot of stuff being done here witch are mainly relevant when stabilization is enabled (position request, some moves (witch are "null" moves in this case), etc...)

FormerLurker, 2019-01-24:
Can you post a video of what is happening?  From the log, the point at which Octolapse takes over is actually:
```
2019-01-24 12:42:57,724 Octolapse has received a position request response.
```

The prior two commands that are sent, M400 and M114 (with m400 being sent at 2019-01-24 12:42:56,491), are executed probably while your printer is still clearing its internal buffer.  The buffer is guaranteed to be clear once Octolapse receives a position response (this is how it knows when to take a snapshot).  The log doesn't show when the triggering command (G1 X74.974 Y104.900 E3.17844) was sent.  That command was held and sent after the snapshot and stabilization code is complete.  However, a response from the printer regarding receipt of the the final snapshot gcode (generated by Octolapse) was received at (probably, it's not exactly easy to tell which response is being responded to) 2019-01-24 12:42:58,063, which means the entire snapshot routine took 339MS, at least according to the logs I'm seeing.  Downloading the snapshot took 265MS, so that means the rest of the process took 74MS, which doesn't seem too bad (but could probably be improved).  The initial M400 + M114 combo probably took some amount of time to execute, but it's impossible to know exactly how much, at least with the current information.  Also, the 'NULL' moves should probably be removed (no detection was added for moves of 0 distance, which could be rectified), and they probably took a few MS each to execute, but it's very difficult to tell how much time they added individually.

A video may clarify what's happening.  I'm sure you are probably seeing a delay greater than what I'm calculating based on the logs, but I just can't tell what's going on from my end yet.

FormerLurker, 2019-01-24:
@gege2b, FYI, the 265MS download time is pretty high.  It took 46 MS to actually save the JPG to disk, so the rest of the time must have been generating a thumbnail and/or transposing the snapshot.  Do you have any transposition options set in your camera settings?  I've never actually measured how long this takes, but it looks like something in the image processing is taking about 200MS, which is indeed pretty high.

gege2b, 2019-01-24:
Hi @FormerLurker , no problem I will post a video tomorow, it's pretty hard to measure how long it pauses, but  I think it's more than 0.3s actually

Just to clarify some things : I'm actually NOT using any stabilization or fancy animation (they're great BTW but I just want a bare timelapse)

The only thing I do is using a trigger gcode, so I expect the snapshot routine to just do a snapshot (and all linked image processing) when the trigger word is spotted, and that's all. But I admit I don't know how all the gears are running inside Octolapse :)

I have neither transposition nor thumbnail generation turned on, But I have set Jpeg quality at 90 and webcam resolution at its maximum, could it be the culprit here ?

gege2b, 2019-01-25:
Here is a little [video ](https://www.youtube.com/watch?v=M1JQxMRn9IY)of the problem
As you can see, it freeze on the upper-left side of the heart being printed (that's where my processing script insert the trigger word for gcode trigger)

the pause seems shorter than on my yesterday's print but it's still here

FormerLurker, 2019-01-25:
So, I've been enhancing the snapshot process, and the following improvements have been added:

1.  The snapshot routine now ends as soon as the snapshot is downloaded or acquired.
2.  Added threads to do all of the image post processing.
3.  Once the post processing is finished, the threads send a 'new thumbnail available' message to the client, which then downloads the latest thumbnail.
4.  Replaced Image.Resize with Image.Thumbnail.  On my desktop this reduced the calculation time from about 84ms for a 1080p image to 38ms.  Mind you this reduction is likely much greater on an RPI!

So, all of that added together probably saves quite a boatload of time.  I will be pushing these changes to a new branch soon.  However, parts of the program are super new, and I'll be running some test prints soon.  Let me know if you are willing to test it, and if you're willing to lose your current settings (they could be saved and then re-applied after downgrading to the latest release if you want).  I'm still working on settings migration, so that doesn't work too well now.  However, it now extracts slicer settings directly from your gcode file (probably), so there is less setup in theory.

Let me know!

FormerLurker, 2019-01-25:
Well, my (limited) testing has shown great success.  I wasn't even sure it was capturing images while looking at the printer, until I checked the octolapse tab.  So it now takes me around 80MS to acquire an image, but the post processing, done in the background, took 213MS, which means it takes about 25% of the time it had been to acquire a snapshot.  

I had zlift and retract enabled, so I'll turn that off and run another test.  If you want to try this out, please download your settings.json file first from OctoPrint settings->Octolapse.  The settings upgrade routine is pretty primitive at the moment, and relatively untested, so there are going to be problems with it.  If you have a backup settings file it will be easy to downgrade.

Then install Octolapse from the following URL:  https://github.com/FormerLurker/Octolapse/archive/new_settings.zip

Are you using cura?  If so, there is a bit of setup in your slicer to get this working, but if not, just go to your printer profile and select 'Automatic' as your slicer type (pretty neat huh?) and run a test print!

If you are using cura, add [this template](https://github.com/tjjfvi) to your start OR end gcode (not both) and re-slice.  That will add all of cura's settings to the gcode.  In the future I'll distill that huge template so that only the settings Octolapse needs are in there.  For now it's everything :)

Let me know if you have any questions.  Also, are you on my discord server (I don't think you are...).  If not, shoot me an email at formerlurker@pm.me and I'll send you an invite.  That will be good if you have any problems or suggestions.

gege2b, 2019-01-25:
great !! I will try this on monday, my printer is at my office at the moment
I'm using Slic3r and it already embed all settings in the gcode file

FormerLurker, 2019-01-26:
@gege2b, I was able to get the delay down to around 80ms total, including image acquisition with zlift and retract turned off :)  I got rid of all of those 'null' commands, which was pretty tough, but had a surprising impact on time, though I'm not exactly sure why.

I'm glad this process was optimized.  I'm working on some new stabilization that will benefit greatly from this enhancement.

gege2b, 2019-02-01:
I'm trying a print right now (I know we are not monday anymore, got a flu problem in between :D )
As long as I can tell, the delay is simply gone... 
At least it's so small that I can't even notice when the screenshot is taken (to be honest, I had to check the snapshot number in octolapse's panel to confirm it was working)

I will post a picture of the same object with and without this fix so you can clearly see how it positively impact print quality

FormerLurker, 2019-02-01:
Woot!  Also, glad you are recovering from the flu.  I just got over two colds in a row.  Was sick for over a month.  Was around too many small children over the holiday :)

I'm really glad I got to the bottom of the delay.  Now all snapshots will be substantially faster for everybody.  There really is no downside.  Also, an ultra low delay will be necessary for the next feature I'm working on: quasi-stabilized timelapses.  The extruder will never move away from the print, but the goal will be to stabilize the timelapse as well as possible.  This kind of timelapse could always be enabled (I believe), even in vase mode and for delicate prints.

I cant wait to see your results!

Also, did you turn on automatic slicer settings detection?  I'm curious how well that feature is working.

gege2b, 2019-02-01:
winter is coming, they said

Can't wait to test this stabilization mode. My printer is pretty prone to oozing (it's a legacy prusa mendel and I still need to adapt an extruder fan since... well, about 6 years) that's why I don't want to move the extruder if not necessary

I did not try automatic slicer settings today because I didn't have a lot if time, but I will do next week for sure

gege2b, 2019-02-06:
I just realized that your "quasi stabilized" new feature is exactly what I did with "my" post processing script ! ( #109  )
can't wait to test this native implementation !

FormerLurker, 2019-02-06:
It took me a while to get around to it, but better late than never!  I had to spent a lot of time reducing parsing time to make this feasible and reasonable.  Had to do a lot of it in C++, but i got it down from around 200 seconds to under 30 for a 38mb sample gcode.

Once the foundation is set it will be pretty easy to add new ones.  I have several ideas :)

FormerLurker, 2019-10-12:
So, I've been able to reduce the preprocessing time to mere seconds for a 30mb sample file on the RPI 3b+!  Also, shortly after my last post I too got the flu and MAN was it bad.  Both me and my wife were out of commission for a couple weeks.  Glad that is over.  I'll be first in line for the flu shot this year!

Since it looks like this is now solved, I'm going to close this issue down, but let me know if you want to test V0.4.  It's coming along nicely (though much slower than I'd hoped)!

closed 2019-10-12 by FormerLurker