FormerLurker/Octolapse #287: When spiralizing outer contours, blobs appear0.

1Ghasthunter1, 2018-12-10:
Hello,
I just ran an octolapse with a vase, spiralized outer contour, and had it smooth the Z, so I just had octolapse take a snap every .2mm layer height. When it did this, a slight blob would form on the print. The result is extremely ugly, and ruins the print. How do I fix this?

FormerLurker, 2018-12-10:
Vase mode prints are definitely the hardest timelapses to implement.  When using vase mode the nozzle should never leave the print, and obviously Octolapse causes that to happen.  I've gotten a few decent vase mode timelapses to work, but I spent a lot of time tuning my retraction/detraction speeds/lengths.  In the end a perfect vase mode print wouldn't have blobs, but will have some exterior perimeter artifacts, unfortunately.  The photos of your print don't show what I would call blobs, but DO show artifacting caused by moving the print head while printing the exterior perimeters.

Can you send me your gcode file used for the above print?  If so I will see if some of the other quality settings in Octolapse can help.  It's possible that the new 'Feature Detection' option can prevent snapshots when printing the exterior of the vase, but it depends entirely on the print in question.

FormerLurker, 2019-10-12:
I'm going to close this issue, but I wanted to let you know that I've added a new feature called 'snap to print' that will reduce artifacts in vase mode (or any other print for that matter).  This mode prevents the extruder from leaving the print, but stabilized the timelapse as much as possible.  There are many options too for achieving different results.  This will be available starting with V0.4, which is in testing now.  Let me know if you'd like to try it out and i can send instructions.  Thanks for posting!

closed 2019-10-12 by FormerLurker