FormerLurker/Octolapse #288: API /setCurrentProfile respond error HTTP 500

anthonyst91, 2018-12-10:
Hello,

I experiment a HTTP 500 issue when trying to call the POST /setCurrentProfile API.
From octoprint.log I get:
TypeError: 'NoneType' object has no attribute '__getitem__'

I can set the profile settings without any issue from the OctoPrint web interface.

Issue encountered for all types of profiles.

#### Version of Octolapse
Octolapse Version: 0.3.4
Fresh install of Octolapse on a fresh install of Octopi

#### Version of OctoPrint
OctoPrint Version: 1.3.9 running on OctoPi 0.15.1 

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. From a HTTP request sender tool: 
POST http://IP_ADDRESS/plugin/octolapse/setCurrentProfile?apikey=XXXXX
With following body:
{
    "client_id": "my_application_name",
    "guid": "7c0eb400-5a3e-45b3-aae9-cc01918a9bb6",
    "profileType": "Printer"
}
Please note that the passed guid exists and it is not the profile currently in use for the Printer section.

#### What should have happened?
HTTP code 200 or 204
And setting changed in the Octolapse config.

#### What happened instead?
HTTP error: 500 
With following body:
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
500 Internal Server Error
The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.

#### Operating System running OctoPrint and Octolapse
OS Name: Raspbian (Octopi)
Os Version: Raspbian GNU/Linux 9 (stretch)

#### Link to plugin_octolapse.log
2018-12-10 12:25:41,257 Printer event received:ClientClosed.
2018-12-10 12:25:42,005 Printer event received:ClientOpened.

#### Link to octoprint.log
2018-12-10 12:25:30,275 - octoprint - ERROR - Exception on /plugin/octolapse/setCurrentProfile [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 269, in set_current_profile_request
    profile_type = request_values["profileType"]
TypeError: 'NoneType' object has no attribute '__getitem__'
2018-12-10 12:25:30,279 - tornado.access - ERROR - 500 POST /plugin/octolapse/setCurrentProfile?apikey=1E0BCD2C0EA642CD81D19F0E26EFB021 (ipv6_address) 15.74ms

anthonyst91, 2018-12-10:
Damn. I finally fixed the issue by adding the following header:
Content-Type: application/json. Now it works fine!

Previously I was using this header:
content-type: application/json
with 'c' and 't' in lowercase. But the POST always returned 500.

FormerLurker, 2018-12-10:
This is an interesting issue, and I'm glad you figured it out!  Yes, the content type is required, as you saw.

Can I ask what you're trying to do?  I love to see people working with the code, and would love some more info if you're willing to give it!

anthonyst91, 2018-12-10:
What's strange is that "content-type" was set, but it was not working until I change it for "Content-Type". That's strange, Flask seems to be case sensitive regarding the headers.
Here's where I found the answer to my issue: https://github.com/pallets/flask/issues/2891

It's finally working well ;)
I've sent you an email few days ago: I'm currently integrating Octolapse in Printoid for OctoPrint:
[https://printoid.net](https://printoid.net)

Because a lot of my user requested me the support of your amazing plugin, I did it :) 
Everything is now working fine, Printoid is able to:
- Switch ON/OFF the plugin
- Display the current state of the plugin (is enabled, is capturing snapshot...)
- Adjust the profiles (select another one in the list of available profiles)
- Enable/disable trigger state, extruder state...

Here is a screenshot of the current implementation:
[Screenshot](https://i.ibb.co/VH3Mkd8/Screenshot-20181210-170658-fr-yochi76-printoid-phones-premium.jpg)

FormerLurker, 2019-01-21:
Hey, I've been lazy about closing out old issues, and since I know Printoid is working with Octolapse (thanks for the copy!), I believe this issue is closed.

Super job with the interface, by the way.  Your app is super slick, and you managed to make your Octolapse integration look like Octolapse :)  Bonus points!

Closed!

closed 2019-01-21 by FormerLurker

AnthonyStephan, 2019-01-21:
Hello! 

Thank you for your great plugin, you've done the huge part of this amazing job :) 
There is a lot of Printoid users that are using the Octolapse integration in the app, your plugin looks very popular :) good job.

All my best,
Anthony