FormerLurker/Octolapse #289: Octolapse stops taking snapshots after wrong sent snap command

Pelzkiwi, 2018-12-11:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Switching from "Fixed Run Length" to "Static FPS 30".
2. Starting print for overnight printing the same way I did before where octolapse worked well. Nothing special

#### What should have happened?
Taking snapshots every time the Snap Command "M117 snap" appears in the code.

#### What happened instead?
Octolapse stopped taking snapshots after 246 layers of 744. I tried the same gcode file twice and it also occured twice. I took a look at the log file and found a mysterious snap command in line 67461:

"2018-12-11 02:48:00,089 Queuing Command: Command Type:None, gcode:M117, cmd: M117 192.168.2.101, tags: set(['trigger:printer.commands', 'source:plugin', 'plugin:ipOnConnect'])
2018-12-11 02:48:00,091 An error occurred while parsing the command string.  Details: Cannot parse the gcode address, multiple periods seen."

Every other snap command looks like this:
"2018-12-11 02:47:27,333 Queuing Command: Command Type:None, gcode:M117, cmd: M117 snap, tags: set(['fileline:493808', 'source:file', 'filepos:13883709'])
2018-12-11 02:47:27,334 GcodeTrigger - Waiting for extruder to trigger."

After that mysterious command all other snap commands were obsolete...

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3 MMU2
Printer Firmware Version: 3.5.0

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Pelzkiwi/c89cdf738783a8b01de61504bbf9dbb3

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Pelzkiwi, 2018-12-31:
The error occured a second time with the same issue in the logfile.
At line 5635 appears the M117 command with the IP Adress instead of the snap command.

The link to the log file: https://gist.github.com/Pelzkiwi/97a4cb8114d8fea3a52a14d1420bf018

FormerLurker, 2018-12-31:
Hi @Pelzkiwi!  The gcode file that you attached in your initial post appears to be a log file.  I see the M117 commands in the log, but would like to see the gcode file too if that's possible.  Thanks!

Pelzkiwi, 2019-01-01:
Hi @FormerLurker Thanks for your response! Sorry I attached the wrong file.

This link is the gcode where the issue occured the first time (I had to shorten it for Gist):
https://gist.github.com/Pelzkiwi/b317702cf8022e3eb2120b5b68220d93
This link is the gcode where the issue occured the second time:
https://gist.github.com/Pelzkiwi/ac97576570e11a2f954f841e63431917

Happy new year :-)

FormerLurker, 2019-10-12:
@Pelzkiwi, OMG so sorry I let this go for so long!  I got sick around this time and I seriously neglected some github issues due to this.  I've been apologizing all day as I'm trying to get caught up for this!

Anyway, I think this bug has been fixed in the upcoming V0.4.  I'm going to leave this issue open, and let me know if you want to try out the new version to see if this is fixed for you.

Thanks and OMG I'M SO SORRY I let this go for  so long.  Is my face red!

FormerLurker, 2019-11-09:
Hey @Pelzkiwi, it's been several weeks since this thread has been active.  I'm going to go ahead and close it now.  If you want me to reopen it, just let me know.  Thanks!

closed 2019-11-09 by FormerLurker

probertson, 2020-02-15:
@FormerLurker @Pelzkiwi I'm hitting this same error. Is there anything I can do to work around it? I will install the 0.4.0rc1.dev2 build to see if that fixes the issue.

FormerLurker, 2020-02-15:
@probertson, try the alpha of dev3 instead if you would.  Install using the plugin manager with the following URL: https://github.com/FormerLurker/Octolapse/archive/devel.zip

probertson, 2020-02-17:
@FormerLurker I tried both of the files where I ran into the issue, and it didn't happen again. I have also printed a couple more files that were exported at the same time with the same settings, and they worked also. So it seems like it's fixed!

Also I really like the improvements in v0.4 😃

FormerLurker, 2020-02-17:
Hooray!  I'm so glad it's working for you!  I'm getting ready to release that version today (0.4.0rc1.dev3), and there were a few changes/fixed that have been applied in the last two days.  You may want to upgrade again using the same link I posted above, or just wait for the actual release on github to drop.

If you have any comments/criticisms/observations regarding the new functionality, I'd love to hear it!  I still have a little time to make changes, and to adjust the verbiage of the release.