FormerLurker/Octolapse #290: Visudo: Still having to use "sudo" even after editing visudo with DLSR

chilicoke, 2018-12-13:
I followed the [DSLR guide](https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera) linked from wiki on a loose test Pi 3B, with a little bit of trouble shooting it all seems to work.

Now I'm trying to set it up on my mounted primary Pi 3B+, already edited visudo but ran into issue while testing `gphoto2 --capture-image` command; as it seems I still need to use `sudo gphoto2 --capture-image` to get it to work, which I did not had to do on my test Pi 3B.

I've tested the current memory inserted from my 3B+ into the test 3B and it's the same issue, need to use sudo to get it to work. I've already spent tons of time setting up this instance of OctoPrint and all the addons, so I'd prefer not having to redo the entire thing from scratch.

Also another weird issue I've encountered with my Canon 350D Rebel in normal mode: After a Pi reboot, USB disconnect, or camera restart. The first command of `sudo gphoto2 --capture-image` will also fail with the below error message, and I would have to switch the shooting mode on the camera dial to another mode and everything will work fine afterwards. Even if I connect the camera, switch the modes first, then issue `sudo gphoto2 --capture-image`, it would still fail. Any idea on how to fix this problem?

Thanks!

>pi@octopi-portable:~ $ gphoto2 --auto-detect
> Model                          Port
> Canon Digital Rebel XT (normal mode) usb:001,008
> pi@octopi-portable:~ $ sudo gphoto2 --capture-image
> Detected a 'Canon:EOS 350D (normal mode)'.
> 
> *** Error ***
> An error occurred in the io-library ('Unknown error'): No error description available
> ERROR: Could not capture image.
> ERROR: Could not capture.
> *** Error (-114: 'OS error in camera communication') ***
> 
> For debugging messages, please use the --debug option.
> Debugging messages may help finding a solution to your problem.
> If you intend to send any error or debug messages to the gphoto
> developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
> gphoto2 as follows:
> 
>     env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image
> 
> Please make sure there is sufficient quoting around the arguments.
> 
> pi@octopi-portable:~ $

#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: issue not within Octolapse

#### What were you doing when the problem occurred

1. modified visudo following the instruction
2. tested `gphoto2 --auto-detect`, camera detected fine
3. tested `gphoto2 --capture-image`. failed
4. tested `sudo gphoto2 --capture-image`. failed
5. turned SLR mode knob to another mode (any) position
6. tested `gphoto2 --capture-image`. failed
7. tested `sudo gphoto2 --capture-image`. worked fine

#### What should have happened?
test `gphoto2 --capture-image` and camera takes an image

#### What happened instead?
1. tested `gphoto2 --capture-image`. failed
2. tested `sudo gphoto2 --capture-image`. failed
3. turned SLR mode knob to another mode (any) position
4. tested `gphoto2 --capture-image`. failed
5. tested `sudo gphoto2 --capture-image`. worked fine

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: newest OctoPi

FormerLurker, 2018-12-18:
Thanks for posting!  Sorry about my late response.  Winter colds are bad.

Regarding the sudoers file, can you run this command and post the results (pardon me if you've already tried this):
```
whereis gphoto2
```
also, can you post your sudoers file (remove any parts that might contain sensitive info.  probably nothing in there is sensitive, but just in case)?  That will help.

The other problem is likely either a camera issue or something within gphoto2 itself.  You can try posting [here](https://github.com/gphoto/gphoto2/issues/new) and ask the author.  This may well be a known issue, or something with an existing workaround.

FormerLurker, 2019-01-21:
Hey, I'm pretty sure we got this working, so I'm going to close this issue now.  I've been lazy about closing things out recently :)

Closed!

FormerLurker, 2019-10-12:
Hmm.  I said I closed this issue last JAN, but apparently clicked the wrong button.  Sorry.  Now it's closed :)

closed 2019-10-12 by FormerLurker