FormerLurker/Octolapse #291: No config menu if Octolapse is Disabled

andrelec1, 2018-12-15:
Octolapse Version: 0.3.4
OctoPrint Version: Octolapse 0.3.4
OctoPrint 1.3.10

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
Try to add another webcam on octolapse, but not directly related.

#### What should have happened?
Edit settings of octolapse ...

#### What happened instead?
I only have access to the Main and About page in config.

[image]

[image]

FormerLurker, 2018-12-15:
So this should probably change.  It made sense when in an earlier version of the interface, but not at all now.  I'll look into the complexity of the change and will get back with an ETA.

FormerLurker, 2019-10-12:
So, this has been fixed in V0.4 (still in testing!)  Let me know if you want to try it out.

Closed!

closed 2019-10-12 by FormerLurker