FormerLurker/Octolapse #292: Printer not in the list, what should I do?

leoheck, 2018-12-17:
Hi, my printer is not in the printers list. 
It is an `Ender3`.
What should I do?

FormerLurker, 2018-12-18:
Hi @leoheck!  It's not too difficult to create a new printer profile.  It's always possible that things won't work 100% with a printer that hasn't been tested, but it would be unusual.

First take a look at [this video](https://www.youtube.com/watch?v=YsxjQhWKz5w), which explains how to create a profile from scratch.  This was made using a previous version, things have changed quite considerably with regards to how exactly you enter your slicer settings.  You can ignore all of the parts where slicer settings are entered in that video, but the rest should be correct.

Next, take a look at [this video](https://www.youtube.com/watch?v=sDyg9lMqMG8) which WILL show you how to enter your slicer settings into the printer profile.  That should get you pretty far.

You can expect a few bugs with your config, so be sure to try a small test print (something fast and easy).  You can use Test Mode if you want things to go even faster (no heating time and no extrusion).  If you run into errors, or if no snapshots are taken, enable the 'Position State' info panel (slider on the Octolapse tab within the 'Current Settings') and take a screenshot before cancelling your print, then post it here.  If I can't figure out what's going on I may ask you to repeat the print with the 'Test Mode - Full Diagnostic' option enabled.  This will cause Octolapse to create a detailed (and huge) log file.  This helps me figure out what's going on.

Good luck!  When you've finished the profile I'll add it to the defaults if you post it.  You can export your settings from OctoPrint Settings (the wrench/spanner icon)->Octolapse.  There's a button to download your settings there.

leoheck, 2018-12-19:
Awesome, that is a good info @FormerLurker. I will try to configure it to increase the list the Octoprint list. Thank you.

FormerLurker, 2019-10-12:
So, you might be interested to know that I've added a default Ender 3 profile, which will be available in V0.4!  In fact there is now an entire repository for Octolapse profiles, so I can add new profiles any time.  You can now configure them by selecting the make and model.  This version is still in testing.  I'm considering this issue closed, but let me know if you want to test the new version, and thank you for your post!

closed 2019-10-12 by FormerLurker

leoheck, 2019-10-13:
@FormerLurker I am glad to hear that, thanks for letting me know. I also have an Ender-5 too. It is a little bit different than the Ender-3, the main difference is that it has the stop ends on back-right. It also has a bed that goes down instead of moving the extruder up.