FormerLurker/Octolapse #294: [Request]

SystemDisc, 2018-12-21:
#### If this is a feature request describe it here
I would like the ability to disable the error message about my print head being out of bounds. I have ae BLTouch mounted on the right side of my print head, and I have configured my printer to consider farthest left and front position that the probe can reach as X0 Y0, which means that when I home the X and Y axis, they are negative values. I do this because I do not want to print in areas where my probe cannot reach. However, my print head can reach that area without problem, and Octolapse shouldn't care in this situation. Hence, I'd like to be able to disable the message.

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.10

FormerLurker, 2018-12-21:
So you dont want the printhead to be able to move to these reachable areas when stabilizing?  Also, the error message wont show unless gcode actually moves into this area, so you may want to look at your start gcode.  I will consider the request in any case, though.  Thanks for your post!

SystemDisc, 2018-12-21:
No, I don't care whether or not Octolapse will move outside of the defined print area for stabalization.

The position of the head after homing is X:-48.00000, Y:1.00000, and Z is some positive integer.
I have this in my startup code:
```
G28 ; home all axes
M420 S1 ; use stored bed levelling data
G1 Z5 F3000 ; lift
G1 X0 Y10 F1500 ; move to prime
G1 Z0.4 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 Y80 E40 F800 ; prime nozzle
G1 Y100 F5000 ; quick wipe
M73 P1 ; enable build progress
```

Here's the message from Octolapse:
```Octolapse - Position Error
Position - Coordinates (X:-48.00000,Y:1.00000,Z:5.00000,E:0.00000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.```

FormerLurker, 2018-12-21:
Ok, for now set xmin to -40 (either in your octoprint profile or in the octolapse printer profile, let me know if you need help with that), and you'll get the results you want.

Also, fyi, i never imagined the home command causing this message, but looking at the code i see why it happens.  I may be able to find a better fix.  The message is intended to prevent stabilization from going out of bounds, no other reason.

FormerLurker, 2018-12-21:
Excuse me, -48.

SystemDisc, 2018-12-21:
Thank you. I'll do that for now.

FormerLurker, 2019-10-16:
I know this is an old issue, but I've finally taken care of the error popups.  The 'out of position error' shouldn't ever be displayed.  See [this issue](https://github.com/FormerLurker/Octolapse/issues/359) for details.

FormerLurker, 2019-11-09:
I'm going to close this out since it has been resolved.  Thanks!

closed 2019-11-09 by FormerLurker