FormerLurker/Octolapse #295: No "OK" Button when attempting to reset to defaults

talldonkey, 2018-12-22:
Octolapse Version: 0.3.4
OctoPrint Version: Version 1.3.10
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I was attempting to reset to defaults as per installation instructions https://github.com/FormerLurker/Octolapse/wiki/Installation
2. Settings->Octolapse->Reset Defaults

#### What should have happened?
3. Warning shows up with a OK or Dismiss button and gets the @#$ out of the way :)

#### What happened instead?
3. Warning shows up w/o a "OK" or "Dismiss" button
"
Restore Default Settings
You will lose ALL of your octolapse settings by restoring the defaults! Are you SURE?
"
4. Warning popup is persists and wont go away unless you reload the web browser

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.15.1

#### Browser and version of browser, operating system running browser

Browser: ChromeOS Version 71.0.3578.98 (Official Build) (64-bit)
Browser OS: MacOS 10.14.2

#### Link to plugin_octolapse.log
2018-12-22 01:27:21,544 Octolapse - loaded and active.
2018-12-22 01:27:27,526 Octolapse - is loading template configurations.

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/talldonkey/dc576ea0a00930df860966e1587dfdc0

#### Link to contents of Javascript console in the browser
[Violation] Forced reflow while executing JavaScript took 89ms
[Violation] Forced reflow while executing JavaScript took 66ms
[Violation] Forced reflow while executing JavaScript took 65ms

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: 
https://photos.app.goo.gl/oQiMoz8VGui5R33C9

FormerLurker, 2018-12-22:
This might take me a while to fix sinve i dont have access to a Mac.  I will see if i can get an open source license for Browser stack to help me debug.

FormerLurker, 2018-12-22:
Oh, i forgot to mention this earlier, but thanks for posting!  I'm sure you're not the only one with this problem, and i dont get much feedback from MacOS users.  Every bit helps!

talldonkey, 2018-12-22:
Thanks for the response!  I don't believe that it's MacOS related.  After
your response, I tried on my Pixelbook running ChromeOS (Version
71.0.3578.98 (Official Build) (64-bit)) and was able to exactly reproduce
the issue - you couldn't tell any difference from the Chrome running on
MacOS and ChromeOS.

Just for fun, I booted up my Linux virtual machine (Pixelbook Crostini,
Debian based) and ran Chrome, Same Symptoms.

Finally, to round this out, tried Safari on MacOS.  Exact same symptoms.

smeg1, 2018-12-22:
Tested on a Mac with safari and I do get the ok or cancel option
[image]

FormerLurker, 2018-12-22:
Hmmm.  Chrome for android and windows working here.  Could it maybe be some other plugin is interfering?  @taildonkey, can you try disabling all other plugins, rebooting, and trying to restore the defaults?

talldonkey, 2018-12-22:
Yes, I think your suspicions are correct about plugin interference.  I
disabled all plugins leaving only Octolapse (including the bundled system
plugin), and it worked!

I'll now proceed through toggling them on one at a time so we can zero in
on the interfering plugin.

FormerLurker, 2018-12-22:
Sweet!  Once you figure out which plugin it is, it should be a simple matter to determine what is causing the problem.  My bet is on css.  Thanks for spending time to help solve this!

talldonkey, 2018-12-22:
Zeroed in on it:  https://github.com/devildant/OctoPrint-ShutdownPrinter

Enabled all plugins except that one --> OK and Close button appear and seem
to function as intended (closing closes, OK looks like it resets stuff)
Enabled all plugins and turned that one too --> Boom repro symptoms!

FormerLurker, 2018-12-22:
And the hard work is done!  Ill take a look at this plugin and will let you know what i find.

devildant, 2018-12-31:
normally i fixed the problem, could you try this version of shutdownprinter: https://github.com/devildant/OctoPrint-ShutdownPrinter/archive/master.zip

FormerLurker, 2018-12-31:
@devildant, that took care of it, thank you so much for the super fast reply and fix!  I am impressed :)

@talldonkey, can you install the newest version of the shutdownprinter plugin by installing from this url within the plugin manger:  https://github.com/devildant/OctoPrint-ShutdownPrinter/archive/master.zip

Let me know if it also works for you, then I'll close my issue in @devildant's github page.

devildant, 2018-12-31:
@talldonkey I am also waiting for your feedback, if it works well, I will make a release.
normally it should work, I made a big cheating (but it's much cleaner): p
Thank you @FormerLurker for your investigation, I admit that I had not thought about the impact on other plugin

FormerLurker, 2018-12-31:
@devildant, you and me both!  These kinds of issues are pretty much unavoidable in an extensible environment like OctoPrint, but with team work we can fix them all (well, some anyway, lol!)  Thanks for your work on the shutdownprinter plugin, it's a great!

talldonkey, 2019-01-05:
Seems to be fixed!

Test Procedure:

Current (Release 0.1.2) version of Shutdown printer inactive/disabled:
OKay/Close button appears on the reset to default 

Re-Enabled Shutdown printer 0.1.2:
Okay/Close button does not appear as expected (repro'd again for confirmation)

Updated to 1.0.1 Release:
OKay/Close button appears on the reset to default

devildant, 2019-01-07:
thx for you feedback i will close the issue on my side