FormerLurker/Octolapse #297: Octolapse navbar button

ssttiijjnn, 2019-01-03:
#### Version of Octolapse
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.3.10 running on OctoPi 0.15.1

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred
OctoPrint was disabled, but still showing the NavBar button. I wanted to disable it. To disable the NavBar button I had to go into settings, enable the plugin, then disable the button in the settings, then disable the plugin again. 

#### What should have happened?
When OctoPrint is disabled the NavBar button should be gone. Or the NavBar button enable disable button in the settings should also be available when the octolapse is disabled.

#### What happened instead?
I had to go into settings, enable the plugin, then disable the button in the settings, then disable the plugin again.  (Because of this it's not a major issue)

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version:  0.15.1

FormerLurker, 2019-01-03:
There are several options within the main settings (octoprint settings->Octolapse->main tab) to control the visibility of the navbar link.  Let me know if those settings work for you.

Thanks for your post!

ssttiijjnn, 2019-01-03:
I can't find any way to edit the navbar button there.

This is my octoprint settings window
[image]

There is no way to enable or disable the navbar button, only a indication if it is enabled or disabled. After opening the settings I only see this
[image]

So i first have to enable the plugin to be able to change the navbar button here (last item)
[image]

And change the navbar tab and then disable the plugin again. Maybe i'm missing somthing

FormerLurker, 2019-01-03:
Set your navbar options within the main settings like so:

[image]

This way the navbar will only show when Octolapse (and a print) is running.

ssttiijjnn, 2019-01-03:
Thanks, that's nice to have.

The issue was mainly focused towards not being able to change settings when the plugin is disabled. Like i said, it's just a minor inconvenience because you can enable it, change it and disable it. This just feels counter intuitive for me though. But if this is ment to be this way then i understand

FormerLurker, 2019-01-03:
I've already gotten requests to change the 'disabled' behavior, but haven't gotten around to it yet :)  The next release should have this fixed.  Originally there were issues where users didn't notice they had the plugin disabled, so i wanted to make it more obvious, but I should have just put a big 'disabled' message somewhere instead!

Again, thanks for submitting your issue, and let me know if you have any other questions/requests!

FormerLurker, 2019-01-21:
@ssttiijjnn, I wanted to let you know that I am adding this feature to the next release (Navbar disabled when Octolapse is disabled).  The current behavior seems very silly now that you pointed this out to me.  Thanks again for your request!

FormerLurker, 2019-10-16:
@ssttiijjnn, I know this is an old issue, but I've finally gotten around to fixing the enabled/disabled problem you mentioned here.  Now when Octolapse is disabled all settings and tabs remain available, but there are obvious visual indicators showing that Octolapse is disabled.  Here is the main settings screen (which was in your original post:

[image]

Here is the tab when disabled:
[image]

FormerLurker, 2019-11-09:
Since this issue has been resolved, I'm going to go ahead and close it down.  Thanks for posting!

closed 2019-11-09 by FormerLurker