FormerLurker/Octolapse #298: [Enhancement] nicer video start and end

Phil1988, 2019-01-06:
Thanks for this awesome plugin!
I dont want to say anything about the problems but asking for an enhancement that might interessed a lot of people who use it.

The videos itself looks amazing, however I dont like that you cant see the start so good and also the finish vision is mostly blocked by the printhead.
Is it possible to capture a few pictures during the start (I am using a start script to prime the nozzle of my CR-10) and take a few pictures after the print is complete (I am using a finish script to move the printheat away and move the printbed to the front).
This would then make an better overall illusion of the whole print.

Thanks for any help, suggestions or work on this idea :)

FormerLurker, 2019-01-08:
@Phil1988,

First, thank you for your post!  It's great to get feedback, as well as feature requests.  However, I'm having trouble understanding some of the specifics.

When you say that, "you can't see the start so good", do you mean that the animation starts too abruptly?  If not, please give me a few more details here.

Second, when you say that, "your vision is mostly blocked by the printhead," do you mean that the pinthead is in some unexpected position?  can you give me some examples, or perhaps link to a video showing this?  This could be some sort of new bug that I haven't seen before.

Regarding your actual feature request, it sounds like you are suggesting adding pre and post roll video to the timelapse.  Currently Octolapse can 'simulate' this by simply repeating the first and/or last frame of the timelapse for a desired number of seconds.  You can look in the 'rendering' settings for that feature.  I've thought about adding actual video pre and post roll, but there are some technical challenges.  I think they could be solved, however.  Let me know if this lines up with what you had in mind.

Phil1988, 2019-01-08:
Thanks for your fast reply @FormerLurker !

For the beginning:
The video starts right at the layer zero with the printhead at its desired position.
This is good for fixed position with the printhead out of sight, but for the animated modes (orbit and printing) there could be a little more pre roll.

For the end: The printhead is at the right position. Octolapse does everything right here.
But for the orbit and printing mode the printhead is in line of sight with my camera position.
So there is no new bug here :)

I just uploaded a video I made yesterday for this purposes.
Please be nice to the low resolution - I didnt had the time to fix it yet ;)
https://youtu.be/8EApcUvx9js

As you see in the video, I already have your "simulated post roll" activated.

If you add this great feature, it would be sufficient if the user has an input field for the pre roll and the post roll, where we can enter some coordinates to take snapshots at.

For example on a 100x100mm print bed on the pre roll: 
X0 Y50
X10 Y50
X20 Y50
X30 Y50
X40 Y50
X50 Y50

Post roll:
X0 Y50
X10 Y50
X20 Y50
X30 Y50
X40 Y50
X50 Y50

I think this would do a huge improvement for the 2 animated modes.

Again: Thanks a lot for making this great plugin/addon. It really is a great feature for 3d printers :)

FormerLurker, 2019-01-08:
That's an interesting idea, thanks for the description!  Let me think this over a bit.

CWempe, 2020-04-27:
I fully support the feature request from @Phil1988 

To get a smooth transition between the start/end (slow growth of the model in the video) and the middle of the print (fast growth in the video) I would recommend to use [some kind of math function](https://math.stackexchange.com/questions/2660962/slow-start-slow-end-fade-curve/2660965) to calculate the photos per layer (or per time).

Maybe this helps to understand what I mean:

[image]

(drawn with https://excalidraw.com/)