FormerLurker/Octolapse #299: Camera Error - script failed

keawepoo, 2019-01-07:
Octolapse Version: Octolapse 0.3.4
OctoPrint Version: Version 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. Selected the gcode file
2. Clicked on "Load and Print"
3. Waited to see what happens
4. Around 8% A red popup alert happens "Octolapse - Camera Error
snapshot_script_error: The snapshot script failed with the following error message: sudo: no tty present and no askpass program specified
The snapshot was not found in the expected directory: '/home/pi/.octoprint/data/octolapse/snapshots/2cee60eb-2c71-46b2-982f-cabae9ef2b0a/354def78-9eea-409a-ad23-ee966dfff4ba/LTAZ6DV3_50mm-tower000000.jpg'."
5. Cancel print

#### What should have happened?
No error?

#### What happened instead?
It errored...

#### Operating System running OctoPrint and Octolapse
Raspberry Pi 3 Model B Plus Rev 1.3

#### Printer model & used firmware incl. version
Printer Model: Lulzbot Taz6
Printer Firmware Version: 1.1.9.28

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: Happens on more then 1 gcode file.

#### Link to settings.json
https://gist.github.com/keawepoo/d73dfbf643a6116d8c6a863f452299f7

#### Link to plugin_octolapse.log
https://gist.github.com/keawepoo/33113bc5c720b84959846d02e2522852

#### Link to octoprint.log
https://gist.github.com/keawepoo/d6ab7e5929f26ad60a72ccaf7145056c

#### Additional information
visudo
https://gist.github.com/keawepoo/892a04f10ffb3f94c3f16061d3141f76

take-snapshot.sh
https://gist.github.com/keawepoo/b8df6bd70119bf5ee97f3afa87081193

#################
I also ran "/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"" - And it came back working. But when I "./take-snapshot.sh" It comes back 
```javascript
pi@octopi:~ $ cd scripts
pi@octopi:~/scripts $ ./take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
[sudo] password for pi:
New file is in location /capt0000.jpg on the camera
File capt0000.jpg exists. Overwrite? [y|n] y
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
The snapshot was not found in the expected directory: ''.
pi@octopi:~/scripts $
```

Thank you.

FormerLurker, 2019-01-08:
Can you run this command:  
```
whereis gphoto2
```

Perhaps Gphoto2 isn't in the default location?  If whereis returns a different path, adjust the last line in your visudo file to match.

Note - If you received two notifications for this response, I apologize.  I accidentally sent an unresponse that missed all of the additional files you posted.  That response has been deleted.

keawepoo, 2019-01-08:
[image]

FormerLurker, 2019-01-08:
Ok, gphoto 2 appears to be installed at /user/local/bin/gphoto2, so change the last line of visudo to this:
```
pi ALL = (root) NOPASSWD: /usr/local/bin/gphoto2
```
then reboot.  That should do it!  Let me know either way, thanks!

keawepoo, 2019-01-08:
It works. I feel dumb for not seeing the /local/ part too... xD Thank you.

FormerLurker, 2019-01-08:
Excellent!  Don't feel bad, I missed it at first too!

Let me know if you have any other issues.  Closed!

closed 2019-01-08 by FormerLurker