FormerLurker/Octolapse #301: Can octolapse render with some missing image?

minw428, 2019-01-08:
Assume that there are some missed snapshot due to the problem of camera or communication.
Does the rendering process finish well in this case?

FormerLurker, 2019-01-08:
Yes it should.  There is a timeout setting within the camera profile.  If this timeout is reached, the snapshot acquisition stopps and the printing process will continue.  Once the print is completed Octolapse will combine whatever images were successful into the final timelapse movie.  Let me know if you have any additional questions!

FormerLurker, 2019-01-21:
I'm going to close this issue.  If you have any additional questions I can easily reopen it.   Thanks for your post!

Closed!

closed 2019-01-21 by FormerLurker