FormerLurker/Octolapse #303: How Octolapse capture picture at correct position

viron1988, 2019-01-15:
Hi,
I have a question and ask for kind help. I'm using the Gcode System Commands plugin to trigger a lighting on gcode when reaching a position. However I found out that the lighting gets triggered before reaching the position. So I found this: http://www.makermusings.com/2015/09/01/making-time-lapse-video-of-3d-prints/
where they use the following code:
G1 X66.533 Y100.927 E29.70731
G1 E28.70731 F1800.00000
G92 E0
G1 Z0.550 F7800.000 ; we put our extra bits after the Z position change command
G1 X0 Y0 F7800 ; move the printer to our ideal location for a photo
G4 P2000 ; pause for 2,000ms (2 seconds)
G4 P1 ; pause for 1ms — these are here to flush the g-code buffer
G4 P1
G4 P1
G4 P1
G4 P1
G4 P1
G4 P1
G4 P1
G4 P1
@execute /home/chris/getframe.py ; take a snapshot
G4 P1000 ; pause for 1 second to let the script execute
G1 X74.636 Y67.136 F7800.000
G1 E1.00000 F1800.00000
G1 X74.636 Y102.136 E1.25459 F1800.000

This is used to trigger the script (in my case lighting). They used the G4 P1 to flush the buffer in arduino.
I found this is still not accurate enough or not the optimal solution. So I'm asking what does Octolapse use for this?

Thanks in Advance

FormerLurker, 2019-01-15:
You can use the before and after snapshot script to trigger the lighting.  Just go to your Octolapse camera settings and add your script paths to the appropriate script fields.

Your scripts MUST be executable by the pi user, but if you're using gcode system commands that should already be true!  Let me know if you have any issues, and definitely let me know if it works!  Lighting is the reason I added the before/after snapshot script section :)

FormerLurker, 2019-01-16:
No, i try to wait the absolute minimum amount of time (no arbitrary waiting period required).  You can add a small delay for your camera to expose property, but maybe 0.25 sec or so (1/FPS*1000MS is ideal for a webcam).  I bypass the buffering issue to, as ill explain below.

First, when i decide it is time to take a snapshot i prevent octoprint from sending more gcode and send my own gcode to stabilize the printhead.  Then i send an m400 (wait for commands to finish) and finally an m114 (request current position).  Most printer firmware will not send the position response until after all buffered gcode is finished executing.  Receiving the position response is the trigger that starts the snapshot process.  Once the image is acquired, the printhead is (usually, unless the next gcode is travel only) returned to the original position.

If m400 doesn't work this way on your printer, adding an appropriate camera delay (see camera settings) will also work.

FormerLurker, 2019-01-16:
Oh, i should mention that i made my first stabilized timelapse with the gcode system command plugin.  It inspired me to create octolapse :)

FormerLurker, 2019-01-21:
I'm going to mark this as closed.  If you have any other issues/questions I can easily reopen it.

For now:  Closed!

Thanks for your submissions!

closed 2019-01-21 by FormerLurker