FormerLurker/Octolapse #304: Error message about retraction and detraction even though they are different

david-mikulin, 2019-01-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: Octolapse 0.3.4

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Set retraction speed to 35.1
2. Set detraction speed to 34.9

#### What should have happened?
I should not receive an error message that the two speeds are the same.

#### What happened instead?
I receive an error message that the two speeds are the same

#### Operating System running OctoPrint and Octolapse

OS Name: Mac OS 
Os Version: 10.14.2

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 Mk3
Printer Firmware Version: 3.5.1

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: [image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-01-19:
I saw this last night but it was too late to respond.  I really thought you found a bug, and in a way you did.  Octolapse should round these values as you type.  The good news is I'm currently working on enhancing and automating the slicer section of the printer profile, so it's a good time to tackle this.  I'll let you know when I have it corrected.  Also, I'm adding automatic slicer settings extraction directly from the printing gcode file.  Would you be interested in testing the dev build once I have it ready?  Let me know, and thanks for your submission!

david-mikulin, 2019-01-19:
Hi,

Sure, I will help test the dev build once you have it ready.

Cheers,
David

PlinyElderATX, 2021-01-30:
I'm using cura and get this error - "detraction" isn't an option in cura.....

FormerLurker, 2021-01-30:
@PlinyElderATX, check out [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#if-you-are-using-cura-follow-these-steps) for getting automatic slicer settings detection working for cura, and you won't need to worry about this anymore :)

FormerLurker, 2021-01-30:
@PlinyElderATX, btw, that's a great username 👍